volkszaehler/volkszaehler.org #483: Diagramme nach dem erneuten Laden der Seite fehlerhaft

BlueAndi, 2016-09-28:
Habe die aktuellen Änderungen von heute ausprobiert und bei mir zeigt sich folgendes Verhalten im IE11 und FF v49.0.1:

Nach dem Aktualisieren der Seite über den Browser wird im FF "nothing to plot" angezeigt und der IE zeigt nur eine Linie an, die nicht passt. Drücke ich auf Stunde, Tag, usw., dann wird alles korrekt angezeigt.

andig, 2016-09-28:
Kannst Du prüfen ob das Problem verschwindet wenn Du auf commit c0ce5bddffc8612cfe1cbe53f45eff1b409cbf33 zurück gehst? Also:

    git reset --hard c0ce5bddffc8612cfe1cbe53f45eff1b409cbf33

Falls nein- wird es besser wenn Du noch weiter zurück gehst?

BlueAndi, 2016-09-28:
Scheint am commit "Resize plot after loading channels" 73507040a6b2fcd21b94829f81e596d519f987b8 zu liegen. Davor gehts, ab dem gehts nicht mehr.

J-A-U, 2016-09-29:
Mit Opera 40.0 fast der selbe Effekt. Allerdings taucht "nothing to plot" nur sehr kurz auf.
Bei mir werden alle Zähler vom Typ "El. Energie (Zählerstände)" als Strich ab 00:00 Uhr dargestellt. Höhe entspricht dem aktuellen Wert. Temperaturen erscheinen gar nicht im Graph, in der Tabelle sind die Werte aber vorhanden.

Die Graphen werden dargestellt nachdem man eine der Tasten betätigt oder automatisch Aktualisiert wird.
Ich kann den Fehler auch reproduzieren wenn ich die komplette Seite refreshe.

andig, 2016-09-29:
Sollte jetzt wieder gehen. 

Leider ist das mehr als hässlich- die schöne Lösung das Chart und die Seite nach dem laden einmal größentechnisch anzupassen funktioniert damit nicht mehr. Die eigentliche Ursache habe ich noch nicht gefunden- mal sehen obs noch eine bessere Lösung geben wird...

closed 2016-09-29 by andig

andig, 2016-09-29:
Könnt ihr mal schauen ob es damit klappt: https://github.com/volkszaehler/volkszaehler.org/pull/486/files 

Damit sollte auch das Resizen der Kanäle wieder funktionieren und die Seite immer auf 100% skaliert werden es sei denn der Plot wird zu klein.

BlueAndi, 2016-09-29:
Mit Deinem eingspielten Bugfix gehts schonmal wieder. Was mir noch auffällt ist das nach einem Reload die Ansicht immer wieder auf der x-Achse sich auf eine Stunde einstellt, d. h. hatte man vorher einen Tag ist es anschließend wieder eine Stunde. War das ursprünglich nicht immer so, wie es zuletzt eingestellt war (Cookie)?

Versuche mir das mit dem pull request 486 noch anzuschauen bzw. muss mal schauen wie ich das in git lokal reinholen kann, bin da kein wirklicher Spezialist.

BlueAndi, 2016-09-29:
Also wenn ich es richtig gemacht habe,

` git fetch origin pull/486/head:pull_request_486`
`git checkout pull_request_486`

dann siehts gut aus. Bei mir werden die Diagramme korrekt nach einem Reload angezeigt. Getestet im IE11 und FF v49.0.1.

andig, 2016-09-29:
Also alles gut?

J-A-U, 2016-09-30:
Ja.