volkszaehler/volkszaehler.org #488: footer issue with chrome

schnello, 2016-10-01:
Hallo.

Habe mit aktuellen Stand vom git das Problem das die Fußzeile nicht richtig skaliert (siehe Bild)
[image]

Getestet mit Chrome, IE edge und Firefox

andig, 2016-10-01:
Mit welchrn Schritten? Nur laden?

Viele Grüße,
Andreas

schnello, 2016-10-01:
Einfach nur die Seite öffnen. Hier liegt leider das Problem jedoch scheinbar lokal bei mir:

> Also ich hab das jetzt nochmal getestet mit dem original stand bzgl der wui.js bzgl Line 45
> :   $('#entity-list').show(); // open entity list by default
> 
> Die hatte ich auskommentiert weil ich schon so viele Kanäle habe... damit die Kanalliste immer zu ist beim öffnen.
> Ist die Zeile aktiv, also die Kanalliste wird beim öffnen vollständig dargestellt, dann passt auch die Fußzeile. Also liegt das Problem leider schon auf meiner Seite indirekt. :(

Kann also geschlossen werden. Wenn es nötig ist bzgl der Darstellung die Kanäle zu laden dann muss es halt sein.

Grüße

andig, 2016-10-02:
> Kann also geschlossen werden. 

:)

closed 2016-10-02 by andig