volkszaehler/volkszaehler.org #502: Show UUID of new channel

J-A-U, 2016-10-25:
Von Gruppen mal abgesehen erstellt man Kanäle selten zum Selbstzweck sondern weil man sie mit Daten füttern möchte. Da wäre es doch angebracht, nach dem Erstellen des Kanals, die UUID gleich anzuzeigen.

Eine Meldung würde genügen, wenns einfacher ist vielleicht auch das Eigenschaften-Feld wie man es vom "i" kennt?

andig, 2016-10-25:
Ich klicke immer einfach auf das (i) und kann dann direkt copy/paste machen. Hint auf dem (i) nutzt ja nix da der sich dann nicht kopieren lässt. Reicht so nicht?

bgewehr, 2016-10-25:
Mir reicht das.

J-A-U, 2016-10-25:
Mir ist halt aufgefallen das wir immer mal wieder darauf verweisen das man Kanäle erstellen muss und wo man die UUID dann zu suchen hat.

Es erscheint mir einfach sinnvoller dem Nutzer die UUID gleich vorn Latz zu knallen.

BlueAndi, 2016-10-31:
Die UUID kurz in einem Bestätigungsdialog nach dem Erzeugen anzeigen, fände ich auch gut.

andig, 2016-11-01:
> Die UUID kurz in einem Bestätigungsdialog nach dem Erzeugen anzeigen, fände ich auch gut.

Das kostet 2 Klicks: (i), (ok).

Eine neue Implementierung kostet immer einen Klick um den Dialog loszuwerden. Ich sehe den Vorteil nicht bzw. mehr Belästigung für den Anwender nur um 1 Klick loszuwerden?

J-A-U, 2016-11-01:
Das ist keine Belästigung des Anwenders, der Anwender erstellt ja den Kanal damit er eine UUID bekommt.

BlueAndi, 2016-11-02:
Der Vorschlag ging nicht in Richtung einen Klick loszuwerden, auch wenn man Usability manchmal vermeintlich über die Anzahl an Klicks definieren mag. Ich hatte für mich gesprochen einfach am Anfang auch die Erwartungshaltung: Kanal anlegen und autom. UUID bekommen, da ich ohne ja nicht mehr weiterkomme.

Cool wäre die UUID auch vorgeben zu können. Ich habe etliche embedded Devices, in denen muss ich die UUID zur Compilezeit festlegen. Nach einem Neuaufsetzen musste ich anschließend in der Datenbank die UUIDs ändern. Ein neu compilieren und flashen der embedded Devices wäre natürlich auch möglich, aber das kostet mehrere Stunden bis hin zum Ausbau. Aber das ist eine andere Geschichte. ;-)

andig, 2016-11-02:
Hi Andreas,

> Der Vorschlag ging nicht in Richtung einen Klick loszuwerden, auch wenn man Usability manchmal vermeintlich über die Anzahl an Klicks definieren mag. Ich hatte für mich gesprochen einfach am Anfang [...]
Es ist immer wieder schön und hilfreich statt der Lösung mal die Anforderung zu sehen.
> Cool wäre die UUID auch vorgeben zu können. Ich habe etliche embedded Devices, in denen muss ich die UUID zur Compilezeit festlegen. Nach einem Neuaufsetzen musste ich anschließend in der Datenbank [...]
Ja, aber eine relativ kurze da es die Lösung schon gibt- zumindest fast für Deinen Fall:

unter middleware/iot/<name des owners>.json bekommst Du die Entity einschliesslich UUID ausgegeben wenn bei dieser der <owner> (ein meist leeres Feld) gepflegt ist.

Gebaut habe ich das mal für meinen ESP8366 der vom Hashwert der SensorUI aus immer den gleichen Kanal wieder finden sollte. 

Viele Grüße, Andreas

BlueAndi, 2016-11-03:
Hi Andreas,

> Einzubauen ist es einfach- mal in die `wui.js` schauen.

Ok, ich schaus mir mal an. Bin eher für embedded C/C++ geeignet. ;-)

> Es ist immer wieder schön und hilfreich statt der Lösung mal die Anforderung zu sehen.

Ja, da hast Du natürlich vollkommen Recht. Man ist einfach zu gern im Lösungsraum unterwegs.

> Ja, aber eine relativ kurze da es die Lösung schon gibt- zumindest fast für Deinen Fall: unter `middleware/iot/<name des owners>.json` bekommst Du die Entity einschliesslich UUID ausgegeben wenn bei dieser der <owner> (ein meist leeres Feld) gepflegt ist. Gebaut habe ich das mal für meinen ESP8366 der vom Hashwert der SensorID aus immer den gleichen Kanal wieder finden sollte.

Ah ... du meinst einen "Nutzer" Kanal anlegen und dort die UUID hinterlegen. Diese könnte man dann von außen über den abfragen, sprich so eine Art Alias, um an die UUID zu kommen. Hmm ... wäre möglich, jedoch habe ich z. T. embedded Devices mit wenig RAM ... das wird da u. U. schon haarig. Muss ich mir im Detail mal anschauen ... danke für den Hinweis!

Gruß
Andi

closed 2016-11-06 by andig