volkszaehler/volkszaehler.org #504: Define and document Content-Security-Policy

ypid, 2016-10-28:
Refer to: https://content-security-policy.com/

Could probably go into: http://wiki.volkszaehler.org/software/middleware/installation

andig, 2016-11-02:
Add this to your `.htaccess`:

    Header set Content-Security-Policy "default-src 'self'; connect-src * ws: wss: http: https:; script-src 'self' 'unsafe-inline'; style-src 'self' 'unsafe-inline';"

Won't add by default as it requires `mod_headers`.

andig, 2016-11-02:
Please feel free to update the wiki.

closed 2016-11-02 by andig

ypid, 2016-11-02:
Thanks!

Why not use:

```ApacheConf
<IfModule mod_headers.c>
    Header set Content-Security-Policy "default-src 'self'; connect-src * ws: wss: http: https:; script-src 'self' 'unsafe-inline'; style-src 'self' 'unsafe-inline';"
</IfModule>
```

?

No problem there with the `headers` module not being loaded.

andig, 2016-11-02:
Nice. Care to file a PR?

ypid, 2016-11-03:
Sure, done #509. Note that I currently have no instance running to test this with. I justed checked back because I wanted for a long time to set one up again and write a Ansible role to do the setup (would go into [debops-contrib](https://github.com/debops-contrib/)). Keep up the good work.

PR #509 (2016-11-20): Add Content-Security-Policy
Work and testing was done by @andig so @ypid committed in the name of
andig as andig did all the work and requested a PR :wink:
Also, the git commit is automatically gpg signed by ypid as all commits
committed by him.

Note that ypid has not tested this policy yet.

Closes: #504
commit bf7dddd2 (2016-11-20): Add Content-Security-Policy (#509)

Work and testing was done by @andig so @ypid committed in the name of
andig as andig did all the work and requested a PR :wink:
Also, the git commit is automatically gpg signed by ypid as all commits
committed by him.

Note that ypid has not tested this policy yet.

Closes: #504