volkszaehler/volkszaehler.org #506: Core Infrastructure Initiative (CII) Best Practices

ypid, 2016-10-28:
Hi,

I would like to point you to the [CII Best Practices Badge Program](https://bestpractices.coreinfrastructure.org). Could you go thought the [criteria](https://bestpractices.coreinfrastructure.org/criteria), [add the project](https://bestpractices.coreinfrastructure.org/projects/new) and keep it up-to-date there?

Refs:

* https://bestpractices.coreinfrastructure.org
* https://twit.tv/shows/floss-weekly/episodes/389

Related to:

* #505

andig, 2016-10-28:
> I would like to point you to the CII Best Practices Badge Program. Could you go thought the criteria, add the project and keep it up-to-date there?

Please feel free to do this if you think it adds value. Won't fix from my side.

ypid, 2016-10-28:
@andig That thing is that me as an outsider it would be very difficult to do that. For someone who is already involved this is much easier. And then they can also take direct action based on the self-assessment :wink:

andig, 2016-10-28:
Potentially but I'm lacking the benefit perspective plus don't see any priority.

ypid, 2016-10-28:
It gives you best practices for FOSS projects at hand and you can self-assess if you already meet them. This will result in higher quality and the like. You could listen (podcast) or watch to https://twit.tv/shows/floss-weekly/episodes/389

closed 2016-11-02 by andig