volkszaehler/volkszaehler.org #555: Merkliste: Erweiterungen fürs Image

andig, 2017-01-13:
- VZ Ordner von `/var/www` nach `/home/pi` verschieben -> kein sudo mehr für viele Befehle notwendig
  - PPM Middleware installieren
  - ggf. Pushserver installieren

frankrichter, 2017-02-09:
aus ML: /tmp als tmpfs sorgt wegen fixer Größe der Ramdisk für Probleme, wenn OS oder Anwendungen dort große Dateien zwischenlagern wollen. Besser /tmp auf SD belassen und Ramdisk-Lösung für /var/log integrieren: https://github.com/azlux/log2ram

frankrichter, 2017-02-10:
Hier schon mal ein auf dem aktuellen Image getestetes Config-File für logrotate:
```
pi@raspberrypi:~ $ cat /etc/logrotate.d/vzlogger
/var/log/vzlogger.log {
        rotate 1
        size=100k
        copytruncate
        missingok
        notifempty
}
```
Wichtig ist `copytruncate`, weil die Datei sonst erst gelöscht und dann neu erstellt wird. Damit kommt vzlogger nicht klar.
`rotate 0` geht natürlich auch, dann  wird die alte Datei nicht rotiert sondern gleich gelöscht.

J-A-U, 2017-10-02:
> •VZ Ordner von  /var/www  nach  /home/pi  verschieben -> kein sudo mehr für viele Befehle notwendig

Check: #636

christian-weiss, 2017-12-21:
It looks link this ticket is a memo than a ticket.

Would be nice if you can slit it into separate tickets:
- install.sh related (the stuff to avoid sudo is done, i think)
- docker image related (file an issue in: https://github.com/volkszaehler/docker)
- sd card related (raspbian; readme: installation recommendations)

J-A-U, 2017-12-21:
> It looks link this ticket is a memo than a ticket.

Yes, it is.

> Would be nice if you can slit it into separate tickets:

I prefer to keep the rpi-image related stuff tight together (kinda memo ;) ).
wiki.volkszaehler.org/development/rpi-image ?

> (the stuff to avoid sudo is done, i think)

Yes.

> docker image related

This is not about docker. If there are similarities, they are not intended.

J-A-U, 2017-12-31:
Ins Wiki übernommen. Sollen wir zu machen?

andig, 2017-12-31:
Ich würds offen lassen bis wir ein entsprechendes Image haben? Zumindest mir liegt GH näher als das Wiki.

andig, 2019-03-21:
Currently discussed on dev ml for vz 2.0, closing here.

closed 2019-03-21 by andig