volkszaehler/volkszaehler.org #557: Add a possibility to get real Min/Max values

SirkoVZ, 2017-01-16:
Currently Min/Max values are returned along with data tuples. However these Min/Max values are not necessarily the real Min/Max values from the time period, as the returing tuples are "condensed" and so the real Min/Max values get lost.

There should be a way to get only Min/Max values without tuples using the REST API.
Somthing like
GET http://demo.volkszaehler.org/middleware.php/data/57acbef0-88a9-11e4-934f-6b0f9ecd95a8.json?minmaxonly=true&from=01-01-2010&to=01-02-2010

The result should be similar to:

{"version":"0.3","data":{"uuid":"14360350-d080-11e2-ab3d-05959e28b10f","from":1484232240000,"to":1484318650000,"min":[1484289380000,83.198764424758],"max":[1484286340000,1192.0081471004]}}

andig, 2017-01-16:
> There should be a way to get only Min/Max values without tuples using the REST API.

Not sure. Especially for grouped requests (hour, day) it might make sense to return min/max per tuple group, too. That would also enable [line/area charts](http://www.telerik.com/clientsfiles/a1c6e0ef-a769-489f-929c-d2499884345a_targets.png?sfvrsn=0).

The effort is considerable- it means rewriting https://github.com/volkszaehler/volkszaehler.org/blob/master/lib/Util/Aggregation.php preferable using the interpreter API instead of SQL. That implies that also the interpreter SQLs need to be rewritten to take min/max calculation into account.

closed 2017-04-10 by andig

SirkoVZ, 2017-04-27:
I extended the VolkszählerApp some weeks ago to show tables with min/max values, but it is not very satisfiable to never see the real min/max values :-(

Sorry when I can't overlook the implications, but wouldn't it be possible to add the min/max values into the aggregation table only? Maybe using a SQL like (example) in Aggragation.php?

`SELECT * FROM volkszaehler.data AS d,
 (SELECT MIN(value) AS mini,
 	      MAX(value) AS maxi
   FROM volkszaehler.data where channel_id=12 and timestamp > 1451602834000 and timestamp < 1454281200000) AS m
WHERE channel_id=12 and timestamp > 1451602834000 and timestamp < 1454281200000 and (m.maxi = d.value
       OR m.mini = d.value);`

andig, 2017-04-27:
I would prefer to get rid of the entire aggregation SQL instead and use the Interpreter logic instead to reduce this complex code. PR would be welcome.

SirkoVZ, 2017-04-27:
hmm, sorry, my PHP is almost not existent