volkszaehler/volkszaehler.org #565: abonnierte öffentliche Kanäle sind inaktiv

frankrichter, 2017-01-26:
seit #560 sind Kanäle, die über Kanal hinzufügen > Öffentliche Kanäle abonniert werden, zunächst nicht aktiv. Ein Reload des Frontends behebt das.
Hinzufügen via Private Kanäle mittels UUID abonniert den gleichen Kanal direkt aktiv.

frankrichter, 2017-01-26:
Eben getestet: "aktiv" wird jetzt gesetzt (Checkbox angehakt, Chart wird geladen).

Allerdings wird auf diesem Weg offenbar keine push subscription mehr erzeugt: Es kommen keine Updates an, und wenn ich die Checkbox deaktiviere, kommt folgender JS-Error:  Uncaught not subscribed to topic (UUID).
Erneutes Aktivieren bringt dann auch funktionierendes push.

andig, 2017-01-27:
> Allerdings wird auf diesem Weg offenbar keine push subscription mehr erzeugt

Nachvollziehbar. So langsam nervt dieses Issue :/

andig, 2017-01-27:
Fixed via https://github.com/volkszaehler/volkszaehler.org/commit/aef52a54a148ca41569965140c4cf8352e4e2a7e

frankrichter, 2017-01-27:
> Fixed via aef52a5

Check.

Jetzt haben wir noch den Fall, dass ein in der DB vorhandenes active=false
immer noch berücksichtigt wird.
Hier weitermachen, neues issue, oder soll ich meine DB von allen active
flags säubern?

andig, 2017-01-27:
> Jetzt haben wir noch den Fall, dass ein in der DB vorhandenes active=false immer noch berücksichtigt wird.

Der sollte eigentlich behoben sein. Hast Du ein Beispiel?

> Hier weitermachen, neues issue, oder soll ich meine DB von allen active flags säubern?

Lieber Ursache finden und fixen.

frankrichter, 2017-01-27:
Am 27. Januar 2017 um 16:35 schrieb andig <notifications@github.com>:

> Jetzt haben wir noch den Fall, dass ein in der DB vorhandenes active=false
> immer noch berücksichtigt wird.
>
> Der sollte eigentlich behoben sein. Hast Du ein Beispiel?
>
Beispiel: abonnier' mal bei mir PV2001 oder PV2012, die haben active=false
in der DB

andig, 2017-01-27:
Getestet- sind bei mir (Chrome) auf Deiner Installation nach Abo alle "active"!

frankrichter, 2017-01-27:
Da muss ich wohl meinen Rechner entsorgen...

frankrichter, 2017-01-27:
Halt, direkt nach Abo sind sie bei mir auch active. Mach mal bitte einen
Reload.

2017-01-27 17:19 GMT+01:00 Frank Richter <frank.richter83@gmail.com>:

andig, 2017-01-27:
> Halt, direkt nach Abo sind sie bei mir auch active. Mach mal bitte einen Reload.

Soviel zum Thema steps to reproduce :/ Jetzt seh ichs auch...

frankrichter, 2017-01-27:
Sorry...

andig, 2017-01-27:
Finally see https://github.com/volkszaehler/volkszaehler.org/commit/6deb4683c9050bef11b3f37a4714bd7d51e1b451

andig, 2017-01-27:
Ich trau mir nicht das issue wieder zu zu machen :O

frankrichter, 2017-01-27:
Ich teste das nachher, aber jetzt erstmal Abendessen...

frankrichter, 2017-01-27:
> Ich trau mir nicht das issue wieder zu zu machen :O

Bei mir gehts, dann übernehme ich das zumachen :-)

andig, 2017-01-27:
war das zäh :O

jahir, 2017-02-01:
Ist das Absicht, daß Kanäle jetzt nach dem Laden _immer_ aktiv sind, auch wenn in der Datenbank active=0 steht?? Vor den Änderungen wurde das beachtet...
(Wobei das alte Verhalten auch nicht so optimal war. M.E. sollte der Aktiv-Zustand aus der DB nur beim Hinzufügen beachtet werden und dann die letzte Einstellung des FE in den cookies gespeichert werden, aber das ist eigentlcih ein anderes Thema :)

frankrichter, 2017-02-01:
Ja, ist Absicht, und die Cookie-Lösung ist auch angedacht. Wir waren uns einig, dass die Speicherung clientseitig mehr Sinn macht als in der DB. Mit dem aktuellen Frontend kann active gar nicht mehr in der DB gespeichert/geändert werden.
Wir hatten parallel ein längere Diskussion dazu in der ML. Der Thread hat allerdings ganz anders angefangen, deshalb heißt er jetzt leider "vzlogger an zweite middleware".
Zur Cookie-Sache wollte ich ein neues issue aufmachen - dann können wir hier wieder zumachen, oder?

andig, 2017-02-01:
> Ist das Absicht, daß Kanäle jetzt nach dem Laden immer aktiv sind, auch wenn in der Datenbank active=0 steht?? Vor den Änderungen wurde das beachtet...

Das habe ich tatsächlich auf Wunsch ausgebaut. M.E. macht nachvollziehbar inaktiv in der DB nicht soviel Sinn.

> (Wobei das alte Verhalten auch nicht so optimal war. M.E. sollte der Aktiv-Zustand aus der DB nur beim Hinzufügen beachtet werden und dann die letzte Einstellung des FE in den cookies gespeichert werden, aber das ist eigentlcih ein anderes Thema :)

Eben- und man stelle sich nur mehrere Devices vor. Wieso überhaupt "inaktive" Kanäle? Der Commit lässt sich wieder zurück drehen, aber dann ist @frankrichter unglücklich.

Speichern im Cookie ist leider nicht trivial da nur die root entities im Cookie stehen, die restliche Struktur aber von der MW geladen wird. Eine einfache Möglichkeit hierarchiche "active" Statuus mit den geladenen Strukturen zu vermergen ist mir bisher nicht eingefallen- wenn Du da eine Idee hast her damit!

Also entweder zurückdrehen oder damit leben bis wir (neues Issue) eine Cookielösung haben. Da bräuchte ich allerdings schlaue Ideen...

frankrichter, 2017-02-02:
> Eben- und man stelle sich nur mehrere Devices vor. Wieso überhaupt "inaktive" Kanäle? Der Commit lässt sich wieder zurück drehen, aber dann ist @frankrichter unglücklich.

Ich hab damit nicht angefangen, ich konnte mit dem bisherigen Zustand (active in der DB) eigentlich ganz gut leben. Aber mit #560 ist die Möglichkeit verschwunden, das active-property in der DB zu setzen/ändern, deshalb war es nur konsequent, ein eventuell in der DB befindliches active auch im Frontend zu ignorieren, damit da kein undurchschaubares Verhalten entsteht.
Inaktive Kanäle finde ich schon sinnvoll: Wenn ich viele Kanäle habe, möchte ich diese aus Komfortgründen gerne alle abonniert haben, aber nicht alle zusammen per default im Chart (weil ich dann nix mehr erkenne...). Gerade für Leute die private Kanäle verwenden (z.B. User von demo) geht das auch nur so, weil ständiges Löschen und wieder Abonnieren keine Option ist, wenn man immer mit UUIDs hantieren muss.
Ich käme also auch mit zurückdrehen klar, aber dann zum Zustand vor #560. Deutlich schicker fände ich allerdings die Cookie-Lösung, dann wäre das hier auch nicht alles umsonst gewesen...

> Speichern im Cookie ist leider nicht trivial da nur die root entities im Cookie stehen, die restliche Struktur aber von der MW geladen wird. 

Reicht es nicht erstmal, wenn man active für die root entities im Cookie speichert? Bisher wurde ein gespeichertes active ja auch von der Gruppe auf die Elemente vereerbt.

andig, 2017-02-02:
> Ich hab damit nicht angefangen, ich konnte mit dem bisherigen Zustand (active in der DB) eigentlich ganz gut leben. Aber mit #560 ist die Möglichkeit verschwunden, das active-property in der DB zu [...]

@frankrichter hat recht. Den Effekt habe ich beim aufräumen des UIs verursacht. Active kann seitdem nicht mehr über den Dialog gesetzt werden sondern nur über Checkbox in der Liste. Das hat aber offensichtlich auch dazu geführt dass active nixht mehr in der DB aktualsisiert wird.

Damit wäre es bereits vorher inkonsistent gewesen zwischen Checkbox in der Liste und active im Dialog.

Damit gibts zwei weitere Optionen:

  - C) wir drehen alles zurück- ist per git reset --hard schnell gemacht
  - D) ich baue das Speichern von active wieder ein- Aufwand/ Risiko wäre überschaubar

Speichern von active für Root entities reicht vmtl nicht für das Verhalten das @jahir erwartet. 

Ich bin für alles offen, die Situation habe ich ja auch (unabsichtlich) verursacht :/

andig, 2017-02-02:
Ps.: tatsächlich wäre es aber mal sinnig das Wunschverhalten komplett zu beschreiben. Was soll z.b. passieren wenn eine Gruppe auf inaktiv gesetzt wird (was sich auf die Childs vererbt)? Soll der Status für alle Childs gespeichert werden?

jahir, 2017-02-02:
Wollte schon was schreiben, aber ihr habt ja das meisten schon selbst aufgelöst :)
Ich meine, daß das benutzen von _active_ auch erst vor nicht allzu langer Zeit eingebaut wurde. Es ging damals darum, eine Möglichkeit zu schaffen, viele Gruppen/Kanäle im FE zu haben, aber nicht immer gleich alle zu laden, so daß man schnell zwischen den Gruppen umschalten kann. Ich finde die Möglichkeit sehr praktisch, einiges mehr (an Gruppen) per cookie im FE gespeichert zu haben, beim Laden aber erstmal nur die wichtigen zu sehen und den Rest bei Bedarf. Es gab damals glaube ich auch die Diskussion, daß man sowas wie "Views" einführt, mit einer Combobox oder so, so daß man zwischen verschiedenen Sammlungen von Gruppen/Kanälen umschalten kann. Die Implementierung wollte sich aber damals niemand antun :)
Könnte sich um den commit ef181e950aeb057f2ac9643f57685e66bb9d6564 handeln, wobei es da wohl nur darum ging, ein !active der gruppe auf alle Mitglieder anzuwenden.

andig, 2017-02-02:
Jetzt bleibt nur noch die Frage: was wollen wir denn :)

frankrichter, 2017-02-02:
Ich wäre in einem ersten Schritt für Speicherung des active flags der root entities in den Cookies. Damit haben wir alles was bisher funktioniert hat, aber pro Client statt zentral.
Für ein Wunschverhalten bei hierarchischen Strukturen haben wir nicht mal eine konsistente Formulierung. In der ML ging es vor kurzem auch darum: [[vz-users] Kanal Status der Checkboxen im frontend Persistenz](http://lists.volkszaehler.org/pipermail/volkszaehler-users/2017-January/010109.html). Das ging aber auch ohne klares Ergebnis zu Ende, zumindest kann ich mit der Beschreibung von Claus nicht so viel anfangen.

frankrichter, 2017-02-02:
Ich hab mir mal was ausgedacht:

Die root entities laden das active flag wie vorgeschlagen aus dem Cookie. Cookie wird bei jedem Checkbox-Klick angepasst, ein Reload ergibt also immer den letzten Stand.

Die gruppierten entities laden das active flag aus einer neuen Spalte aus der Tabelle entities_in_aggregator (statt aus properties wie bisher). Default ist active=true.
Das hat den schönen Effekt, dass ein Kanal in verschiedenen Gruppen sein und dort unterschiedlichen Status haben kann.
Für die Vererbung gilt beim Laden der Kanäle, dass bei deaktivierten parents auch alle childs deaktiviert sind, dass aktive parents aber inaktive childs haben können.
In die DB geschrieben wird ein geänderter Status nur, wenn zu diesem Zeitpunkt der übergeordnete Knoten aktiv ist.

Zwei Beispiele, wie das gemeint ist:

* Eine Gruppe in root mit active=false (Cookie) wird im Frontend als collapsed angezeigt. Finde ich gut und soll so bleiben. Jetzt expandiere ich die Gruppe und aktiviere einen einzelnen Kanal. Kanal wird natürlich angezeigt, aber in die DB wird nichts geschrieben, weil die Gruppe ja inaktiv ist und beim nächsten Reload wieder collapsed geladen wird, da macht ein aktiver Kanal darin keinen Sinn.

* Ich setze die übergeordnete Gruppe auf aktiv. Cookie wird angepasst. Alle childs erhalten den Status aus der DB (active=true per default). Jetzt kann ich einzelne childs per Checkbox deaktivieren oder wieder aktivieren, diese Änderung landet in der DB und wird damit beim nächsten Reload berücksichtigt, d.h. Gruppe wird aktiv und expanded angezeigt mit gewünschtem Subset an aktiven Kanälen.

Was meint ihr dazu? Nachteil wäre wohl, dass es eine Änderung am DB-Model dafür braucht, d.h. Update-Script für bestehende DBs und/oder Fallback für nicht angepasste DB erforderlich. Fallback wäre dann einfach die Vererbung von active vom Wurzelknoten auf alle childs.

andig, 2017-02-02:
Warum sollte die eine Art Status im Browser persistiert werden, aber die Andere in der DB? Wenn dann alle in der DB. In unterschiedlichen Tabellen ist ne witzige Idee, verfrixkelt das DB Schema aber ziemlich. Dann lieber einen Serialisierungsmechanismus für groups -> cookies überlegen imho.

andig, 2017-02-02:
Hier mal ein einfacher Vorschlag für die Root entities. Habs mit Gruppen noch nicht getestet, insbesondere nicht ob die beim Laden jetzt falls inaktiv auch hübsch zuklappen.

frankrichter, 2017-02-02:
> Warum sollte die eine Art active Status im Browser persistiert werden, aber die Andere in der DB?

Die DB war nur die Notlösung, weil du meintest dass es kompliziert sei die hierarchische Struktur als Cookie abzubilden. Wenn es geht, bin ich für die Cookie-Lösung.

Abgesehen von der technischen Realisierung: was ist mit der vorgeschlagenen Logik (inaktiv wird immer vererbt, aktive parents können aber inaktive childs haben)?

PR teste ich gleich.

frankrichter, 2017-02-03:
Ich nerve bestimmt, aber hilft ja nix. Ich hab nochmal einen Fall gefunden, in dem ein in der DB stehendes active=false noch ausgewertet wird: Beim Abonnieren von öffentlichen Gruppen erhalten die childs offenbar das active flag aus der DB. Nach Reload sind alle aktiv.

frankrichter, 2017-02-24:
Das im letzten comment beschriebene Verhalten lässt sich verhindern, indem man das `if` um 
[entity.js#L80](https://github.com/volkszaehler/volkszaehler.org/blob/master/htdocs/js/entity.js#L80) ausbaut, damit werden dann auch childs mit DB-Relikt active=0 konsistent behandelt.

andig, 2017-02-26:
> Das im letzten comment beschriebene Verhalten lässt sich verhindern, indem man das if um
entity.js#L80 ausbaut, damit werden dann auch childs mit DB-Relikt active=0 konsistent behandelt.

Was heisst das jetzt in Summe und angesichts https://github.com/volkszaehler/volkszaehler.org/pull/586?

frankrichter, 2017-02-27:
#586 rührt den active state nirgends an und hilft demnach hier nix (eben getestet).

Das if auszubauen war ne schlechte Idee, weil es die Cookie-Funktion kaputt macht. Sorry, das hätte ich hier gleich wieder rauseditieren können.
Besser, weil verträglich mit Cookie, ist die in #584 vorgeschlagene Lösung, die mir aber auch nicht gefällt.

`parseJSON` ist eigentlich nicht die richtige Stelle, um ein aus der DB angeliefertes active zu beseitigen, weil man dort nicht mehr unterscheiden kann, ob es aus der DB kam oder vom Cookie. Deswegen wäre IMHO der richtige Ansatz, alle active flags noch vor Weiterverarbeitung aus der MW-Antwort zu killen.

Ich finde da 2 Stellen, wo das passieren müsste:

1. für Cookie und private Kanäle in `Entity.prototype.loadDetails`

2. für öffentliche Kanäle in `Middleware.prototype.loadEntities`

An beiden Stellen findet sich auch schon Code um active aus dem JSON zu löschen:

1. `delete json.entity.active;` ([entity.js#L279](https://github.com/volkszaehler/volkszaehler.org/blob/master/htdocs/js/entity.js#L279))

Das funktioniert immerhin für das entity selbst, aber nicht für active innerhalb von eventuell vorhandenen children[].
Das müsste vom Prinzip her so aussehen:
```
delete json.entity.active;
delete json.entity.children[i].active;
delete json.entity.children[i].children[j].active;
...
```

2. `json.active = false;` ([middleware.js#L51](https://github.com/volkszaehler/volkszaehler.org/blob/master/htdocs/js/middleware.js#L51))

Das tut glaub ich gar nichts, weil das JSON, das es bearbeiten soll, folgende Struktur hat:
`{"version":"0.3","entities":[{"uuid":"...",...,"active":true,...,"children":[{},{}]},{...}]}`
Außerdem ist für konsistentes Verhalten löschen sicher besser als explizit false setzen, deshalb müsste das analog zu oben ungefähr so aussehen:
```
delete json.entities[i].active;
delete json.entities[i].children[j].active;
...
```

Im kontrollierten Test-Setup (1 Gruppe, 1 Kanal mit active=false) funktioniert das so auch.
Leider hab ich keinen Plan, wie man das sauber coden könnte, damit alle Elemente des Arrays und eventuell vorhandene children[] rekursiv behandelt werden.

Oder wäre es stattdessen vielleicht einfacher, die MW davon abzuhalten, active überhaupt rauszuschicken?

frankrichter, 2017-02-27:
> Das tut glaub ich gar nichts

Da hab ich mich wohl verhauen. Ich bin erst davon ausgegangen, dass `json` vor und hinter `forEach` das gleiche ist, aber so ist es wohl nicht.
Ändert aber nichts daran, dass active der children[] ausgefiltert werden sollte.

andig, 2017-02-27:
> Oder wäre es stattdessen vielleicht einfacher, die MW davon abzuhalten, active überhaupt rauszuschicken?

Das schaue ich mir an- sollte nicht so schwierig sein wäre vmtl. auch der bessere Fix gewesen :/

andig, 2017-02-27:
> Das funktioniert immerhin für das entity selbst, aber nicht für active innerhalb von eventuell vorhandenen children[].

> Das müsste vom Prinzip her so aussehen:

Ok, middleware.js#L51 fehlen die Children. Das Top Active funktioniert, das forEach läuft ja schon auf dem `entities` array. 

Die Lösung müsste wohl einschl. Children so aussehen:

````
		json.entities.forEach(function(json) {
			// fix https://github.com/volkszaehler/volkszaehler.org/pull/560
			json.active = false;
			var entity = new Entity(json, this.url);
			entity.each(function(child) {
				child.active = false;
			}, true); // recursive
			this.public.push(entity);
		}, this);
````

frankrichter, 2017-02-27:
Meinst du `entity.eachChild`?

andig, 2017-02-27:
yep

frankrichter, 2017-02-27:
Hab das eben mal reingehackt, im Prinzip funktioniert es. Allerdings sollte es `active = true` heißen, sonst machen wir das hier behandelte Problem zum Standardverhalten: öffentliche Gruppe abonnieren  > alle children inaktiv.
Dürfte ja eigentlich nicht schaden, wenn die mit active = true in `middleware.public` stehen, außer für das Dropdown-Feld in Kanäle hinzufügen wird es ja nicht gebraucht.

frankrichter, 2017-02-27:
Ich hab's mal so wie es bei mir funktioniert als PR zusammengestellt.

andig, 2017-02-27:
> Allerdings sollte es active = true heißen

Das können wir jetzt machen da das automatische `subscribe` aus der Entity raus ist.

PR #587 (2017-02-28): Fix active state for child entities
Fix #565
closed 2017-02-28 by andig