volkszaehler/volkszaehler.org #568: favicon wird nicht angezeigt

bgewehr, 2017-02-05:
das favicon wird nicht mehr angezeigt.

[image]

andig, 2017-02-05:
Kann ich nicht nachvollziehen- bitte mal git pull und neu laden.

andig, 2017-02-05:
Retested on demo, working.

frankrichter, 2017-02-05:
Habe ich allerdings auch. Cache löschen und Apache Neustart bereits getestet. Pfade für die Icons sind ok, werden aber nicht gefunden.

andig, 2017-02-05:
Na das is ja mal geil. Schauen wir ggf auch am Mittwoch an...

BerndGewehr, 2017-02-05:
Ich bekomme auch auf dem iPhone kein Volkszähler Icon, sondern das Chart als Icon

frankrichter, 2017-02-05:
Apache mag keinen Ordner namens `icons` im DocumentRoot, weil es dazu ein konfiguriertes Alias gibt:
[http://nixmash.com/php/solution-for-forbidden-icons-directory-in-apache/](http://nixmash.com/php/solution-for-forbidden-icons-directory-in-apache/)

andig, 2017-02-05:
Also machen wir einfach /ico draus statt Apache Changes zu erfordern?

Viele Grüße, Andreas

frankrichter, 2017-02-05:
Wird wohl das beste sein.

closed 2017-02-06 by andig