volkszaehler/volkszaehler.org #570: Versucht das Frontend IMMER ein live Update zu demo.volkszaehler.org?

bgewehr, 2017-02-05:
[image]

frankrichter, 2017-02-05:
Ja, ist normal. Wenn du Demo nicht verwendest, kannst du es in der options.js deaktivieren.

andig, 2017-02-05:
Das FE versucht ein Live update zu jeder konfigurierten MW.

closed 2017-02-05 by andig

frankrichter, 2017-02-05:
Wäre es nicht zu überlegen, die demo-MW in der Github-Version auzukommentieren? Dass sich jemand das Frontend selbst installiert, aber demo als MW nutzt, halte ich für ein eher exotisches Setup.

andig, 2017-02-05:
Warum- stört die?

frankrichter, 2017-02-05:
Es stört nicht wirklich, ist aber IMHO für die meisten Nutzer eine unnötige Option und erzeugt halt bei jedem Laden eines Nutzer-Frontends Anfragen an den demo-Server (öffentliche Kanäle und Push).

jahir, 2017-02-05:
Zumindest ist es verwirrend, wie schon dieser issue zeigt...
Abgesehen davon sehe ich das auch als privacy issue, demo.volkszaehler.org könnte damit viele clients tracken (die wenigsten dürften das auskommentieren). Und solange dort auch kein push-server läuft, bringt es nicht mal einen Mehrwert.

andig, 2017-02-06:
In 6 Jahren hat das 1x gestört- und auch nur weil Bernd in der Dev Konsole unterwegs war. Imho works as designed.

jahir, 2017-02-06:
Wir haben den push server schon seit 6 Jahren?? Wie die Zeit vergeht...
Mich hat's auch schon gestört, bzw. ich hab mich einfach gewundert (klar, als ich in der dev-console unterwegs war, sonst sieht man das ja nicht). Wobei die Meldung natürlich auch bei meinem eigenen Server kommt und dort immer ein 503 im Log erzeugt. Wäre vielleicht nicht schlecht, wenn man das einfach pro middleware abschalten könnte, sehe ich aber auch nur als minor issue an...

bgewehr, 2017-02-15:
Hab's jetzt manuell auskommentiert. Ist in Ordnung so.