volkszaehler/volkszaehler.org #571: Wenn live update aktiv ist, verschwinden die Werte aus dem Frontend

BerndGewehr, 2017-02-05:
Ich habe das Update mit gut pull gemacht.
Die Werte erscheinen alle wie gewohnt. Wenn der erste live-Update erfolgt, verschwinden die Werte Zeile für Zeile und kommen auch nicht mehr wieder.

BerndGewehr, 2017-02-05:
[image]
[image]
[image]

bgewehr, 2017-02-15:
[image]

Nun mit dem letzten Stand aus dem Repo neues Fehlerbild: Kein Kontakt zum Pushserver.

bgewehr, 2017-02-15:
Log sagt Service vzpusherser entered failed state. Ich muss sagen, am System sonst habe ich nichts geändert. Sudo git stash, sudo git pull, manuell options.js angepasst für push und install.sh ausgeführt.

bgewehr, 2017-02-15:
Neuinstallation:

```
root@volkszaehler:/home/pi# ./install.sh
volkszaehler.org installation script

checking prerequisites: php: /usr/bin/php
 mysql: /usr/bin/mysql
 awk: /usr/bin/awk
 sed: /bin/sed
 grep: /bin/grep
 wget: /usr/bin/wget
 mktemp: /bin/mktemp
 mkdir: /bin/mkdir
 git: /usr/bin/git

checking php version: 5.6.22-0+deb8u1 >= 5.4, ok

volkszaehler setup...
volkszaehler path? [/var/www/volkszaehler.org]
/var/www/volkszaehler.org already exists. remove it and get new git clone? (you have to type 'Yes' to do this!) [n] Yes
git clone volkszaehler.org into /var/www/volkszaehler.org
Cloning into '/var/www/volkszaehler.org'...
remote: Counting objects: 10800, done.
remote: Total 10800 (delta 0), reused 0 (delta 0), pack-reused 10800
Receiving objects: 100% (10800/10800), 7.10 MiB | 897.00 KiB/s, done.
Resolving deltas: 100% (5879/5879), done.
Checking connectivity... done.

checking composer...
/var/www/volkszaehler.org /home/pi
composer not found, downloading...
All settings correct for using Composer
Downloading...

Composer (version 1.3.2) successfully installed to: /var/www/volkszaehler.org/composer.phar
Use it: php composer.phar

/home/pi
composer: /var/www/volkszaehler.org/composer.phar

installing dependencies...
/var/www/volkszaehler.org /home/pi
Do not run Composer as root/super user! See https://getcomposer.org/root for details
Loading composer repositories with package information
Updating dependencies
Your requirements could not be resolved to an installable set of packages.

  Problem 1
    - Installation request for react/http dev-master#cd15204bd15d106d7832c680e4fb0ca0ce2f5e30 -> satisfiable by react/http[dev-master].
    - react/http dev-master requires react/socket ^0.5 -> satisfiable by react/socket[v0.5.0] but these conflict with your requirements or minimum-stability.
````

andig, 2017-02-15:
Was hat das mit dem Issue zu tun? Bitte nur ein Problem je Issue.

bgewehr, 2017-02-15:
Hab ein neues aufgemacht.

bgewehr, 2017-02-17:
So, nun bin ich auf dem letzten Repo-Stand und mein Push-Server läuft auch wieder. Danke erst mal dafür.

Dennoch bleibt der Fehler, dass die Werte meiner abonnierten Kanäle beim ersten live refresh verschwinden und auch nicht mehr wiederkommen. Lösche ich das Abo und abonniere erneut, bleiben die Werte stehen. Ich versuche mal zu parsen...

bgewehr, 2017-02-17:
[image]

An dieser Stelle ist die gesamte Zeile bereits gelöscht und wird neu gezeichnet - wenn this.data.rows nicht undefined wäre...

andig, 2017-02-17:
Hallo Bernd,

Ich hab Dich mehrfavh gefraht ob Du wirklich aktuelles repo oder meinen PR verwendest.. jetzt ist die Antwort plötzlich ja.  

Ganz ehrlich: ich habe keine Zeit für Beschäftigungsgherapie und bin nicht zu Deiner Bespassung da.

Ich verziechte ab jetzt draus Deine Probleme zu bearbeiten.

Sorry,
Andreas

BerndGewehr, 2017-02-17:
Ich habe auch wirklich mehrfach mein Verzeichnis gelöscht und neu geklont, das Repo sollte also auch wirklich jededmal frisch gezogen sein.

Ich habe im anderen Issue schon entschuldigend erwähnt, dass ich aber versehentlich eine ältere install.sh im home Verzeichnis und nicht aus dem misc/tools verwendet hatte und diese letztendlich den Unterschied ausgemacht hat, obwohl ich das an der commit history des files nicht nachvollziehen konnte. 

Dass es einen Issue mit den dependencies gab ist doch unbestritten, oder?

Wenn ich der Einzige mit dem refresh Fehler bin, dann können wir hier ja auch aufhören, dann muss es an was auch immer bei mir liegen.

Hab ganz sicher nicht absichtlich Deine Zeit verschwendet.

bgewehr, 2017-02-17:
Ich habe auch wirklich mehrfach mein Verzeichnis gelöscht und neu geklont, das Repo sollte also auch wirklich jededmal frisch gezogen sein.

Ich habe im anderen Issue schon entschuldigend erwähnt, dass ich aber versehentlich eine ältere install.sh im home Verzeichnis und nicht aus dem misc/tools verwendet hatte und diese letztendlich den Unterschied ausgemacht hat, obwohl ich das an der commit history des files nicht nachvollziehen konnte.

Dass es einen Issue mit den dependencies gab ist doch unbestritten, oder?

Wenn ich der Einzige mit dem refresh Fehler bin, dann können wir hier ja auch aufhören, dann muss es an was auch immer bei mir liegen.

Hab ganz sicher nicht absichtlich Deine Zeit verschwendet.

frankrichter, 2017-02-18:
Hallo Bernd,

ich hab mit dem aktuellen Stand aus dem Repo keine Probleme mit der Tabelle, diese wird bei ankommenden Push-Updates laufend aktualisiert.
Schieb mal testweise einen Kanal aus der Gruppe raus auf die root-Ebene und abonniere den einzeln. Ändert das was am Verhalten?

Gruß
Frank

bgewehr, 2017-02-18:
Immer wenn ich einen Kanal oder eine Gruppe von Kanälen frisch abonniert habe, bleibt er brav stehe und der refresh funktioniert. Wenn ich aber jetzt die Seite im Browser neu lade, verschwinden die Werte. Auch dann, wenn ich einen einzelnen Kanal direkt ohne Gruppen verwende.

frankrichter, 2017-02-19:
Kann ich bei mir nicht nachvollziehen. Mal einen anderen Browser probiert/Cache gelöscht?

bgewehr, 2017-02-20:
Ja, verschiedene Desktop und mobile Browser mit gelöschtem Cache führen zum gleichen Ergebnis. Wenn es niemanden sonst betrifft, dann weiß ich auch nicht weiter. Dies war der Grund für den total-reset und die Neuinstallation mit all dem, was dann geschah. Aber ich möchte hier auch keinen verärgern, wenn das nur bei mir auftritt - mein Pech, muss ich eben solange suchen, bis ich was gefunden habe. Danke erst mal!

bgewehr, 2017-02-22:
Ich habe jetzt mit frischem vz-Image und neuem Clone vom Git (der sich übrigens fehlerfrei installiert hat) wieder das gleiche Ergebnis. Die Wertezeilen verschwinden beim ersten live Update und bleiben leer.

Umgebung:
Meine DB liegt auf dem NAS, MariaDB auf synology
Pushserver zusätzlich mit /socket für node-red
In options.js push aktiviert und Demo.vollszaehler.org rausgenommen
volkszaehler.conf.php angepasst
Browser cache gelöscht

Rest ist Image-Standard.

Was kann ich tun, um den Fehler einzugrenzen?

frankrichter, 2017-02-22:
> Pushserver zusätzlich mit /socket für node-red

Hast du mal ohne /socket getestet?

> In options.js push aktiviert

D.h. `live: 8082`? Das läuft bei mir ohne den zusätzlichen Port über die in der Doku empfohlene Apache-Config.

andig, 2017-02-22:
Sollte am FE Verhalten egtl (tm) nix ändern. 

Viele Grüße,
Andreas

frankrichter, 2017-02-22:
Besonders wahrscheinlich erscheint mir das auch nicht, ich suche halt nach Unterschieden zwischen seiner und meiner (funktionierenden) Config...

bgewehr, 2017-02-22:
Notiz:
Wenn ich die Kanalgruppe neu abonniere, werden alle Werte korrekt angezeigt und die live Updates laufen.
Wenn ich den Browser manuell mit F5 aktualisiere (=load channels from cookie) verschwinden sie.

Die population des .rows js Objekts findet dabei in zwei verschiedenen Routinen statt, richtig? Dann ist die read_from_cookie evtl die, die fehlerhaft abläuft (evtl. nur bei meinen Daten?)

frankrichter, 2017-02-22:
Zeig mal die properties von einem betroffenen Kanal: `middleware.php/channel/UUID.json`
Und den Inhalt des Cookies `vz_entities`

bgewehr, 2017-02-23:
Properties:

````
{"version":"0.3","entity":{"uuid":"635481a0-6fcd-11e2-8587-eda25226f8bf","type":"powersensor","active":true,"color":"fuchsia","cost":0.24,"public":true,"style":"lines","title":"Strom","yaxis":"auto"}}
````

BerndGewehr, 2017-02-23:
Cookie kommt später...

frankrichter, 2017-02-24:
Ändert sich was wenn du den Kanal per Checkbox deaktivierst und wieder aktivierst?

BerndGewehr, 2017-02-24:
Ja, dann wird die Zeile angezeigt und auch aktualisiert.

frankrichter, 2017-02-24:
Nach Reload ist aber trotzdem wieder kaputt?

BerndGewehr, 2017-02-24:
Ja, genau!

frankrichter, 2017-02-24:
Ich kanns leider hier nicht nachvollziehen, aber schau dir mal #560 und #565 an, da hat sich einiges an active/inactive und push subscription geändert.
Der Grund für #560 war http://lists.volkszaehler.org/pipermail/volkszaehler-users/2017-January/010197.html
Andres Problem liest sich eigentlich genau wie deins, die Frage ist nur warum der Fix für ihn gut war und für dich nicht.

frankrichter, 2017-02-24:
Was du auch noch testen könntest: In der DB die Zeile mit active=true für einen betroffenen Kanal raus löschen. Das wird neuerdings nicht mehr dort gespeichert und soll eigentlich auch nicht mehr ausgewertet werden, wird es aber zumindest noch für Kanäle innerhalb von Gruppen, siehe mein letzter Post in #565.

frankrichter, 2017-02-24:
Jetzt kann ich es reproduzieren: wenn ich einen Kanal mit active=true in eine Gruppe schiebe, habe ich das Problem auch.
Aber das wollte ich schon weiter oben ausschließen: https://github.com/volkszaehler/volkszaehler.org/issues/571#issuecomment-280809585.
Offenbar hast du den Kanal nur zusätzlich solo abonniert, aber nicht aus der Gruppe entfernt. Wenn du das gemacht hättest was ich vorgeschlagen habe, wären wir vermutlich schon weiter.

frankrichter, 2017-02-24:
@andig: offenbar ist #560 nur für Kanäle auf root-Ebene gefixt, nicht für Kanäle in Gruppen.

frankrichter, 2017-02-24:
@BerndGewehr: Wenn du das Problem sofort und endgültig los werden willst, wirf alle active-Einträge aus der properties-Tabelle in der DB. Bin mir relativ sicher dass dann Ruhe ist.

BerndGewehr, 2017-02-24:
Hab ich gemacht. Es existiert nun kein active property mehr in der DB. Abonnement wurde beendet und neu eingerichtet. Leider gleiches Ergebnis. Nach Abonnement alles gut, nach browser reload verschwinden die Zeilen nach und nach beim Refresh. Soll ich die Gruppen mal löschen?

frankrichter, 2017-02-24:
Shit... Was ist mit Kanälen die in keiner Gruppe sind? Das geht bei mir
definitiv.

BerndGewehr, 2017-02-24:
Ich habe die Tabelle entities in aggregator geleert. Der öffentliche Kanal Strom wird nun beim Abo richtig geladen und bleibt nach dem Page reload stabil. Sorry, ich hatte Deine Aufforderung nicht so verstanden, habe nur einzeln abonniert, aber der Kanal war weiterhin in der Gruppe drin.

frankrichter, 2017-02-24:
Ich hab bei mir jetzt eine Testgruppe gemacht mit 3 Kanälen: 

1. keine active in DB > Tabelle leer bei Push-Update
2. active=1 > Tabelle leer bei Push-Update
3. active=0 > Tabelle ok

Also kommt bei nicht vorhandenem active wohl noch ein default active=1 von irgendwoher.
Wenn du active=0 überall ergänzen würdest, hättest du zumindest einen temporären Fix - sorry dass ich dich gerade ermuntert habe alle zu löschen...

frankrichter, 2017-02-24:
Wenn du in https://github.com/volkszaehler/volkszaehler.org/blob/master/htdocs/js/entity.js#L80 aus dem `true` ein `false` machst, klappt es auch bei nicht vorhandenem active flag.
Ich steck da allerdings nicht weit genug drin um zu sagen ob das irgendwelche Nebenwirkungen hat...

frankrichter, 2017-02-24:
@andig ich glaube es geht um diese Stelle: [entity.js#L95](https://github.com/volkszaehler/volkszaehler.org/blob/master/htdocs/js/entity.js#L95), dieses subscribe macht glaub ich nur Ärger.
Ich hab das auskommentiert und keinen Ärger mehr mit push bei child entities.

frankrichter, 2017-02-24:
Ne, perfekt ist das auch nicht: Wenn ich die Gruppe neu abonniere, bekommen die childs jetzt keine push subscription mehr, erst nach einem Reload. Es müsste also idealerweise eine Fallunterscheidung rein, ob es sich um ein neues oder um ein aus dem Cookie geladenes Abo handelt.

frankrichter, 2017-02-24:
@BerndGewehr auch wenn das noch nicht ideal ist, wäre es sicher ein brauchbarer Workaround für dich:
In `https://github.com/volkszaehler/volkszaehler.org/blob/master/htdocs/js/entity.js#L95` die 3 folgenden Zeilen auskommentieren, dann kommen zwar bei neuem Abo nicht sofort push updates, aber nach einem Reload ordnungsgemäß mit intakter Tabelle und unabhängig von eventuellen active flags in der DB.

BerndGewehr, 2017-02-24:
Probiere ich aus! Danke!

BerndGewehr, 2017-02-24:
Es sieht für mich so aus, als ob die Gruppen ihren aktiv Status immer noch aus der DB beziehen. Ich kann nämlich nicht im Browser Cookie speichern, ob sie aktiv sein sollen, aber eine Änderung in der DB bewirkt das. Nachdem aber jetzt die oberste Ebene meiner Hierarchie, das Gebäude, active=0 hat, das aber aus dem Cookie gelesen wird (bleibt persistent unabhängig vom DB wert), habe ich nun einen Zustand, der erst mal funktioniert, ohne in der entity.js etwas geändert zu haben.
Gebäude dBactive=0, im Cookie = 1
Gruppe Zähler dbactive=1, cookie egal
Kanäle dbactive=0, Cookie egal
Erzeugt eine erfolgreiche Anzeige inkl. refresh, auch nach Neuladen der Seite.

frankrichter, 2017-02-24:
Speicherung von active im Cookie ist aktuell nur für root entities implementiert (alles andere ist nicht trivial, Story dazu in #565 mittendrin).
childs von Gruppen sind per default aktiv, außer sie haben active=0 in der DB, das liegt an [entity.js#L79](https://github.com/volkszaehler/volkszaehler.org/blob/master/htdocs/js/entity.js#L79).
Deine Lösung mag für dich im Moment sein, ist aber nix für die Zukunft, weil bei neu erstellten Kanälen/Gruppen gar kein active mehr in der DB landet - manuell in der DB rumzupfuschen ist ja auch nix...

> Erzeugt eine erfolgreiche Anzeige inkl. refresh, auch nach Neuladen der Seite.

Wahrscheinlich sogar _nur_ nach Reload, oder? Wenn du eine Gruppe mit childs, die active=0 in der DB haben, neu abonnierst, sind die childs erstmal nicht aktiv, zumindest ist das hier so (siehe #565 ganz unten).

BerndGewehr, 2017-02-24:
Richtig, habe für die Gruppe der Ebene 2 ein active=1 gesetzt. Lädt sofort richtig mit live refresh.

frankrichter, 2017-02-24:
Ok, dann ist es bei verschachtelten Gruppen nochmal anders.
Und was passiert wenn du die Gruppe "Zähler" als root entity abonnierst?

frankrichter, 2017-02-24:
Ach so: ich hab vorhin auch festgestellt, dass die push subscription ein Löschen des Kanals (Cookies) überlebt. Du musst also so testen: Root-Gruppe (Gebäude) löschen, Reload des dann leeren Frontends, dann neu abonnieren. Dann kriegst du wahrscheinlich auch keine Updates direkt nach dem Abo, oder?

frankrichter, 2017-02-24:
@andig was ich oben bzgl. Fallunterscheidung vorgeschlagen habe, geht glaub ich in die falsche Richtung.
Ich denke subscribe hat in parseJSON eigentlich nichts verloren, weil es für nicht abonnierte Kanäle keinen Sinn macht.
Wenn ich das richtig sehe, wird die push subscription für ein neu abonniertes root entity in [wui.js#L228](https://github.com/volkszaehler/volkszaehler.org/blob/master/htdocs/js/wui.js#L228) erzeugt. Das müsste dann an dieser Stelle auch rekursiv für die childs passieren, oder?

BerndGewehr, 2017-02-24:
Also ich habe jetzt das Abo gelöscht, dann reload, dann neu abonniert, Gebäude active, Gruppe Zähler active, Kanäle inactive, keine Daten. Neu Laden -> alle Kanäle active, Daten und live updates OK.

frankrichter, 2017-02-24:
Yep, so hab ich das erwartet. Damit müsstet du doch mal leben können bis wir eine schicke Lösung haben, oder?

BerndGewehr, 2017-02-24:
Ja, klar!!

frankrichter, 2017-02-24:
@andig Ich hab grad mal durchgeschaut, es gibt ja 4 Wege wie in neuer Kanal ins Frontend kommt:

1. aus den Cookies > da werden rekursiv push subsriptions erzeugt (vermutlich in [init.js#L145](https://github.com/volkszaehler/volkszaehler.org/blob/master/htdocs/js/init.js#L145))
2. Abo öffentlicher Kanal > da erfolgt nur subscribe für root entity, nicht rekursiv
3. Abo privater Kanal > keine push subscription
4. Erstellung neuer Kanal > ungetest, aber vermutlich auch keine push subscription

Kriegen wir zumindest 2 - 4 vereinheitlicht, wenn wir in `vz.wui.addEntity` ein rekursives `entity.subscribe()` einbauen?

andig, 2017-02-24:
@frankrichter gute Analyse :)

Das subcribe aus parsejson raus zu nehmen habe ich auch schon überlegt, mir waren es aber einfach zuviele Stellen wo es wieder rein gemusst hätte- aber da läufts ja drauf hinaus.

  1) ruft m.W. `loadDetails` auf um Gruppen/Eigenschaften zu laden
  2) wurde vorab schon vollständig geladen, da dürfte nur `subscribe` fehlen
  3) das ist dann noch ein Bug
  4) kein Bug weil solange der Kanal nicht da ist kann ja auch kein Logger auf den Kanal pushen.

Long Story Short: 1/2 in Ruhe anschauen, 3 sollte analog 1) sein.

frankrichter, 2017-02-24:
Danke für die Blumen - noch effektiver wär das wenn ich etwas mehr von JavaScript verstehen würde - da muss ich noch viel lernen...

hab ein Lösung für 2-4: in wui.js 
nach [L141](https://github.com/volkszaehler/volkszaehler.org/blob/master/htdocs/js/wui.js#L141) `entity.subsribe();` und
nach [L143](https://github.com/volkszaehler/volkszaehler.org/blob/master/htdocs/js/wui.js#L143) `child.subscribe();` einfügen
[L228](https://github.com/volkszaehler/volkszaehler.org/blob/master/htdocs/js/wui.js#L228) kann dann raus

zu Punkt 4: tut aber auch nicht weh wenn es damit auch erschlagen wird, oder? Ich könnte ja die angezeigte UUID des neuen Kanals sofort in die gerade editierte vzlogger.conf packen und den Logger starten. Dann bekomme ich sofort Push Updates ohne Frontend Reload :-)

frankrichter, 2017-02-24:
Mein erster PR ;-)
Ich glaub' das ist übersichtlicher als meine Vorschläge aus einem Haufen Kommentare rauszufiltern...

andig, 2017-02-26:
Zur Doku:

  - laden aus Cookie:
    1. neue rudimentäre Entities erzeugen
    2. für _alle_ Entities Details laden
    3. subcribe rekursiv ausführen wenn die MW jeweils connected ist

  - laden aus "public":
    1. (inaktive) Entity aus der public Liste entnehmen
    2. active setzen
    3. via addEntity rekursiv subscriben

  - laden aus "private":
    1. neue rudimentäre Entity erzeugen (analog cookie)
    2. Details laden (analog cookie)
    3. via addEntity rekursiv subscriben

Ich überlege nochmal ob sich nicht cookie analog private umsetzen ließe- dann liefe alles nach einem Muster

andig, 2017-02-26:
Funktioniert https://github.com/volkszaehler/volkszaehler.org/pull/586 für Euch? Analog Version von Frank mit etwas anderer Syntax.

frankrichter, 2017-02-26:
Testen kann ich erst später, aber ist ja im Prinzip genau das gleiche.

closed 2017-02-26 by andig

frankrichter, 2017-02-26:
pulled, tested, approved

frankrichter, 2017-02-26:
> Ich überlege nochmal ob sich nicht cookie analog private umsetzen ließe- dann liefe alles nach einem Muster

Klingt gut. Naheliegend wäre ja, in init.js einfach für jedes aus dem Cookie geladene Entity `addEntity` aufzurufen. Das ist aber wahrscheinlich nix, weil das jedesmal ein `drawPlot` auslösen würde. Im Moment werden ja erst alle Daten geholt und dann geplottet, das ist sicher effizienter (und schöner...).