volkszaehler/volkszaehler.org #576: Add new Entity definition

SirkoVZ, 2017-02-13:
Ich hätte gern für meine Zisterne in
/lib/Definition/EntityDefinition.json:
 {
                 "name"                  : "filllevel",
                 "optional"              : ["resolution","tolerance", "cost", "local"],
                 "icon"                  : "waterdrop.png",
                 "unit"                  : "l",
                 "interpreter"           : "Volkszaehler\\Interpreter\\SensorInterpreter",
                 "model"                 : "Volkszaehler\\Model\\Channel",
                 "translation"           : {
                         "de" : "Füllstand",
                         "en" : "Fill Level",
                         "fr" : "Niveau de remplissage"
                 }
}

andig, 2017-02-15:
Entweder so oder Du benutzt den "generischen" Sensor, da kannst Du auch selbst die Einheit mit angeben. Wenn so dann gerne PR ;)

SirkoVZ, 2017-02-16:
PR ist da :-)

andig, 2017-02-16:
Einfacher ist es wenn Du PR https://github.com/volkszaehler/volkszaehler.org/pull/578 verlinkst ;)

andig, 2017-02-21:
Merged.

closed 2017-02-21 by andig