volkszaehler/volkszaehler.org #577: Änderungen in https://github.com/reactphp/http müssen eingearbeitet werden

bgewehr, 2017-02-15:
````
    "require": {
        "php": ">=5.3.0",
        "ringcentral/psr7": "^1.2",
        "react/socket": "^0.5",
        "react/stream": "^0.4.4",
        "evenement/evenement": "^2.0 || ^1.0"
````

Bei volkszaehler.org

````
    "require": {
        "php": ">=5.6.0",
        "doctrine/orm": "^2.5",
        "doctrine/dbal": "^2.5",
        "andig/dbcopy": "dev-master",
        "symfony/console": "^2.6|^3.0",
        "symfony/http-kernel": "^2.6|^3.0",
        "symfony/http-foundation": "^2.6|^3.0",
        "symfony/routing": "^2.6|^3.0",
        "cboden/ratchet": "0.3.*",
        "react/socket": "0.4.*",
        "react/http": "dev-master#cd15204bd15d106d7832c680e4fb0ca0ce2f5e30"
````

andig, 2017-02-15:
Was tust Du genau? Mit sauberer Installation reproduzierbar? Der Stand von "react/http": "dev-master#cd15204bd15d106d7832c680e4fb0ca0ce2f5e30" hat sich nicht geändert, insofern gibts da nix nachzuziehen?

bgewehr, 2017-02-16:
Ich habe das install.sh script benutzt und auf die Frage des neuen clones Yes geantwortet.

Dann läuft das soweit, bis der composer die o. a.  Fehlermeldung mit den dependencies auswirft.

andig, 2017-02-16:
Das ist wirklich hässlich. Müssen wir abwarten, siehe https://github.com/composer/composer/issues/6166

andig, 2017-02-16:
Bitte mal testen.

BerndGewehr, 2017-02-16:
Leider nein:

Loading composer repositories with package information
Updating dependencies
Your requirements could not be resolved to an installable set of packages.

  Problem 1
    - Installation request for react/http dev-master#cd15204bd15d106d7832c680e4fb0ca0ce2f5e30 -> satisfiable by react/http[dev-master].
    - react/http dev-master requires react/socket ^0.5 -> satisfiable by react/socket[v0.5.0] but these conflict with your requirements or minimum-stability.

BerndGewehr, 2017-02-16:
Mit 0.4.* geht's.

Allerdings läuft dann der Pushserver nicht:
PHP Fatal error:  Call to undefined method React\Http\Request::getBody() in /var/www/volkszaehler.org/lib/Server/HttpReceiver.php on line 54
ro

andig, 2017-02-16:
> Mit react/http 0.4.* geht's

Interessiert nicht :( Hast Du meine Änderung überhaupt ausprobiert?

BerndGewehr, 2017-02-16:
Naja, ich hab das wiederholt, was ich vorher auch schon gemacht hatte. Reicht das oder nicht?

BerndGewehr, 2017-02-16:
Ich habe den Code manuell in die composer.json eingefügt. Läuft durch.

Pushserver läuft trotzdem nicht.
Volkszaehler Push Server
Listening for updates on 5582. Clients may connect at 8082.
PHP Fatal error:  Call to undefined function GuzzleHttp\Psr7\parse_request() in /var/www/volkszaehler.org/vendor/react/http/src/RequestParser.php on line 98

andig, 2017-02-16:
Bitte neues Issue mit Inhalt composer.lock

closed 2017-02-16 by andig