volkszaehler/volkszaehler.org #580: Vzpushserver startet nicht

BerndGewehr, 2017-02-16:
Fehlermeldung:
pi@volkszaehler:~ $ sudo /usr/bin/php /var/www/volkszaehler.org/misc/tools/push-server.php
Volkszaehler Push Server
Listening for updates on 5582. Clients may connect at 8082.
PHP Fatal error:  Call to undefined function GuzzleHttp\Psr7\parse_request() in /var/www/volkszaehler.org/vendor/react/http/src/RequestParser.php on line 98

Composer.lock:

{ "_readme": [ "This file locks the dependencies of your project to a known state", "Read more about it at https://getcomposer.org/doc/01-basic-usage.md#composer-lock-the-lock-file",         "This file is @generated automatically" ], "content-hash": "374e75c3b44eb42dd87ac361f9e1afbb", "packages": [ { "name": "andig/dbcopy", "version": "dev-master", "source": { "type": "git", "url": "https://github.com/andig/dbcopy.git", "reference": "00481a1dd6905cfa51aafb96c3eff08b5bf0c169" }, "dist": { "type": "zip", "url": "https://api.github.com/repos/andig/dbcopy/zipball/00481a1dd6905cfa51aafb96c3eff08b5bf0c169", "reference": "00481a1dd6905cfa51aafb96c3eff08b5bf0c169", "shasum": "" }, "require": { "doctrine/dbal": "^2.5", "symfony/console": "^2.6|^3.0" }, "bin": [                 "dbcopy.php" ], "type": "project", "autoload": { "psr-0": { "DatabaseCopy\\": "./" } }, "notification-url": "https://packagist.org/downloads/", "license": [ "MIT" ], "homepage": "http://github.com/andig/dbcopy/", "time": "2016-12-27 16:09:21" }, { "name": "cboden/ratchet", "version": "v0.3.6", "source": { "type": "git", "url": "https://github.com/ratchetphp/Ratchet.git", "reference": "84df35d2a6576985b9e81b564d3c25809f8d647e" }, "dist": { "type": "zip", "url": "https://api.github.com/repos/ratchetphp/Ratchet/zipball/84df35d2a6576985b9e81b564d3c25809f8d647e", "reference": "84df35d2a6576985b9e81b564d3c25809f8d647e", "shasum": "" }, "require": { "guzzle/http": "^3.6", "php": ">=5.3.9", "react/socket": "^0.3 || ^0.4", "symfony/http-foundation": "^2.2|^3.0", "symfony/routing": "^2.2|^3.0" }, "type": "library", "autoload": { "psr-4": { "Ratchet\\": "src/Ratchet" } }, "notification-url": "https://packagist.org/downloads/", "license": [ "MIT" ], "authors": [ { "name": "Chris Boden", "email": "cboden@gmail.com", "role": "Developer" } ], "description": "PHP WebSocket library", "homepage": "http://socketo.me", "keywords": [ "Ratchet", "WebSockets", "server", "sockets" ], "time": "2017-01-06T14:36:36+00:00" }, { "name": "doctrine/annotations", "version": "v1.3.1", "source": { "type": "git", "url": "https://github.com/doctrine/annotations.git", "reference": "bd4461328621bde0ae6b1b2675fbc6aca4ceb558" }, "dist": { "type": "zip", "url": "https://api.github.com/repos/doctrine/annotations/zipball/bd4461328621bde0ae6b1b2675fbc6aca4ceb558", "reference": "bd4461328621bde0ae6b1b2675fbc6aca4ceb558", "shasum": "" }, "require": { "doctrine/lexer": "1.*", "php": "^5.6 || ^7.0" }, "require-dev": { "doctrine/cache": "1.*", "phpunit/phpunit": "^5.6.1" }, "type": "library", "extra": { "branch-alias": { "dev-master": "1.4.x-dev" } }, "autoload": { "psr-4": { "Doctrine\\Common\\Annotations\\": "lib/Doctrine/Common/Annotations" } }, "notification-url": "https://packagist.org/downloads/", "license": [ "MIT" ], "authors": [ { "name": "Roman Borschel", "email": "roman@code-factory.org" }, { "name": "Benjamin Eberlei", "email": "kontakt@beberlei.de" }, { "name": "Guilherme Blanco", "email": "guilhermeblanco@gmail.com" }, { "name": "Jonathan Wage", "email": "jonwage@gmail.com" }, { "name": "Johannes Schmitt", "email": "schmittjoh@gmail.com" } ], "description": "Docblock Annotations Parser", "homepage": "http://www.doctrine-project.org", "keywords": [ "annotations", "docblock",                 "parser" ], "time": "2016-12-30T15:59:45+00:00" }, { "name": "doctrine/cache", "version": "v1.6.1", "source": { "type": "git", "url": "https://github.com/doctrine/cache.git", "reference": "b6f544a20f4807e81f7044d31e679ccbb1866dc3" }, "dist": { "type": "zip", "url": "https://api.github.com/repos/doctrine/cache/zipball/b6f544a20f4807e81f7044d31e679ccbb1866dc3", "reference": "b6f544a20f4807e81f7044d31e679ccbb1866dc3", "shasum": "" }, "require": { "php": "~5.5|~7.0" }, "conflict": { "doctrine/common": ">2.2,=2.4,=5.3.2" }, "require-dev": { "phpunit/phpunit": "4.*", "symfony/console": "2.*||^3.0" }, "suggest": { "symfony/console": "For helpful console commands such as SQL execution and import of files." }, "bin": [ "bin/doctrine-dbal" ], "type": "library", "extra": { "branch-alias": { "dev-master": "2.5.x-dev" } }, "autoload": { "psr-0": { "Doctrine\\DBAL\\": "lib/" } }, "notification-url": "https://packagist.org/downloads/", "license": [ "MIT" ], "authors": [ { "name": "Roman Borschel", "email": "roman@code-factory.org" }, { "name": "Benjamin Eberlei", "email": "kontakt@beberlei.de" }, { "name": "Guilherme Blanco", "email": "guilhermeblanco@gmail.com" }, { "name": "Jonathan Wage", "email": "jonwage@gmail.com" } ], "description": "Database Abstraction Layer", "homepage": "http://www.doctrine-project.org", "keywords": [ "database", "dbal", "persistence", "queryobject" ], "time": "2017-02-08T12:53:47+00:00" },         { "name": "doctrine/inflector", "version": "v1.1.0", "source": { "type": "git", "url": "https://github.com/doctrine/inflector.git", "reference": "90b2128806bfde671b6952ab8bea493942c1fdae" }, "dist": { "type": "zip", "url": "https://api.github.com/repos/doctrine/inflector/zipball/90b2128806bfde671b6952ab8bea493942c1fdae", "reference": "90b2128806bfde671b6952ab8bea493942c1fdae", "shasum": "" }, "require": { "php": ">=5.3.2" }, "require-dev": { "phpunit/phpunit": "4.*" }, "type": "library", "extra": { "branch-alias": { "dev-master": "1.1.x-dev" } }, "autoload": { "psr-0": { "Doctrine\\Common\\Inflector\\": "lib/" } }, "notification-url": "https://packagist.org/downloads/", "license": [ "MIT" ], "authors": [ { "name": "Roman Borschel", "email": "roman@code-factory.org" }, { "name": "Benjamin Eberlei", "email": "kontakt@beberlei.de" }, { "name": "Guilherme Blanco", "email": "guilhermeblanco@gmail.com" }, { "name": "Jonathan Wage", "email": "jonwage@gmail.com" }, { "name": "Johannes Schmitt", "email": "schmittjoh@gmail.com" } ], "description": "Common String Manipulations with regard to casing and singular/plural rules.", "homepage": "http://www.doctrine-project.org", "keywords": [ "inflection", "pluralize", "singularize", "string" ], "time": "2015-11-06T14:35:42+00:00" }, { "name": "doctrine/instantiator", "version": "1.0.5", "source": { "type": "git", "url": "https://github.com/doctrine/instantiator.git", "reference": "8e884e78f9f0eb1329e445619e04456e64d8051d" }, "dist": { "type": "zip", "url": "https://api.github.com/repos/doctrine/instantiator/zipball/8e884e78f9f0eb1329e445619e04456e64d8051d", "reference": "8e884e78f9f0eb1329e445619e04456e64d8051d", "shasum": "" }, "require": { "php": ">=5.3,=5.3.2" }, "type": "library", "extra": { "branch-alias": { "dev-master": "1.0.x-dev" } }, "autoload": { "psr-0": { "Doctrine\\Common\\Lexer\\": "lib/" } }, "notification-url": "https://packagist.org/downloads/", "license": [ "MIT" ], "authors": [ { "name": "Roman Borschel", "email": "roman@code-factory.org" }, { "name": "Guilherme Blanco", "email": "guilhermeblanco@gmail.com" }, { "name": "Johannes Schmitt", "email": "schmittjoh@gmail.com" } ], "description": "Base library for a lexer that can be used in Top-Down, Recursive Descent Parsers.", "homepage": "http://www.doctrine-project.org", "keywords": [ "lexer", "parser" ], "time": "2014-09-09T13:34:57+00:00" }, { "name": "doctrine/orm", "version": "v2.5.6", "source": { "type": "git", "url": "https://github.com/doctrine/doctrine2.git", "reference": "e6c434196c8ef058239aaa0724b4aadb0107940b" }, "dist": { "type": "zip", "url": "https://api.github.com/repos/doctrine/doctrine2/zipball/e6c434196c8ef058239aaa0724b4aadb0107940b", "reference": "e6c434196c8ef058239aaa0724b4aadb0107940b", "shasum": "" }, "require": { "doctrine/cache": "~1.4", "doctrine/collections": "~1.2", "doctrine/common": ">=2.5-dev,=2.5-dev,=5.4", "symfony/console": "~2.5|~3.0" }, "require-dev": { "phpunit/phpunit": "~4.0", "symfony/yaml": "~2.3|~3.0" }, "suggest": { "symfony/yaml": "If you want to use YAML Metadata Mapping Driver" }, "bin": [ "bin/doctrine", "bin/doctrine.php" ], "type": "library", "extra": { "branch-alias": { "dev-master": "2.6.x-dev" } }, "autoload": { "psr-0": { "Doctrine\\ORM\\": "lib/" } }, "notification-url": "https://packagist.org/downloads/", "license": [ "MIT" ], "authors": [ { "name": "Roman Borschel", "email": "roman@code-factory.org" [... cut ...]  "src/" } }, "notification-url": "https://packagist.org/downloads/", "license": [ "BSD-2-Clause" ], "description": "PSR HTTP Message implementations", "homepage": "https://github.com/zendframework/zend-diactoros", "keywords": [ "http", "psr", "psr-7" ], "time": "2017-01-23T04:53:24+00:00" } ], "aliases": [], "minimum-stability": "dev", "stability-flags": { "andig/dbcopy": 20, "react/http": 20 }, "prefer-stable": true, "prefer-lowest": false, "platform": { "php": ">=5.6.0" }, "platform-dev": [] }

BerndGewehr, 2017-02-17:
War leider nur mobil, heute Abend gern formatiert, sorry!

andig, 2017-02-17:
Fang bitte nochmal mit neuem git clone an und beschreib alle Schritte bis dahin. Wie gesagt- bei mir tut es!

BerndGewehr, 2017-02-17:
- Verzeichnis komplett gelöscht
- install.sh neu ausgeführt
- neuer clone, composer läuft durch, setup beendet
- Zur Vorsicht apt-get update, apt-get upgrade
- Fehler beim Start des Pushservers:

```` 
root@volkszaehler:/var/www/volkszaehler.org# /usr/bin/php /var/www/volkszaehler.org/misc/tools/push-server.php
Volkszaehler Push Server
Listening for updates on 5582. Clients may connect at 8082.
PHP Fatal error:  Call to undefined function GuzzleHttp\Psr7\parse_request() in /var/www/volkszaehler.org/vendor/react/http/src/RequestParser.php on line 98
````

andig, 2017-02-17:
Ok, also das passiert wenn sich der erste Client verbindet?

bgewehr, 2017-02-17:
Habe mal composer update ausgeführt. 36 installs und 1 Update wurde ausgeführt. Nun läuft es.

Ich fange noch mal von vorn an und berichte.

BerndGewehr, 2017-02-17:
Kennst Du den Moment, in dem einem auffällt, dass man sich schämen muss?

Ich habe mit einer älteren Kopie der install.sh im home-Verzeichnis gearbeitet. Mit der gab es die Fehler.

Neue install.sh geladen, keine Fehler. Habe mir die commit-history der install.sh angesehen und dafür keinen Grund erkannt, aber das scheint es gewesen zu sein. Tut mir leid!

andig, 2017-02-17:
Hallo Bernd,

Ich hab Dich mehrfach gefragt ob Du wirklich aktuelles repo oder meinen PR verwendest.. jetzt ist die Antwort plötzlich ja.  

Ganz ehrlich: ich habe keine Zeit für Beschäftigungstherapie und bin nicht zu Deiner Bespassung da.

Ich verzichte ab jetzt draus Deine Probleme zu bearbeiten. Du musst Dir einen Anderen suchen- meine Zeit ist mir dafür zu kostbar.

Sorry,
Andreas

frankrichter, 2017-02-19:
Ich hab vorhin auch mal einen neuen Clone angelegt und konnte den Push-Server ebenfalls nicht starten. Fehlermeldung wie bei Bernd:

`PHP Fatal error:  Call to undefined function GuzzleHttp\Psr7\parse_request() in /var/www/volkszaehler.org/vendor/react/http/src/RequestParser.php on line 98`

Vergleich mit älterem Clone ergab, dass im neu erstellten das Verzeichnis guzzlehttp/psr7/ unterhalb von vendor/ gar nicht existiert.

Hab dann noch ein bisschen rumgetestet und festgestellt, dass guzzlehttp/psr7/ wohl seit neuestem nur noch mit `composer install`, aber nicht mit `composer install --no-dev` installiert wird.
Da install.sh auch nur `--no-dev` installiert, sollte das behoben werden.

bgewehr, 2017-02-19:
Das erklärt vielleicht auch, warum es mach composer Update plötzlich ging. Leider bin ich nicht so fit, um diese Zusammenhänge zu kapieren...

frankrichter, 2017-02-19:
Denke ich auch. Die von dir genannten 36 installs/1 update sind genau die
Differenz zwischen require-dev und no-dev.

andig, 2017-02-19:
Was sagt `git log`?

Bei mir:

````
commit 7cccaa90c55e8871dd7aa11172e16770d5255dd8
Author: andig <cpuidle@gmx.de>
Date:   Fri Feb 17 15:02:39 2017 +0100

    Workaround for react/http changes (#579)

frankrichter, 2017-02-19:
Genau das.

andig, 2017-02-19:
Ich kanns immer noch nicht reproduzieren, habe aber eine Idee. Bitte `composer update --no-dev` versuchen.

frankrichter, 2017-02-19:
Hab eben nochmal neu angefangen, alles ok.
Den Stand ohne guzzlehttp kann ich gerade auch nicht reproduzieren. Keine Ahnung was ich da gestern gemacht hab. Könnte höchstens sein dass da eine composer.lock im Spiel war, die irgendwie Ärger gemacht hat.
Mea culpa für die Verwirrung, von mir aus schließen.

bgewehr, 2017-02-20:
Ja, so ging es mir leider auch. Irgendwann ging es dann...

bgewehr, 2017-02-20:
Heute geht es nach einem power-on Neustart übrigens wieder nicht:

````
pi@volkszaehler:~ $ /usr/bin/php /var/www/volkszaehler.org/misc/tools/push-server.php
Volkszaehler Push Server
Listening for updates on 5582. Clients may connect at 8082.
PHP Fatal error:  Call to undefined function GuzzleHttp\Psr7\parse_request() in /var/www/volkszaehler.org/vendor/react/http/src/RequestParser.php on line 98
````

bgewehr, 2017-02-20:
install.sh vom frisch gezogenen repo geht auch nicht mehr:

````
pi@volkszaehler:/var/www/volkszaehler.org/misc/tools $ ./install.sh
volkszaehler.org installation script

checking prerequisites: php: /usr/bin/php
 mysql: /usr/bin/mysql
 awk: /usr/bin/awk
 sed: /bin/sed
 grep: /bin/grep
 wget: /usr/bin/wget
 mktemp: /bin/mktemp
 mkdir: /bin/mkdir
 git: /usr/bin/git

checking php version: 5.6.29-0+deb8u1 >= 5.6, ok

volkszaehler setup...
volkszaehler path? [/var/www/volkszaehler.org]
/var/www/volkszaehler.org already exists. Remove it and get new git clone? (you have to type 'Yes' to do this!) [n]

checking composer...
composer: /var/www/volkszaehler.org/composer.phar

installing dependencies...
/var/www/volkszaehler.org /var/www/volkszaehler.org/misc/tools
Loading composer repositories with package information          Reading composer.json of react/http (0.3)             Could not fetch https://api.github.com/repos/andig/http/contents/composer.json?ref=cf04457ef433cb128fd089b4b71abd4a6085bc7e, please create a GitHub OAuth token to go over the API rate limit
Head to https://github.com/settings/tokens/new?scopes=repo&description=Composer+on+volkszaehler+2017-02-20+0745
to retrieve a token. It will be stored in "/home/pi/.composer/auth.json" for future use by Composer.
Token (hidden):
No token given, aborting.
You can also add it manually latReading composer.json of react/http (0.5)
Could not fetch https://api.github.com/repos/andig/http/contents/composer.json?ref=1fa41f84f183c14fb0c2fa005c911e501bd19fbb, please create a GitHub OAuth token to go over the API rate limit
Head to https://github.com/settings/tokens/new?scopes=repo&description=Composer+on+volkszaehler+2017-02-20+0746
to retrieve a token. It will be stored in "/home/pi/.composer/auth.json" for future use by Composer.
````

andig, 2017-02-20:
> install.sh vom frisch gezogenen repo geht auch nicht mehr:

Da auch das a) **überhaupt** nichts mit dem gemeldeten Fehler zu tun hat und b) die Fehlermeldung im Text ziemlich eindeutig ist aber gelesen und vielleicht gegoogelt werden will gehe ich davon aus dass das Ursprungsproblem gelöst ist.

closed 2017-02-20 by andig

SirkoVZ, 2017-02-21:
ich hab auch Probleme mit dem ppm,
bei mir gibt es keine /var/www/volkszaehler.org/vendor/bin/ppm mehr
die composer.json enthält nicht mehr:

        "php-pm/php-pm": "dev-master",
        "php-pm/httpkernel-adapter": "dev-master"

(aber das(?):
"repositories": [
        {
            "type": "vcs",
            "url": "https://github.com/andig/http"
        }
    ],
)

Hängt das damit zusammen?

ich hab gemacht
- sudo git pull
- composer.phar update

=> kein  /var/www/volkszaehler.org/vendor/bin/ppm mehr da

SirkoVZ, 2017-02-21:
ok, beim 2. Versuch hab ich die ppm wieder, aber:
```
pi@raspberrypi /var/www/volkszaehler.org $ /usr/bin/php /var/www/volkszaehler.org/vendor/bin/ppm start -c /var/www/volkszaehler.org/etc/ppm.json
PHP Fatal error:  Class 'PHPPM\Commands\StartCommand' not found in /var/www/volkszaehler.org/vendor/php-pm/php-pm/bin/ppm on line 27
```

andig, 2017-02-21:
> die composer.json enthält nicht mehr

@SirkoVZ was auch immer Du getan hast- die Installationsanleitung gilt weiterhin ;)

SirkoVZ, 2017-02-21:
ok, nochmal von vorn:
- mein Stand von gestern (letztes git pull vom 3.2)
```
pi@raspberrypi /var/www/volkszaehler.org $ sudo git reset --hard origin/master
HEAD ist jetzt bei ee1023e added EntityDefinition filllevel (#578)
pi@raspberrypi /var/www/volkszaehler.org $ sudo git pull
Already up-to-date.
pi@raspberrypi /var/www/volkszaehler.org $ sudo ./composer.phar update
Do not run Composer as root/super user! See https://getcomposer.org/root for details
Loading composer repositories with package information                                                                                                                                 Updating dependencies (including require-dev)           - Removing php-pm/httpkernel-adapter (dev-master)
  - Removing php-pm/php-pm (dev-master)
  - Removing mkraemer/react-pcntl (v2.0.0)
  - Removing symfony/process (v3.2.1)
  - Removing monolog/monolog (1.22.0)
  - Removing doctrine/collections (v1.3.0)
  - Installing doctrine/collections (v1.4.0)
    Loading from cache

  - Removing doctrine/common (v2.6.2)
  - Installing doctrine/common (v2.7.2)
    Loading from cache

  - Removing doctrine/dbal (v2.5.5)
  - Installing doctrine/dbal (v2.5.12)
    Loading from cache

  - Removing symfony/debug (v3.2.1)
  - Installing symfony/debug (v3.2.4)
    Loading from cache

  - Removing symfony/console (v3.2.1)
  - Installing symfony/console (v3.2.4)
    Loading from cache

  - Removing symfony/http-foundation (v3.2.1)
  - Installing symfony/http-foundation (v3.2.4)
    Loading from cache

  - Removing symfony/event-dispatcher (v3.2.1)
  - Installing symfony/event-dispatcher (v3.2.4)
    Loading from cache

  - Removing symfony/http-kernel (v3.2.1)
  - Installing symfony/http-kernel (v3.2.4)
    Loading from cache

  - Removing symfony/routing (v3.2.1)
  - Installing symfony/routing (v3.2.4)
    Loading from cache

  - Removing react/stream (v0.4.5)
  - Installing react/stream (v0.4.6)
    Loading from cache

  - Removing react/socket (v0.4.4)
  - Installing react/socket (v0.4.6)
    Loading from cache

  - Removing cboden/ratchet (v0.3.5)
  - Installing cboden/ratchet (v0.3.6)
    Loading from cache

  - Removing symfony/yaml (v3.2.1)
  - Installing symfony/yaml (v3.2.4)
    Loading from cache

  - Removing sebastian/object-enumerator (2.0.0)
  - Installing sebastian/object-enumerator (2.0.1)
    Loading from cache

  - Removing sebastian/comparator (1.2.2)
  - Installing sebastian/comparator (1.2.4)
    Loading from cache

  - Removing phpunit/php-code-coverage (4.0.4)
  - Installing phpunit/php-code-coverage (4.0.5)
    Loading from cache

  - Removing myclabs/deep-copy (1.5.5)
  - Installing myclabs/deep-copy (1.6.0)
    Loading from cache

  - Removing phpunit/phpunit (5.7.5)
  - Installing phpunit/phpunit (5.7.14)
    Loading from cache

  - Removing zendframework/zend-diactoros (1.3.8)
  - Installing zendframework/zend-diactoros (1.3.10)
    Loading from cache

  - Removing react/dns (v0.4.3)
  - Installing react/dns (v0.4.4)
    Loading from cache

  - Removing guzzlehttp/psr7 (1.3.1)
  - Installing guzzlehttp/psr7 (1.4.0)
    Loading from cache

  - Installing ringcentral/psr7 (1.2.1)
    Loading from cache

Package guzzle/common is abandoned, you should avoid using it. Use guzzle/guzzle instead.
Package guzzle/stream is abandoned, you should avoid using it. Use guzzle/guzzle instead.
Package guzzle/parser is abandoned, you should avoid using it. Use guzzle/guzzle instead.
Package guzzle/http is abandoned, you should avoid using it. Use guzzle/guzzle instead.
Writing lock file
Generating autoload files

```

Danach ist tatsächlich der SymLink /var/www/volkszaehler.org/vendor/bin/ppm zu 
ppm -> ../php-pm/php-pm/bin/ppm
weg, ebenso wie /var/www/volkszaehler.org/vendor/php-pm komplett weg ist

siehe beim composer update oben:
 - Removing php-pm/php-pm (dev-master)

Das liegt wohl daran, daß im https://github.com/volkszaehler/volkszaehler.org/blob/master/composer.json
die zwei Zeilen weg sind?
        "php-pm/php-pm": "dev-master",
        "php-pm/httpkernel-adapter": "dev-master"

SirkoVZ, 2017-02-21:
verdammt, ich bin im falschen issue, bei mir geht's gar nicht um den push server (der geht), sondern um den        php-pm

andig, 2017-02-21:
Ich weiss ich nerve, aber:

@SirkoVZ (was auch immer Du getan hast) die Installationsanleitung gilt weiterhin. Ums deutlicher zu sagen: https://github.com/volkszaehler/volkszaehler.org/tree/master/misc/tools#installation

SirkoVZ, 2017-02-21:
naaaagut ;-)
`composer require php-pm/php-pm:dev-master php-pm/httpkernel-adapter:dev-master`
war noch nötig.
Danke