volkszaehler/volkszaehler.org #588: Fehler bei unsubscribe in verschachtelten Gruppen

frankrichter, 2017-02-27:
Setup: 
Gebäude/Gruppe/Kanal, push aktiv

Aktion:
Checkbox von Gebäude deaktivieren

Resultat:
Uncaught not subscribed to topic ...

vermutlich wird unsubscribe mehrfach versucht, ich konnte die Ursache aber nicht finden

andig, 2017-02-27:
Quick and dirty- einfach try/catch einbauen?

````
`Entity.prototype.unsubscribe = function() {
	var mw = vz.middleware.find(this.middleware);
	if (mw.session) {
		mw.session.unsubscribe(this.uuid);
	}
};

frankrichter, 2017-02-27:
Von mir aus gerne.

Wahrscheinlich erschlägt das dann auch das eben entdeckte Problem, dass verschachtelte Gruppen nicht per Cookie auf inaktiv gesetzt werden können (Cookie wird nicht gespeichert, wahrscheinlich wegen dem vorher auftretenden Fehler). Ohne push klappt das nämlich.

frankrichter, 2017-02-27:
Hab die Ursache gefunden, doppelte Rekursion in `activate`.
Und ja, die Cookie-Sache geht damit jetzt auch.

PR #589 (2017-03-02): Removed duplicate recursion
Fix #588
closed 2017-03-02 by andig