volkszaehler/volkszaehler.org #590: Vereinheitlichung Anzeigeverhalten Gruppen/Gebäude/Nutzer

frankrichter, 2017-02-27:
aktuell werden nur inaktive Gruppen eingeklappt, für Gebäude und Nutzer gilt das nicht:
```javascript
vz.entities.inheritVisibility = function() {
	vz.entities.each(function(entity, parent) {
		// inherit active state if parent
		if (entity.type !== "group" && entity.parent !== undefined) {
			if (entity.active !== entity.parent.active) {
				entity.activate(entity.parent.active);
			}
		}

		// collapse groups if inactive
		if (entity.type == "group" && entity.active === false) {
			entity.activate(false, entity.parent, true);
			$('#entity-' + entity.uuid + '.aggregator').removeClass('expanded').collapse();
		}
	}, true);
};
```

mein Versuch, das mittels `definition.model` für alle aggregators zu regeln, funktioniert leider nicht:
```javascript
vz.entities.inheritVisibility = function() {
	vz.entities.each(function(entity, parent) {
		// inherit active state if parent
		if (entity.definition.model !== 'Volkszaehler\\Model\\Aggregator' && entity.parent !== undefined) {
			if (entity.active !== entity.parent.active) {
				entity.activate(entity.parent.active);
			}
		}

		// collapse groups if inactive
		if (entity.definition.model == 'Volkszaehler\\Model\\Aggregator' && entity.active === false) {
			entity.activate(false, entity.parent, true);
			$('#entity-' + entity.uuid + '.aggregator').removeClass('expanded').collapse();
		}
	}, true);
};
```

frankrichter, 2017-02-27:
Funktioniert doch, hab mich selbst verarscht...
Ist das was für einen PR?

andig, 2017-02-27:
Her damit! Zur Sicherheit aber prüfen ob entity.definition wirklich gesetzt ist...

frankrichter, 2017-02-27:
Meinst du prüfen durch Test (hab ich) oder prüfen im Code (das kriegst du sicher besser hin)?

closed 2017-02-28 by andig