volkszaehler/volkszaehler.org #607: Update issues

WolfgangFahl, 2017-04-17:
I have updated my Volkszähler installation following the instructions at:

http://wiki.volkszaehler.org/howto/git

The first issue I encountered was a http 500 and an entry in the apache 2 error log:

`Mon Apr 17 07:58:05.353126 2017] [core:alert] [pid 13522] [client 2.0.0.102:56945] /var/www/volkszaehler.org.2017/.htaccess: Require not allowed here`

to fix this I had to add Authconfig to the AllowOverride options in the vz configuration of apache:
```
Alias /vz2017/  "/var/www/volkszaehler.org.2017/htdocs/"
<Directory /var/www/volkszaehler.org.2017>
  Options Indexes FollowSymLinks MultiViews
  AllowOverride FileInfo Limit Options Indexes Authconfig
  Require all granted
</Directory>
```

you might want to add this info to the documentation.

The next issue is a display of "Network Error" - Internal Server Error dialog. If I ignore this I get
Javascript Runtime Error "TypeError: vz.capabilities.definitions.entities is undefined"

I have cleared the vz_ cookie and the cache to no avail.

Please advise

andig, 2017-04-17:
For the internal server error please look into the apache log file. Most likely an incomplete installation of config missing.

andig, 2017-04-17:
@WolfgangFahl could you add the missing AuthConfig to the wiki?

J-A-U, 2017-04-17:
@WolfgangFahl  Ich bin der Meinung das du die Änderung an der falschen Stelle ins Wiki gepackt hast.
Die Seite die du da editiert hast bezieht sich ganz konkret auf den Umgang mit git, Hinweise zu Apache sind da unpassend.

Mir ist die Ursache für das Problem nicht ganz klar (mit Webserverkonfig kenn ich mich nicht aus), scheint mir aber eins zu sein das wir auch bei Neuinstallationen haben. Würde dann hier hingehören: http://wiki.volkszaehler.org/software/middleware/installation

J-A-U, 2017-04-21:
Nachdem ich mich jetzt eingelesen habe bin ich mir sicher das es ein grundsätzliches Problem ist das sowohl Apache [2.4](https://httpd.apache.org/docs/2.4/howto/auth.html#theprerequisites) als auch [2.2](https://httpd.apache.org/docs/2.2/howto/auth.html#theprerequisites) betrifft, in der Installationsanleitung für die MW aber nicht behandelt wird.

Daher die Installationsanleitung korrigiert (den Teil zu Debian Jessie muss ich noch prüfen) und aus dem git Howto entfernt.

andig, 2017-04-21:
Yup, scheint mir auch so.

andig, 2017-04-22:
Ich habe noch ein Problem mit der Apache Conf gefunden: falls /middleware/... funkionieren soll wird auch

    AddType application/x-httpd-php .php

In den MIME Types benötigt, siehe https://serverfault.com/questions/372733/apache-file-negotiation-failed

J-A-U, 2017-04-22:
Hm, ok.

Je genauer ich mir die bestehende Anleitung im Wiki anschau umso falscher kommt sie mir vor.  Gerade auch im Bezug auf die sehr knappe Anpassung unter "Jessie": `<Directory />` ist bei mir aufm Rpi gar nicht in Konfiguriert.

andig, 2017-04-23:
Ich mache mal wieder auf. Mir ist auch aufgefallen dass die RewriteRules nicht vernünftig funktionieren wenn die `mod_negotiation` geladen ist. Wir brauchen also noch ein `Options -MultiViews`

TobiasScheinert, 2017-07-30:
Hi,
ich bin mir nicht ganz sicher, ob das hier rein gehört. Ich hatte meinen Apache 2.4.10 (Raspbian) folgendermaßen eingerichtet (ich habe `+MultiViews` und `-MultiViews` ausprobiert):

```
DocumentRoot /srv/www/empty
Alias /volkszaehler "/srv/www/volkszaehler/htdocs"
<Directory "/srv/www/volkszaehler/htdocs/">
  Options +Indexes +FollowSymLinks -MultiViews
  AllowOverride FileInfo Limit Options Indexes Authconfig
  Require all granted
</Directory>
```
An und für sich hat alles einwandfrei funktioniert, sprich das Frontend hat funktioniert, und ich konnte auch problemlos per vzlogger Daten importieren. Allerdings schreibt mir dann der Apache bei jedem Zugriff ` AH01630: client denied by server configuration: /srv/www/empty/data` ins Log.

Wenn ich den `DocumentRoot` auf `/srv/www/volkszaehler/htdocs` setze verschwindet die Fehlermeldung. Ich vermute mal, dass der Apache die Art und Weise, wie der `middleware.php` die Parameter übergeben werden, nicht gut findet. Vielleicht weiß ja von euch jemand Rat.

Gruß Tobias

andig, 2017-07-31:
@TobiasScheinert das scheint mir eine sehr individuelle Apache Config zu sein. Hat nix mit dem Standard VZ Setup zu tun? Für Diskussionen bitte die ML benutzen...

christian-weiss, 2017-12-21:
This ticket was not updated for some month. Problem seams to be resolved or discussed in mailing list.
Please close this ticket by now.

J-A-U, 2017-12-21:
> Problem seams to be resolved or discussed in mailing list.

No and no.

christian-weiss, 2017-12-21:
@WolfgangFahl Do you still have that problem? If yes, please update to the latest "release" following the wiki. If the issue still remains please provide your log files (apache and php).

J-A-U, 2017-12-21:
I'm pretty sure either WolfgangFahl or TobiasScheinert don't have a problem running their vz. 

But, we have a generic issue with our apache configuration in terms of wiki documentation and the delivered rpi-image.

WolfgangFahl, 2018-04-25:
I have not tried to update again but I'am trying to do so soon.

WolfgangFahl, 2018-11-04:
Heute habe ich einen neuen Versuch gemacht.
`git pull
composer update`

Dann wieder probiert .
Erste Meldung "Network error - Internal Server error"

WolfgangFahl, 2018-11-04:
I am getting frustrated. If the software would give a hint where to look that would be a major improvement. The apache log files are empty. Where would I find the relevant php files?
Which configuration files need modifications except apache?
Does the database need an upgrade?

WolfgangFahl, 2018-11-04:
Now checking installation steps:
`sudo apt-get install git-core libapache2-mod-php7.0 php7.0-cli php7.0-mysql php-apcu mysql-server mysql-client php7.0-xml php-mbstring ca-certificates apache2
`
added php-apcu-bc and php-apcu

same error message ...

WolfgangFahl, 2018-11-04:
Ok the solution is simple I feel a bit stupid but would recommend to update the documentation.
Of course volkszaehler.conf.php needs to be merged from the old installation ...
If the software would detect a missing volkszaehler.conf.php that would be a major improvement. E.g. mediawiki automatically goes into installation mode in this case. For an update it would be helpful to ask for the old configuration file and copy the relevant infos. It would also be helpful to split the configuration file into a part with the credentials and a part with the other configurations (that are more volatile). Also the configuration should not be in the same directory as the stuff that is handled by git - a pull request would accedentialy give passwords and the like to the git repository ...

andig, 2018-11-04:
> I am getting frustrated. If the software would give a hint where to look that would be a major improvement.

Wie ich vor ganz langer Zeit schrieb:

> For the internal server error please look into the apache log file. Most likely an incomplete installation of config missing.

Das Wiki kannst Du selbst uodaten!

andig, 2018-11-04:
> Also the configuration should not be in the same directory as the stuff that is handled by git - a pull request would accedentialy give passwords and the like to the git repository ...

Die Dateinamen sind unterschiedlich. In 8 Jahren VZ war das kein Problem.

J-A-U, 2018-11-04:
Um es ein wenig zu erläutern: Die Beispielkonfig wird bei Erstinstallation einmalig kopiert und fertig. Die lokale Konfigurationsdatei, mit den sicherheitsrelevanten Daten, ist fürs git explizit ausgeblendet. Sie wird daher nicht hochgeladen. 

Falls du eine Möglichkeit siehst ein Script zu schreiben das alte und neue Konfigurationen ineinanderwürfelt sind wir für deine Unterstützung dankbar.

andig, 2019-03-22:
Ich denke das Thema ist erledigt. Das VZ 2.0 Image wird vstl. ohne Apache funktionieren.

closed 2019-03-22 by andig

WolfgangFahl, 2020-10-02:
The problem reappears in 2020 after upgrading to php 7.4. Again i need to investigate the configuation issues instead of getting hints by the software itself.

andig, 2020-10-03:
Mir ist nicht bekannt, dass wir ein neues Image mit php 7.4 haben? Für Änderungen Deiner lokalen Konfiguration bist Du natürlich selbst verantwortlich. Wenn es hier ein konkretes Problem gibt gerne neues Issue aufmachen.

WolfgangFahl, 2020-10-04:
PHP 7.4 ist auf der Serverseite und da tauchen erneut irreführende Fehlermeldung auf:
[image]
für die Software aus 2018 die bisher lief  und konfiguriert ist.
und
[image]
für die Software mit aktuellem stand, die nicht konfiguriert ist. Mein Problem ist nun wie ich von diesen beiden Meldungen aus zu einem funktionierenden System komme.

andig, 2020-10-04:
Wie gesagt. Mach gerne ein neues issue auf. Einschl. Beschreibung was Du ggü funktionierendem System geändert hast, Apache Log und Blick in die Browserkonsole.