volkszaehler/volkszaehler.org #612: Frontend: push data handling

frankrichter, 2017-04-26:
Verbesserungsvorschläge für Umgang mit Push-Daten im Frontend:
- [ ] Zoom/Navigation ohne Verlust der Live-Daten
Idee: Push-Daten werden in `entity.pushData.tuples` gespeichert und überleben so die Neubelegung von entity.data bei MW-Request. Für den Plot werden die push tuples in `entitiy.data.tuples` intgeriert, allerdings anders als aktuell mit Priorität für Push-Daten.

- [ ] Auto-Reload von Kanälen ohne Push-Updates ohne Verlust der Live-Daten anderer Kanäle
Idee: selektiver Reload oder analog zu oben nach dem Reload MW-Tuples durch gespeicherte Push-Tuples ersetzen.

- [ ] Aktivierung/Deaktivierung des Chart-Updates per UI für Fälle, in denen die ständige Aktualisierung hinderlich ist (vgl. #431)
Idee: Navigations-Button "scroll to present" wirkt  im Fall von `(vz.wui.tmaxnow)` beim ersten Klick als Pause-Knopf, beim zweiten Klick scrollt er wie üblich zur aktuellen Zeit und aktiviert die Aktualisierung des Charts wieder. Schick wäre Visualisierung mit Play/Pause-Symbolen.

andig, 2017-04-26:
Die ersten beiden Punke erscheinen adhoc relativ aufwendig. Der Dritte gefällt mir sehr gut. Siehst Du da irgendwelche Nebenwirkungen? Scheint auf Anhieb fast trivial?

frankrichter, 2017-04-26:
> Die ersten beiden Punke erscheinen adhoc relativ aufwendig.

Ist klar dass das einiges an Arbeit ist. Sind ja auch nur Vorschläge.
Die Usability würde sicher profitieren, wenn die Live-Daten nicht bei der ersten Interaktion weg wären, momentan gilt ja "am besten nicht anfassen", wenn man die hohe Auflösung erhalten möchte.

> Der Dritte gefällt mir sehr gut. Siehst Du da irgendwelche Nebenwirkungen? Scheint auf Anhieb fast trivial?

Bei aktivem push ist der Button weitgehend nutzlos, wenn man sich am rechten Rand befindet. Wer kein push nutzt, hat mit dem Button zur Zeit eine Möglichkeit, einen sofortigen Reload auszulösen. Das wäre dann einen Klick weiter weg - vielleicht auch nicht ideal.

andig, 2018-02-08:
> Das wäre dann einen Klick weiter weg - vielleicht auch nicht ideal.

Wäre die Frage. Alternativ könnte man ja day/plus/minus oder Ähnliches drücken um einen Reload auszulösen und wieder in den "live" Modus zu kommen. Richtig sexy finde ich's aber auch nicht.

Nebenbei frage ich mich noch ob es hilfreich wäre wenn die aktuelle Zoomstufe (day, month, year etc) auch optisch am Buttonstatus zu erkennen wäre?

frankrichter, 2018-02-08:
Also doch lieber ein zusätzlicher, dedizierter Play/Pause-Button? Dafür könnten wir "Automatisch aktualisieren" aus dem Optionen-Menü kicken. Das ist dort eh sehr weit weg vom Plot und nicht besonders intuitiv zu bedienen.

> Nebenbei frage ich mich noch ob es hilfreich wäre wenn die aktuelle Zoomstufe (day, month, year etc) auch optisch am Buttonstatus zu erkennen wäre?

Absolut. Erstmal quick and dirty, z.B. mit eckigen Klammern um den Button-Text?

andig, 2018-02-08:
Idee: wenn ich rechts bin ist der button play/pause. In dem Fall gibts ja noch den anderen „rechts“ button der egtl. nutzlos ist. 
Wenn ich nicht rechts bin ist der button forward (wie jetzt).
Kein Zusatzknopf. Status über Schattierung?

frankrichter, 2018-02-08:
Das ist ja im Prinzip mein Vorschlag von ganz oben. Klingt praktikabel.
Schattierung ist natürlich auch gut.