volkszaehler/volkszaehler.org #614: Aggregation mit SensorInterpreter zeigt falsche Werte

fibroin, 2017-05-02:
Hallo zusammen,
bei dem Auslesen von Daten für bestimmte Perioden(tag/monat) über die Middleware ist mir aufgefallen, dass Werte aus dem Sensor-Interpreter falsch ausgegeben werden, wenn für einen Zeitraum nicht regelmäßig Daten vorhanden sind. z.B. eine PV-Anlage, die Nachts keine Werte loggt.
Ich nutze MySql und auch die Aggregation-Tabelle.
Beim täglichen Aggegieren wurde bereits falsch gerechnet und die Werte stimmten nicht.
Ich habe festgestellt, dass bei der Berechnung der Zeitraum nicht einheitlich ist, sondern immer bis zum vorherigen Messwert.
Berechne ich zum Beispiel die Stundenwerte und habe bei meiner PV-Anlage am morgen oder abend nur 45 minutenwerte, wird die summe auf 45 minuten bzw. bis zum letzten Vortagswert gemittelt. Ich muss aber davon ausgehen, dass der Wert für 60 min. gilt.
Ich habe daher folgende Änderung gemacht:

lib/Util/Aggregation.php Zeile 209:
```
- 'SUM(agg.val_by_time) / (MAX(agg.timestamp) - MIN(agg.prev_timestamp)), ' .
+ 'SUM(agg.val_by_time) / (MAX(agg.timestamp) - UNIX_TIMESTAMP(DATE_SUB(FROM_UNIXTIME(MAX(agg.timestamp)/1000, "%Y-%m-%d %H:%i:%s"),INTERVAL 1 ' . $level . '))*1000), ' .
```
Das funktioniert nun wunderbar bei der Aggegation.

Werte, die nicht aus der Aggregation-Tabelle kommen, stimmen ebenfalls nicht.
Hier habe ich noch keine Änderungen vorgenommen, aber im Prinzip müsste es in die selbe Richtung gehen wie oben.
Ich habe gesehen, es gibt da noch einen unterschied im Optimizer mit und ohne Aggregation, bin aber noch nicht genau durchgestiegen, wo die welche Anpassungen nun hin müssen.

Gruß
Andreas

frankrichter, 2017-05-02:
Hallo Andreas,

stell mal bitte Beispieldaten zur Verfügung (CSV-Export für einen Tag, einmal normal und einmal mit options=raw).

Grüße
Frank

andig, 2017-05-02:
> Beim täglichen Aggegieren wurde bereits falsch gerechnet und die Werte stimmten nicht.

Bitte konkretes Beispiel, gerne auch Testcase (siehe test Ordner).

> Ich habe festgestellt, dass bei der Berechnung der Zeitraum nicht einheitlich ist, sondern immer bis zum vorherigen Messwert.

Ja, zwangsläufig, Anderenfalls würde zum Anfang des Intervalls Zeit fehlen. 

Ganz banal: woher soll die Middleware wissen dass da keine Meßwerte hingehören? Ab wann soll sie annehmen dass das so ist? Alle Meßwerte werden gleichberechtigt behandelt. Wenn Du Lücken brauchst dann sollten da NULLEN geschrieben werden.

Soweit WFM.

fibroin, 2017-06-16:
Hallo,
es hat leider etwas länger gedauert. Mir fehlte es etwas an Zeit.
Ich musste noch genauer schauen, was das Problem ist.

Angehängt habe ich 2 CSV-Files. Die erste normal, die zweite mit options=raw
[pv string 1 pdc-normal.csv.txt](https://github.com/volkszaehler/volkszaehler.org/files/1081730/pv.string.1.pdc-normal.csv.txt)
[pv string 1 pdc-raw.csv.txt](https://github.com/volkszaehler/volkszaehler.org/files/1081731/pv.string.1.pdc-raw.csv.txt)

Erstmal vorab: Warum habe ich nachts keine Nullen mehr geschrieben? Weil ich der Ansicht bin, dass nullen keine Daten sind, und nur Speicher unnötig füllen. Dass das nun Probleme mit sich bringt, habe ich nicht erwartet.
Dennoch bin ich der Meinung, dass eine gute Software mit allen Fällen umgehen können sollte.

Dass es Abweichungen von den tatsächlichen Werten gab, habe ich festgestellt, als ich mir eine eigene Anzeige programmieren wollte, die mir Stunden-/Tageswerte aus einem gewünschten Zeitraum als Diagramm darstellt. Die angezeigten Werte habe ich dann mit korrekten Werten verglichen und Abweichungen die gefunden.

Mein erster Lösungsversuch aus dem ersten Post hat aber doch nicht funktioniert.

Folgendes habe ich nun festgestellt: Bei der gewichteten Ermittlung von Daten wird in der Agg-Zwischentabelle der allererste prev_timestamp = NULL gesetzt (ist ja nicht vorhanden).
Das führt dann aber dazu, dass bei der Berechnung für den ersten Zeitraum(mit GROUP BY) der MIN(agg.prev_timestamp) zu spät ist, nämlich vom 2. Wert. In meiner CSV wäre das dann der 13.6. 5:00. Es muss aber eigentlich der vorhergehende sein, also 12.6. 22:02. Dadurch ist der Zeitraum zu klein und der resultierende Wert zu groß.
Beim täglichen Aggregieren resultiert daraus, dass der erste Stundenwert zu groß ist, und beim Tageswert(da nur einer ermittelt wird) alle Werte zu groß sind.

Meine erste Lösung zielte darauf ab, den Zeitraum immer auf den gewünschten (zb. Stunde) zu setzen. Da das Frontend aber mit den tatsächlichen Timestamps die Werte interpretiert, funktioniert das nicht.

Ich habe jetzt den ersten prev_timestamp mit dem nötigen Wert gefüllt, und nun werden die Werte von der Middleware korrekt ausgegeben.
In der `Util\Aggregation.php` habe ich nach 

> `if ($weighed_avg) {`

folgenden code eingefügt 

> 1. `if ($mode == 'delta' and $channel_id) {`
> 2. `     $prev_sql = '@prev_timestamp := MAX(timestamp) FROM aggregate WHERE type = ' . $type . ' AND channel_id = ' . $channel_id;`
> 3. `} else {`
> 4. `     $prev_sql = '@prev_timestamp := NULL';`
> 5. `}`

und die Zeile mit dem CROSS JOIN geändert:

> `'CROSS JOIN (SELECT '. $prev_sql .') AS vars ' .`

Auch mit den Werten, die zeitlich nach der Aggregation-Tabelle liegen, habe ich etwas geändert.
Dort fehlt in der `MySQLAggregateOptimizer.php` die gewichtete Ermittlung für Perioden.
Auch hier muss das System von oben verwendet werden, zumindest für den Zeitraum nach der Aggregation. Die Abfragen für die Zeiträume vor und nach der Aggregationstabelle müssen getrennt werden, sonst ist der erste prev_timestamps für den Zeitraum nachher falsch.
Hier ist meine Version dieser Datei. Geändert ist nur die Funktion optimizeDataSQL.

[MySQLAggregateOptimizer.php.txt](https://github.com/volkszaehler/volkszaehler.org/files/1081811/MySQLAggregateOptimizer.php.txt)

Der Code ist jetzt zwar noch nicht schön geschrieben, aber er funktioniert erstmal. Das ist erstmal das wichtigste.
Von meiner Seite war das erstmal alles.
Wenn noch Fragen dazu da sind, ich versuche diesmal schneller zu antworten :-)

Ich würde mich freuen, wenn die Anpassungen übernommen würden. So können alle davon profitieren.

Gruß
Andreas

andig, 2017-06-17:
> Ich habe festgestellt, dass bei der Berechnung der Zeitraum nicht einheitlich ist, sondern immer bis zum vorherigen Messwert.

Korrekt, das ist bisher by design. Dahinter steckt die Annahme, dass es "viele" Meßwerte gibt, d.h. deutlich mehr als zu betrachtende Intervalle. Das gilt auch für "normale" Abfragen ohne Aggregation. 

Was wir dabei explizit nicht machen ist fiktive Meßwerte an Periodengrenzen zu interpolieren. Das wäre aufwändig und auch immer von der Darstellungsform (lines, steps etc) abhängig. 

Das betrifft aber jegliche Abfragen/Aggregation, ob Sensor oder nicht. Bevor ich mich jetzt durch alle Werte und SQLs wühle- könntest Du bitte nochmal beschreiben wann Dein Problem auftritt und wann nicht (normal vs aggregation, sensor vs sonstige, jedes intervall oder nur fehlende nullen)?

> Erstmal vorab: Warum habe ich nachts keine Nullen mehr geschrieben? Weil ich der Ansicht bin, dass nullen keine Daten sind, und nur Speicher unnötig füllen. Dass das nun Probleme mit sich bringt, [...]

Bitte konkreter. Die SW geht mit dem Fall wie oben beschrieben um. Prev_ts ist immer der lette vorherige- unabhängig davon wie alt dieser ist. Ob Nullen oder nicht ist dabei irrelevant, wichtig ist zu verstehen dass die Middleware keine Daten erfindet. Wenn es keine gibt werden daher auch keine Nullen "erfunden" sondern es ist schlicht kein neuer Messwert vorhanden. Die Middlewäre rätselt nicht!

andig, 2017-06-17:
Long Story short: ich habe noch nicht verstanden welches Problem Du konkret lösen willst. Wäre schön das verständlicher zu bekommen.

@frankrichter vielleicht bist Du ja schon weiter als ich...

frankrichter, 2017-06-17:
Ich steige da leider auch noch nicht durch. Mir fehlen Beispieldaten, die die angeblich fehlerhafte Aggregation belegen.

andig, 2017-06-19:
> Ich steige da leider auch noch nicht durch. Mir fehlen Beispieldaten, die die angeblich fehlerhafte Aggregation belegen.

Also fangen wir nochmal vorne an

> bei dem Auslesen von Daten für bestimmte Perioden(tag/monat) über die Middleware ist mir aufgefallen, dass Werte aus dem Sensor-Interpreter falsch ausgegeben werden, wenn für einen Zeitraum nicht regelmäßig Daten vorhanden sind. z.B. eine PV-Anlage, die Nachts keine Werte loggt.
Ich nutze MySql und auch die Aggregation-Tabelle.

Was liest Du wie aus? Abfrage?

> Beim täglichen Aggegieren wurde bereits falsch gerechnet und die Werte stimmten nicht.

Welche Werte stimmen nicht- in der Aggregationstabelle? Auszug der Rowwerte und der Tabelle?

> Folgendes habe ich nun festgestellt: Bei der gewichteten Ermittlung von Daten wird in der Agg-Zwischentabelle der allererste prev_timestamp = NULL gesetzt (ist ja nicht vorhanden).

In der aggregate Tabelle gibt es keine Spalte `prev_timestamp`. Meinst Du eine SQL Abfrage? Bei Befüllung der Aggregate Tabelle oder der Abfrage der Werte?

> Das führt dann aber dazu, dass bei der Berechnung für den ersten Zeitraum(mit GROUP BY) der MIN(agg.prev_timestamp) zu spät ist, nämlich vom 2. Wert. In meiner CSV wäre das dann der 13.6. 5:00. Es [...]
Beim täglichen Aggregieren resultiert daraus, dass der erste Stundenwert zu groß ist, und beim Tageswert(da nur einer ermittelt wird) alle Werte zu groß sind.

@fibroin Unklar. Wir würden gerne helfen bzw Fehler korrigieren, brauchen aber abgesichts der komplexen Materie eine detaillierte, nachvollziehbare Fehlerbeschreibung bevor es an die Bewertung von Lösungen geht. 

Bisher fehlt in dieser Hinsicht leider alles :(

andig, 2017-06-26:
ping @fibroin

fibroin, 2017-06-26:
So, ich versuche nochmal etwas klarer zu werden:

> könntest Du bitte nochmal beschreiben wann Dein Problem auftritt und wann nicht (normal vs aggregation, sensor vs sonstige, jedes intervall oder nur fehlende nullen)?

Das problem tritt auf 
1) bei der Aggregation(schreiben der Aggregationstabelle) beim Sensor in Zeiträumen mit fehlenden Nullen
2) Beim Abruf von Daten, die noch nicht in der Aggregationstabelle sind. Abruf mit group.
Ich würde vorschlagen, zuerst einmal Fall 1 zu behandeln. Dann wir Fall 2 auch schnell klar.

Es tritt nicht auf bei Accumulator. Impulse habe ich nicht im Einsatz.
Bei einer Middleware-Abfrage ohne group tritt das Problem auch nicht auf.

> Was liest Du wie aus? Abfrage?

Ich rufe die Middleware auf z.B. mit:
http://localhost/vz/middleware.php/data/30252f70-db8f-11e3-86a2-53bbdf0a5161.json?from=01-01-2017&to=09-04-2017&group=day
Group mit day und hour

> Welche Werte stimmen nicht- in der Aggregationstabelle? Auszug der Rowwerte und der Tabelle?

In der Aggregations-Tabelle stimmen Werte nicht.
Immer der erste Wert von jedem Level (hour, day) pro Ermittlung stimmt nicht. 
Ich habe noch einmal die Tabelle von oben(=data-Tabelle) angehängt und für den ersten Tag die aggregierten Werte in day und hour ergänzt.
In rot markiert sind die falsch berechneten Werte. In grün die korrekten Werte.
Ist so wahrscheinlich anschaulicher und nachzuvollziehen.

[pv string 1 pdc-normal.xlsx](https://github.com/volkszaehler/volkszaehler.org/files/1103409/pv.string.1.pdc-normal.xlsx)

Der Unterschied zwischen der falschen und der richtigen Berechnung ist der, dass bei der falschen beim min/max für den Zeitraum die erste Zeile "fehlt". Bitte auch die Formeln vergleichen.

Unter Frontend seht ihr einmal den Tageswert und darunter die Summe der Stundenwerte. Bei der richtig-Spalte sind die Werte identisch.

> In der aggregate Tabelle gibt es keine Spalte prev_timestamp. Meinst Du eine SQL Abfrage? Bei Befüllung der Aggregate Tabelle oder der Abfrage der Werte?

Sorry, ja ich meine die SQL-Abfrage in der "lib\Util\Aggregation.php" in "function aggregateChannel"

> Ob Nullen oder nicht ist dabei irrelevant, wichtig ist zu verstehen dass die Middleware keine Daten erfindet. Wenn es keine gibt werden daher auch keine Nullen "erfunden" sondern es ist schlicht kein neuer Messwert vorhanden. Die Middlewäre rätselt nicht!

Erfunden wird hier natürlich nichts. Es werden nur aus den vorhandenen Daten die Mittelwerte gebildet.
Der Wert um 5:59 von 0,002 kWh muss korrekterweise als Gesamtwert für die ganze Nacht angesehen werden. Dennoch kann ich hier sagen, dass der Wert nur von 5:00 bis 5:59 gilt, denn sonst wäre um 4:59 ja ein Wert gespeichert worden. Da dieser fehlt, muss vom letzen Wert am Vortag bis 4:59 immer 0 gewesen sein, auch wenn keine Nullen in der Datenbank geschrieben sind.

Gruß

andig, 2017-06-27:
> Das problem tritt auf
> 1. bei der Aggregation(schreiben der Aggregationstabelle) beim Sensor in Zeiträumen mit fehlenden Nullen
> 2. Beim Abruf von Daten, die noch nicht in der Aggregationstabelle sind. Abruf mit group.

Ich denke die Fälle sind ohnehin identisch. Die Ermittlung zum Schreiben der Werte bei 1) wird gem. 2) durchgeführt.

> Der Unterschied zwischen der falschen und der richtigen Berechnung ist der, dass bei der falschen beim min/max für den Zeitraum die erste Zeile "fehlt". Bitte auch die Formeln vergleichen.

Ich glaube ich verstehe grob worauf Du hinaus willst, kann es aber im Code nicht nachvollziehen:

	'SELECT channel_id, ? AS type, ' .
		'MAX(agg.timestamp) AS timestamp, ' .
		'COALESCE( ' .
			'SUM(agg.val_by_time) / (MAX(agg.timestamp) - MIN(agg.prev_timestamp)), ' .
			$aggregationFunction .
		') AS value, ' .
		'COUNT(agg.value) AS count ' .
	'FROM ( ' .
		'SELECT channel_id, timestamp, value, ' .
			'value * (timestamp - @prev_timestamp) AS val_by_time, ' .
			'COALESCE(@prev_timestamp, 0) AS prev_timestamp, ' .
			'@prev_timestamp := timestamp ' .
		'FROM data ' .
		'CROSS JOIN (SELECT @prev_timestamp := NULL) AS vars ' .
		'WHERE ';

Der allererste `prev_timestamp` ist immer 0. Bei `max()-min()` wird die Periode also eher zu groß also zu klein und das Ergebnis nach Division müsste zu klein werden. Dein Excel zeigt aber dass der Wert zu groß ist.

In Deiner Tabelle finde ich noch nicht die "echten" Daten aus der Aggregationstabelle für das jeweilige Level. Könntest Du die- zusätzlich zu den Formeln- noch ergänzen?

andig, 2017-06-27:
Mhhm. Mein Code oben stammt aus dem "next" branch. Damit sehe ich nicht den von Dir beschriebenen Fehler (bei mir müsste der Wert zu klein werden, bei Dir wird er zu groß). Deshalb erschien mir Deine Beschreibung nicht plausibel.

Kannst Du im Code oben (aggregate.php) mal bitte schauen was passiert wenn Du:

    GREATEST(0, IF(@prev_timestamp = NULL, NULL, @prev_timestamp)) AS prev_timestamp

durch die obige Version ersetzt:

    COALESCE(@prev_timestamp, 0) AS prev_timestamp

oder stattdessen meinen "next" Branch nimmst? 

Dann müsste der erste Wert je Periode bei DELTA Aggregation weiterhin falsch sein, aber diesmal (viel) zu klein. Kannst Du das reproduzieren?

Falls ja verstehe ich auch Deinen Fix. Bevor ich das einbaue muss ich mir allerdings Gedanken über einen guten Testfall machen da die Komplexität in dem Bereich zu hoch für spontane Änderungen ist.

fibroin, 2017-06-29:
Hallo,
ich verwende den master-Branch.

> In Deiner Tabelle finde ich noch nicht die "echten" Daten aus der Aggregationstabelle für das jeweilige Level. Könntest Du die- zusätzlich zu den Formeln- noch ergänzen?

Die Werte aus der Aggregationstabelle sind die gleichen, die unter "Wert in Agg-Tabelle" in meiner Tabelle stehen. (bis auf minimale rundungsdifferenzen). Die Formeln sind gleich mit denen aus Middleware & Frontend, außer dass ich hier mit Excel-Datum rechne anstatt mit Timestamp.

> Kannst Du im Code oben (aggregate.php) mal bitte schauen was passiert wenn Du: `COALESCE(@prev_timestamp, 0) AS prev_timestamp`
Dann müsste der erste Wert je Periode bei DELTA Aggregation weiterhin falsch sein, aber diesmal (viel) zu klein. Kannst Du das reproduzieren?

Ja, kann ich reproduzieren. Für den Tag bekomme ich dann einen Wert von 0.02894.... 
Umgerechnet wären das dann 0,695 Wh für diesen Tag. Also wie du schon sagtest, wesentlich kleiner. In Excel komme ich aufs gleiche Ergebnis.

> Der allererste prev_timestamp ist immer 0. 

Genau. Im master-branch ist der erste prev_timestamp immer NULL. MIN() beachtet NULL nicht, also wird der prev_timestamp der zweiten Zeile(=Timestamp der ersten Zeile) genommen. Im next-Branch wird 0 für min() genommen.
Das Hauptproblem ist, dass bei der Aggregation der Zeitraum anders ist, als beim Auslesen aus der Datenbank.
Eigentlich ist der Wert sogar falsch, wenn nachts nullen geschrieben wurden, da ja der erste Wert der neuen Periode als min genommen wird und nicht wie beim Interpretieren im Frontend der letzte der Vorperiode. Hier ist der Fehler natürlich wesentlich kleiner, aber trotzdem vorhanden.

Meine Idee bei dem Fix ist, dass der letzte Aggregations-Timestamp aus der Tabelle in MIN() landet, der gleiche, der auch beim Frontend verwendet wird.

Wenn man die Aggregation komplett abschaltet, wird der Fehler nicht auftreten, da ja bei der Interpretation die erste Zeile entfernt wird. Einzig wird der Timestamp als MIN verwendet. Hier wird im Grunde schon das gleiche gemacht, wie bei meinem Fix.

andig, 2017-06-30:
> Ja, kann ich reproduzieren

Gut. Ich baue Deinen Fix mit ein, muss aber auch noch ein paar andere Punkte mit aufräumen.

andig, 2017-07-01:
Bitte mal testen- nebenbei ist die Aggregation jetzt auch dramatisch schneller geworden da ich einen MySQL Bug gleich mit ge-work-arounded habe :)

closed 2017-07-04 by andig

andig, 2017-07-07:
See https://github.com/volkszaehler/volkszaehler.org/pull/621 for a better option to rebuild the aggregation table than using `clear`.