volkszaehler/volkszaehler.org #623: UI Bug in der Begrenzten übersicht

milefork, 2017-07-18:
Moin,

hab seit längerem mal wieder nen pull gemacht.
Was mir auffällt bei der für mich neuen Ansicht, wenn man jetzt auf Tag drückt geht das ganze schön von 00:00 bis zur Aktuellen Zeit.
Dabei kommt dann der Fehler wenn sich die Ansicht automatisch Aktualisiert, dann läuft der Ausschnitt z.B auf 00:08 und so weiter also ein "sliding window" ist das gewünscht? Ich fände es nicht gut.

Ach und die Zoom Stufen verstehe ich nicht bzw. finde ich jetzt nicht mehr schön. Sind die logarithmisch geworden?

Gruß

andig, 2017-07-18:
Diskussionen und Fragen bitte auf der Mailingliste- konkrete Fehlermeldungen hier. Was den Zoom angeht: einfach 2x auf den Knopf drücken.

> Ach und die Zoom Stufen verstehe ich nicht bzw. finde ich jetzt nicht mehr schön. Sind die logarithmisch geworden?

Das verstehe ich jetzt nicht ;)

milefork, 2017-07-18:
Ja es ist ja ein Fehler. Pass auf Drück auf Tag in deinem Volkszähler dann hast du ja eine Ansicht von 00:00 bis zur jetzigen Uhrzeit. Wenn du jetzt auf den Autorefresh wartest hast du eine Ansicht von 00:08 bis zur jetzigen Uhrzeit.
Also was ich damit meine ist du Hast nicht mehr den kompletten tag sondern, es fehlen dir dann -X Minuten am Anfang und das ist definitiv ein Fehler.

Gruß

frankrichter, 2017-07-18:
Das Frontend zeigt immer ein konstantes Intervall an, damit ist klar dass sich bei einem Update die linke Grenze ebenfalls verschiebt. Ein Fehler wäre es dann, wenn es anders konzipiert wäre - ist es aber nicht.

andig, 2017-07-18:
Ping @frankrichter hast du evtl ein wenig Zeit das anzuschauen?

frankrichter, 2017-07-18:
Kann ich machen. Die Frage ist nur, wo wollen wir hin? Variabler Zoom, d.h. Änderung der Intervalllänge bei jedem Auto-Update (und weiter gedacht bei jedem Push-Update)? Könnte mir vorstellen dass das nicht sehr sexy aussieht.

milefork, 2017-07-20:
Könnte man nicht "stati" setzen, welche sich merken ob man vorher Stunde/Tag/Monat/Jahr ausgewählt hat und bei diesen dann die Intervalllänge bei Updates anpassen.
Und wenn man dann wieder auf den >> oder >| Button drückt die "stati" wieder vergessen werden und man ein konstantes Intervall hat?

frankrichter, 2017-07-20:
Damit würde ja aber genau das passieren was ich oben angedeutet habe: Bei jedem Update (bei "Stunde" dauert das nur wenige Sekunden, bei Push-Betrieb noch schneller) wird der Chart mitsamt x-Achse und Gitternetz etwas zusammengestaucht. Ich hab die Befürchtung dass das weder optisch besonders schön noch besonders intuitiv ablesbar wird.
Und was passiert wenn die gewählte Periode voll ist? Weiter auszoomen, oder Sprung zur nächsten Periode und wieder bei Intervalllänge 0 starten?

@andig Wie siehst du das? Machen könnte man das schon, aber überzeugt dass es besser ist bin ich bisher nicht...

@milefork Da steht noch eine Erklärung aus, was du mit

> Ach und die Zoom Stufen verstehe ich nicht bzw. finde ich jetzt nicht mehr schön. Sind die logarithmisch geworden?

meinst.

milefork, 2017-07-20:
> Ach und die Zoom Stufen verstehe ich nicht bzw. finde ich jetzt nicht mehr schön. Sind die logarithmisch geworden?

Damit meine ich das wenn man z.B. auf Stunde drückt, und dann auf den (-) Button dann geht er eine Stunde zurück, drückt man noch mal noch eine, drückt man noch mal dann auf einmal 2 Stunden zurück also von dem Intervall/Ausschnitt.

frankrichter, 2017-07-20:
+ halbiert das angezeigte Intervall, - verdoppelt es, das war schon immer so.

milefork, 2017-07-20:
@frankrichter okay, dann viel mir das jetzt nur so deutlich auf durch die für mich neue Intervall Teilung. Sry.

andig, 2017-09-06:
Ich würde mal sagen kein Bug. Wenns einen konkreten Vorschlag gibt wie die Implementierung zu ändern ist gerne her damit. Bisher hatte niemand eine wirklich gute Idee.

closed 2017-09-06 by andig