volkszaehler/volkszaehler.org #639: JpGraph is not PHP7 compatible

J-A-U, 2017-10-09:
Composer installed JpGraph 3.5.
For PHP7 we need 4.0.0 or newer.

andig, 2017-10-09:
Wer benutzt das denn noch- entfernen?

Viele Grüße,
Andreas

andig, 2017-10-09:
@J-A-U auch wenns drin bliebe- was ist das Problem? 4.0 sollte installiert werden wenn Du das manuell machst?

J-A-U, 2017-10-09:
> Wer benutzt das denn noch- entfernen?

Das sollte besser Christian auf der ML beantworten wozu er das braucht. "[vz-users] Missing format in middleware"

> 4.0 sollte installiert werden wenn Du das manuell machst?

Leider funktioniert die 4.0 auch noch nicht wie gewünscht.
`./composer.phar require --update-no-dev jpgraph/jpgraph:4.0.2`

Abfrage:
`http://localhost/middleware.php/data/***UUID****.png`

Antwort:
`{"version":"0.3","exception":{"message":"The function imageantialias() is not available in your PHP installation. Use the GD version that comes with PHP and not the standalone version.","type":"JpGraphExceptionL","code":0}}`

andig, 2017-10-09:
> Das sollte besser Christian auf der ML beantworten wozu er das braucht. "[vz-users] Missing format in middleware"

Dann bitte hier im Ticket melden. Das kann eigentlich nur an Fehler der Composernutzung liegen.

> The function imageantialias() is not available in your PHP installation. Use the GD version that comes with PHP and not the standalone version.

Ist GD2 installiert? Ohne das entsprechende PHP Modul wirds nix.

J-A-U, 2017-10-09:
> Das kann eigentlich nur an Fehler der Composernutzung liegen.

Composer installiert ab Werk den master Zweig von jpgraph. Das ist aber Version 3.5 und nicht PHP7 tauglich.
Deswegen hab ich eigentlich das Issue aufgemacht.

Zwischenzeitlich haben wir ausgetüftelt wie man 4.0 installiert (s.o.), aber da gibts dann das erwähnte  Folgeprobleme

> Ist GD2 installiert?

Jupp:
/etc/php/7.0/apache2/conf.d/20-gd.ini,

Ich hab an anderer Stelle ein Workaround gefunden:  https://github.com/mailwatch/MailWatch/issues/386

Der funktioniert in sofern das jpgraph endlich antwortet. Aber dann fehlen mir hier die TrueTypeFonts (raspian lite). Wobei danach gleich das nächste Problem kam...
Ich fürchte da hängt ein riesen Rattenschwanz mit dran. :/

andig, 2017-10-09:
> Composer installiert ab Werk den master Zweig von jpgraph. Das ist aber Version 3.5 und nicht mit PHP7 kompatibel.

Sorry, aber das stimmt so global einfach nicht:

	❯ composer require jpgraph/jpgraph
	Using version ^4.0 for jpgraph/jpgraph                        
	./composer.json has been updated
	Loading composer repositories with package information

Deshalb bitte steps to reproduce.

> aber da gibts dann das erwähnte Folgeprobleme ... Ich hab an anderer Stelle ein Workaround gefunden

https://github.com/mailwatch/MailWatch/issues/386 ist für mich keine Erklärung. Die Funktion wird an einer einzigen Stelle aufgerufen und die fangen wir ab, siehe https://github.com/ztec/JpGraph/search?utf8=✓&q=imageantialias&type=

So stochern wir im Nebel. 

> Aber dann fehlen mir hier die TrueTypeFonts 

Das Fontproblem müsste sich per Nachinstallation lösen lassen?

J-A-U, 2017-10-09:
> Sorry, aber das stimmt so global einfach nicht:

Und dennoch hat composer mit hier auf auf der PHP7 Spielwiese die 3.5 installiert. Augenscheinlich weil dies im Installscript nach master verlangt wird.
JpGraph wird ja auch nur installiert wenn man das explizit bejaht.

> Deshalb bitte steps to reproduce.

Ja, mach ich gleich. Ich bin gerade dabei den Ausgangszustand herzustellen.

> mailwatch/MailWatch#386 ist für mich keine Erklärung. 

Ok.

> So stochern wir im Nebel.

Deswegen hab ich vor ein paar Minuten die Notbremse gezogen.

> Das Fontproblem müsste sich per Nachinstallation lösen lassen?

Lief leider auch nicht glatt. Die von mir nachinstallierten Fonts lagen an der falschen Stelle.

J-A-U, 2017-10-09:
> Deshalb bitte steps to reproduce.

Ich installiere jpgraph praktisch wie in install.sh vorgesehen:
`pi@raspberrypi:~/volkszaehler.org $ ./composer.phar require --update-no-dev jpgraph/jpgraph:dev-master`
`pi@raspberrypi:~/volkszaehler.org $ ./composer.phar update`

`http://-IP-/middleware.php/data/-UUID-.png`
`{"version":"0.3","exception":{"message":"The installed PHP version (7.0.19-1) is not compatible with this release of the library. The library requires at least PHP version 5.1.0","type":"JpGraphExceptionL","code":0}}`

`pi@raspberrypi:~/volkszaehler.org $ cat vendor/jpgraph/jpgraph/lib/JpGraph/VERSION`
`Version: v3.5.0b1`
So wirds sicher nix.

Also andere Version:
`pi@raspberrypi:~/volkszaehler.org $ ./composer.phar require --update-no-dev jpgraph/jpgraph:4.0.2`
`pi@raspberrypi:~/volkszaehler.org $ ./composer.phar update`

`pi@raspberrypi:~/volkszaehler.org $ cat vendor/jpgraph/jpgraph/lib/jpgraph/VERSION`
`Version: v4.0.2`

`http://-IP-/middleware.php/data/-UUID-.png`
`{"version":"0.3","exception":{"message":"The function imageantialias() is not available in your PHP installation. Use the GD version that comes with PHP and not the standalone version.","type":"JpGraphExceptionL","code":0}}`

Soweit ich das jetzt sehe haben wir beim Support über die ML im Bezug auf Composer wirklich was vergeigt, deswegen sieht das jetzt ein wenig anders aus.
Ein png bekommen wir aber trotzdem nicht, die 4.0.2 hat auch Probleme.

andig, 2017-10-09:
> Ich installiere jpgraph praktisch wie in install.sh vorgesehen:

Erster Fehler (also meiner ;). Du zwingst ihn ja master zu nehmen, also tut er's auch:

    ./composer.phar require --update-no-dev jpgraph/jpgraph

oder 

    ./composer.phar require --update-no-dev jpgraph/jpgraph:^4.0

> Soweit ich das jetzt sehe haben wir beim Support über die ML im Bezug auf Composer wirklich was vergeigt, deswegen sieht das jetzt ein wenig anders aus.

Was meinst Du? PNG schaue ich mir separat nochmal an.

J-A-U, 2017-10-09:
> Was meinst Du?

Die Rückmeldung der MW sah anders aus, vermutlich weil composer update gefehlt hatte oder die Reihenfolge falsch war.

Ich schätze es ist ok wenn wir bei Christian die Installation mal soweit reparieren das er die Fehlermeldung bekommt wie ich sie jetzt habe?

andig, 2017-10-11:
> Ich schätze es ist ok wenn wir bei Christian die Installation mal soweit reparieren das er die Fehlermeldung bekommt wie ich sie jetzt habe?

Ja, install.sh habe ich schon korrigiert.

andig, 2017-10-25:
ping @J-A-U ist erledigt, oder?

J-A-U, 2017-10-25:
> `http://-IP-/middleware.php/data/-UUID-.png`
`{"version":"0.3","exception":{"message":"The function imageantialias() is not available in your PHP installation. Use the GD version that comes with PHP and not the standalone version.","type":"JpGraphExceptionL","code":0}}`

Die Fehlermeldung ist noch akut. Extra Issue?

andig, 2017-10-25:
> mailwatch/MailWatch#386 ist für mich keine Erklärung. Die Funktion wird an einer einzigen Stelle aufgerufen und die fangen wir ab, siehe https://github.com/ztec/JpGraph/search?utf8=✓&q=imageantialias&type=

Ich blicks nicht. Kannst Du mal ein `?debug=1` anhängen oder mir Zugang per PM geben?

J-A-U, 2017-10-25:
> Kannst Du mal ein ?debug=1 anhängen

{
	"version": "0.3",
	"exception": {
		"message": "The function imageantialias() is not available in your PHP installation. Use the GD version that comes with PHP and not the standalone version.",
		"type": "JpGraphExceptionL",
		"code": 0,
		"file": "\/home\/pi\/volkszaehler.org\/vendor\/jpgraph\/jpgraph\/lib\/jpgraph\/src\/jpgraph_errhandler.inc.php",
		"line": 113,
		"backtrace": [
			{
				"file": "\/home\/pi\/volkszaehler.org\/vendor\/jpgraph\/jpgraph\/lib\/jpgraph\/src\/gd_image.inc.php",
				"line": 110,
				"function": "RaiseL",
				"class": "JpGraphError"
			},
			{
				"file": "\/home\/pi\/volkszaehler.org\/vendor\/jpgraph\/jpgraph\/lib\/jpgraph\/src\/themes\/UniversalTheme.class.php",
				"line": 85,
				"function": "SetAntiAliasing",
				"class": "Image"
			},
			{
				"file": "\/home\/pi\/volkszaehler.org\/vendor\/jpgraph\/jpgraph\/lib\/jpgraph\/src\/jpgraph_theme.inc.php",
				"line": 66,
				"function": "SetupGraph",
				"class": "UniversalTheme"
			},
			{
				"file": "\/home\/pi\/volkszaehler.org\/vendor\/jpgraph\/jpgraph\/lib\/jpgraph\/src\/jpgraph.php",
				"line": 1132,
				"function": "ApplyGraph",
				"class": "Theme"
			},
			{
				"file": "\/home\/pi\/volkszaehler.org\/lib\/View\/JpGraph.php",
				"line": 103,
				"function": "SetScale",
				"class": "Graph"
			},
			{
				"file": "\/home\/pi\/volkszaehler.org\/lib\/Router.php",
				"line": 177,
				"function": "__construct",
				"class": "Volkszaehler\\View\\JpGraph"
			},
			{
				"file": "\/home\/pi\/volkszaehler.org\/lib\/Router.php",
				"line": 128,
				"function": "handleRaw",
				"class": "Volkszaehler\\Router"
			},
			{
				"file": "\/home\/pi\/volkszaehler.org\/htdocs\/middleware.php",
				"line": 44,
				"function": "handle",
				"class": "Volkszaehler\\Router"
			}
		]
	},
	"debug": {
		"database": "pdo_mysql",
		"time": 0.12394,
		"uptime": 1353040,
		"load": [
			0.06,
			0.49,
			0.47
		],
		"commit-hash": "9accdd2c1fa2d0a3756496fbcbc704c4675487d9",
		"php-version": "7.0.19-1",
		"messages": []
	}
}

> oder mir Zugang per PM geben?

Mail ist raus.

andig, 2017-10-25:
Geh mal hier her: https://github.com/volkszaehler/volkszaehler.org/blob/master/lib/View/JpGraph.php#L119

und ändere das so:

		if (function_exists('imageantialias')) {
			$this->graph->img->SetAntiAliasing(true);
		}
		else {
			function imageantialias() { }
		}
		
Leider gibts im JpGraph Code noch eine Stelle wo das aufgerufen wird, die Funktion muss also existieren...

J-A-U, 2017-10-25:
Keine Änderung.

andig, 2017-10-25:
Hast Du mal wie vorgeschlagen `?debug=1` angehängt? Dann siehst Du im Stack Trace dass der Fehler schon in Zeile 103 beim `SetScale` kommt. Also einfach den geänderten Block mal hochschieben vor das `SetScale`.

J-A-U, 2017-10-25:
> Hast Du mal wie vorgeschlagen ?debug=1 angehängt?

Ja, ich kann wegen mangelnder Erfahrung mit der Ausgabe aber nicht viel anfangen.
Mich irritiert allerdings das wir genau die Funktion einbinden die laut Fehlermeldung nicht zur Verfügung steht.

> Also einfach den geänderten Block mal hochschieben vor das SetScale.

Jetzt wird Zeile 110 angemeckert, Verhalten ansonsten identisch.

andig, 2017-10-26:
> Mich irritiert allerdings das wir genau die Funktion einbinden die laut Fehlermeldung nicht zur Verfügung steht.

Nee, wir deklarieren sie als leere Funktion so dass sie ab da existiert.

Jetzt verstehe ich aber den Fehler: wir deklarieren die Funktion in einem namespace- damit ist sie nicht global und nicht von JpGraph aufrufbar. Ich muss schauen wie sich das elegant machen ließe.

andig, 2017-10-31:
ping @J-A-U

J-A-U, 2017-10-31:
Oh, hab das Update deines Kommentars nicht mitbekommen.

Ich schau mal ob ich auf PHP7.2 hoch komme und das ne Lösung bereitstellt.

J-A-U, 2017-11-01:
Für Debian Stretch wird kein 7.2 ausliefert werden.
Selbst compilieren klemmts bei mir an allen möglichen Ecken und Enden.

Nachdem ich jetzt erfahren hab das der Nutzer den es betrifft den erwähnten Workaround anwendet ohne in der ML was gesagt zu haben ist meine Motivation das Problem zu lösen ziemlich gesunken.

andig, 2017-11-01:
> Für Debian Stretch wird kein 7.2 ausgeliefert werden.

Kann ja nicht- 7.2 ist noch nichtmal released. Allerdings sollte auch 7.1 mit gd2 kommen, siehe https://packages.debian.org/source/stretch/php7.0

Aber wenn Motivation unten dann lass uns das Thema zu machen ;)

J-A-U, 2017-11-01:
GD ist, wie schon erwähnt, aktiv. Habs gerade auch nochmal für CLI kontrolliert.

Naja. Wie dem auch sei...

closed 2017-11-01 by J-A-U

andig, 2017-11-01:
Root cause vmtl. hier: https://lists.debian.org/debian-user/2015/03/msg00283.html

> It's about the fact that Debian, by it's policy, does not use bundled libgd from PHP source, and unfortunately upstream libgd had (and still has) missing functions.