volkszaehler/volkszaehler.org #640: Gruppen nach reload leer seit PR#626

J-A-U, 2017-10-10:
Wenn man eine Gruppe (oder Gebäude) abonniert oder per URL aufruft ist diese erstmal wie gewohnt in der Tabelle des FE vorhanden und die untergeordneten Kanäle werden dargestellt (Tabelle und Graph).
Nach einem Reload der gesamten Seite taucht nur noch die Gruppe in der Tabelle auf, keine Kanäle und (wenn man keine einzelnen Kanäle abonniert hat) ein leerer Graph.

Ich konnte das Verhalten bis PR#626 (e2469cb) zurückverfolgen. Davor war das Verhalten nicht zu beobachten.

jahir, 2017-10-10:
Kann ich bestätigen. Workaround: Details der Gruppe öffnen, Bearbeiten, Speichern.

andig, 2017-10-12:
So, macbook repariert, getestet. Ich kanns nicht nachvollziehen. Gibt mir jemand HTTP/SSH Zugang zum testen? Habt ihr APCU installiert? Performance MW?

BlueAndi, 2017-10-12:
Ich weiss nicht ob es das gleiche Problem ist, aber ich verwende das Debian Docker Image von Dir Andi.
Wenn ich wie folgt vorgehe, sind nach einem Browser Refresh alle in der Gruppe deaktiviert:
1. Gruppe deaktivieren (Ordner)
2. Einen in der Gruppe aktivieren
3. Browser refresh

Gehe ich anders vor, sind nach einem Browser Refresh all in der Gruppe aktiviert:
1. Gruppen aktivieren (Ordner)
2. Einen in der Gruppe deaktivieren
3. Browser refresh

Getestet via IE 11 und FF V56.0 mit jeweils gleichem Verhalten.

andig, 2017-10-12:
> Ich weiss nicht ob es das gleiche Problem ist, aber ich verwende das Debian Docker Image von Dir Andi.

Das ist was Anderes. Die Gruppensichtbarkeit wird immer vom obersten Knoten bestimmt. Das ist das erwartete Verhalten bis wir eine coolere Lösung haben.

J-A-U, 2017-10-13:
> Gibt mir jemand HTTP/SSH Zugang zum testen?

Erst am Wochenende, bin heute den ganzen Tag unterwegs.

> Habt ihr APCU installiert? Performance MW?

Apcu betrifft ja schon das PHP-Paket? Das ist drauf.
Ansonsten Standard-Installation per Script, keine PPM.

andig, 2017-10-14:
Es gibt ein Cachingproblem. Workaround `devmode=true`, für eine richtige Lösung muss ich etwas Research machen, evtl. muss ich das zurück bauen :(

andig, 2017-10-14:
Bitte mal testen- damit sollte es auch mit `devmode=false` wieder laufen.

J-A-U, 2017-10-15:
Leider keine Besserung.

andig, 2017-10-15:
Wtf. Zur Not Apache durchgestartet? Das *muss gehen* (tm). Darf ich da nochmal rein schauen?

closed 2017-10-15 by andig