volkszaehler/volkszaehler.org #652: (Temperatur-)Skalierung auf Y-Achse verliert Bezugspunkt

nuccleon, 2017-12-19:
Aus der Mailingliste: [Temperatur Skalierung auf Y-Achse](https://marc.info/?l=volkszaehler-dev)

> Als Beispiel greife ich auf vergleichbare Graphen zurück die ich hier
habe. Raum- und Außentemperatur. Alle Kanäle außer die beiden sind
ausgeblendet. Aktuell gibts keine Minusgrade draußen, also bleibt die
Skalierung mit 0-25°C unverändert, auch wenn nur 1 Kanal dargestellt
wird.
>
>Gehe ich nun aber Tageweise zurück bis zu einer Stelle mit Minusgraden
wird neu skaliert. -5-25°C.
Lass ich jetzt nur einen der beiden Kanäle anzeigen verändert sich die
Skalierung. Bei Außentemperatur von -5-5°C. Bei Raum von 20-22°C.
Der Bezug auf 0° bleibt jetzt auch gebrochen wenn man in Zeiträume
zurückkehrt in denen nur Plusgrade im Graph auftauchen.
>
>Der Effekt tritt auch nur auf wenn die Skalierung nach unten
unterbrochen wird, wird wegen Überschreitung nach oben neu skaliert
bleibt der Bezug 0°C erhalten.

andig, 2017-12-19:
Passiert das nur beim reinscrollen in einen entsprechenden +/- Bereich oder auch wenn schon initial entsprechende Werte existieren und nur Kanäle versteckt werden? Kann ich mir das evtl. in einer Installation (Link per PM) mal anschauen?

nuccleon, 2017-12-19:
Passiert auch ohne scrollen. Aktivieren / deaktivieren eines Kanals reicht.
Link kommt per PM

andig, 2018-03-13:
@nuccleon schau mal bitte ob das für Dich funktioniert. Sollte ziemlich vollständig sein solange Du nicht selber in den Options Axen angelegt hast- da gibts Sonderfälle die sich so nicht behandeln lassen. Das wäre dann aber ein anderes Thema....

nuccleon, 2018-03-14:
> Javascript Runtime Error
> TypeError: axis is undefined

andig, 2018-03-14:
Bäh! Ich sehe nicht in welcher Zeile das passieren soll? Zeilennummer? Magst Du Deine vollständige Datei mal posten? Bitte nochmal prüfen.

andig, 2018-03-18:
Gibts neue Erkenntnisse? Sonst merge ich‘s so- dann findet der Nächste den Fehler 😬

nuccleon, 2018-03-18:
Sorry, bin noch nicht dazu gekommen :-/Ich teste morgen Abend, versprochen...

andig, 2018-03-19:
> Ich könnte dir per PN mal nen link zu meiner Installation geben, dann kannst du dir das mal ankucken

Ja, mach mal bitte- HTTP reicht (cpuidle@gmx.de), dazu UUID vom Rootkanal um das anzuschauen.

nuccleon, 2018-03-20:
Also der PR scheint okay.
Wenn er gemerged ist können wir das Issue zu machen.

closed 2018-03-20 by andig