volkszaehler/volkszaehler.org #666: DB Port konfigurieren

J-A-U, 2018-01-28:
Die Middleware erlaubt derzeit keinen anderen als den Standard Port für die Datenbank. in der Konfig ist keine entsprechende Option vorgesehen.

Die führt zum Beispiel in Verbindung mit einem Wechsel von MariaDB5 auf MariaDB10 zu Problemen weil die neuere Version zusätzlich installiert wird und auf Port 3307 statt 3306 lauscht.

andig, 2018-01-28:
> Die Middleware erlaubt derzeit keinen anderen als den Standard Port für die Datenbank. In der Konfig ist keine entsprechende Option vorgesehen.

Gib einfach einen `port` an?

J-A-U, 2018-01-28:
Meinst du 
`$config['db']['host']				= '127.0.0.1:3307';`

oder
`$config['db']['port']				= '3307';`

andig, 2018-01-28:
Entweder port separat oder gleich connect string angeben http://docs.doctrine-project.org/projects/doctrine-dbal/en/latest/reference/configuration.html

J-A-U, 2018-01-28:
Ok.
Wäre es vielleicht angebracht den Parameter für Port in die Beispielkonfig mit einzubinden?

andig, 2018-01-28:
Klar- wollen wir's so lassen oder gleich auf Connectstring ändern um die längliche Config gleich etwas auszudünnen? Evtl. mit dem Link zu doctrine? Magst Du einen PR machen?

closed 2018-01-28 by J-A-U