volkszaehler/volkszaehler.org #669: No MIN MAX AVG CUR values on duplicated values

Feilner, 2018-02-28:
I have some items in more than one group.
The MIN MAX AVG CUR values are only shown on the first item. No matter if first item ist visible (group is opened) or not.

See Picture e.g. for the "TempSoll EG Esszimmer"

[image]

andig, 2018-03-05:
Could you check if the issue is resolved in my development repo (https://github.com/andig/volkszaehler.org, next branch)? It already includes a bunch of fixes and new features that have not yet been merged.

Feilner, 2018-03-05:
Hello AndiG,

yes the issue is solved on your branch!

Do you have a time plan when your branch is integraded into the master?

Best regards,

Oliver

andig, 2018-03-05:
Not currently. You can continue using that branch if you want.

andig, 2019-03-21:
Should be fixed by recent merges into master.

closed 2019-03-21 by andig