volkszaehler/volkszaehler.org #673: Relative Zeitangaben in Frontend-URL-Parameter

l3d00m, 2018-03-04:
Hallo,

gibt es eine Möglichkeit, relative Zeitangaben wie z.B. `midnight` oder `last hour` im Frontend beim `from=`-Parameter zu benutzen? In dem [dazugehörigen Code](https://github.com/volkszaehler/volkszaehler.org/blob/master/htdocs/js/functions.js#L178) ist zwar die Rede von "speaking timestamps", allerdings unterstützt javascripts `new Date()` das nicht.

Deshalb passiert auch folgendes, wenn ich bei mir z.B. `http://192.168.111.50/?from=midnight` aufrufe:

```
middleware.php/data/[uuid here].json?unique=1520157901390&from=NaN&to=1520157901045&tuples=478:

Invalid time format: 'NaN'
```

PHP's date Funktion unterstützt ja zB solche Zeitangaben, wenn ich also obige middleware URL mit `from=midnight` aufrufe, klappt das ohne Probleme.

andig, 2018-03-04:
Aktuell geht das nur beim Aufruf der Middleware, aber nicht des Frontends.

andig, 2018-03-08:
@l3d00m schau Dir mak locutus (http://locutus.io/php/datetime/strtotime/) an. Wenns damit geht einfach die from/to Parameter da durchschicken, PR merge ich gerne.

l3d00m, 2018-03-09:
Vom Prinzip her habe ich da jetzt schon was am Laufen, ich bin mir nur noch nicht sicher, wo die Funktion aus deinem Link am besten hin soll (im Moment habe ich sie einfach in den functions.js gepastet). Wenn das geklärt ist, werde ich das dann die Tage durchtesten und einen PR machen.

andig, 2018-11-18:
I've just noticed there is also https://github.com/agenda/human-intervalwhich sounds great.

PR #800 (2020-03-15): Allow relative time parameters in frontend, implements #673
See discussion in #797. I've made a new pr as I now used the strtotime locutus library, which I manually minified. Its footprint is much smaller with 4.7KB

closes #673
closed 2020-03-15 by andig