volkszaehler/volkszaehler.org #696: basic write protection for data.

tbessenreither, 2018-07-27:
Damit ich Kanäle öffentlich machen kann ohne das mir jemand rein schreibt habe ich in meiner Installation eine einfache Token basierte Authentifizierung eingebaut. Evtl hat ja jemand Interesse daran.

Vorab, ich ziehe mir den token über $_REQUEST['writetoken']. Ich weiß nicht ob das evtl aus anderen Datenquellen Probleme macht, evtl muss dann die eine Zeile angepasst werden.

Berechnet wird der Token mit sha256 aus dem zusammengesetzten String
`$uuid.'-'.Util\Configuration::read('writeprotection.sha256Salt')`

Integriert ist es im DataController::add($uuid)
`public function add($uuid) {
		$channel = $this->ef->get($uuid, true);
		
		if (!$channel instanceof Model\Channel) {
			throw new \Exception('Adding data is only supported for channels');
		}

		if(Util\Configuration::read('writeprotection.active') === true){
			$securityHash = hash('sha256', $uuid.'-'.Util\Configuration::read('writeprotection.sha256Salt'));
			if(!isset($_REQUEST['writetoken'])){
				throw new \Exception('missing writetoken parameter');
			}
			$submittedHash = $_REQUEST['writetoken'];
			if($_REQUEST['writetoken'] !== $securityHash){
				throw new \Exception('write permission denied');
			}
		}`

und kann in der config wie folgt aktiviert werden:
`$config['writeprotection'] = array(
	'active' => true,
	'sha256Salt' => 'insert your hash salt here'
);`

andig, 2018-07-29:
Hallo Tobias,

In Deinem Fall scheint es immer noch möglich zu sein Kanäle zu löschen- keine gute Idee ;)

Unter https://github.com/andig/volkszaehler.org/blob/next/etc/volkszaehler.conf.template.php#L127 gibt es bereits eine vollständige Implementierung für Multi-User Betrieb mittels JSON Web Tokens in Form von Firewallregeln.

Wurde Anfang des Jahres auch auf der Mailingliste diskutiert, hat es bisher aber nicht in dieses Repository rüber geschafft.

Viele Grüße,
Andreas

tbessenreither, 2018-07-30:
Gut zu wissen das da evtl noch was kommt.
Das löschen habe ich bei mir schon gepatcht, ich hab das hier nur nicht mehr rein geworfen. Im Code oben ist auch noch eine Zeile zu viel ging nur darum wo man das Prinzipiell auf die schnelle selbst einbauen könnte.

Das Ticket hier kann denke ich zu gemacht werden.
Grüße
Tobias

closed 2019-03-21 by andig