volkszaehler/volkszaehler.org #716: Falsche Berechnung bei Ausreisern

Falke-Design, 2019-01-15:
Hallo,
in der Tagesgrafik werden Ausreiser falsch berechnet. (Oder ich hab eine Einstellung nicht gesetzt 😅 ).
In der Tagesgrafik wird der Ausreiser mit ~260W berechnet und angezeigt, zoomt man aber rein, werden die richtigen ~1,8kW angezeigt.
Hier ein [Video](https://www.dropbox.com/s/709dgk99ur4zg2f/volkszaehler%20-%20falsche%20berechnung.mp4?dl=0).

Hier noch die Daten in [JSON ](https://www.dropbox.com/s/c16y5qkvzu29wvh/data_15_01_19.json?dl=0)und [CSV](https://www.dropbox.com/s/fx7se2e02alk1uk/data_15_01_19.csv?dl=0).

Lg

frankrichter, 2019-01-15:
Hi,

das passt schon so, weil in der Tagesansicht über mehrere Datensätze
gemittelt wird. In den Middleware-Daten siehst du das an der dritten Stelle
im Tupel [timestamp, value, rows]. rows gibt an, aus wievielen
Rohdatensätzen ein Tupel berechnet wurde.

Grüße
Frank

Falke-Design, 2019-01-15:
Danke für die rasche Antwort aber wo finde ich diese Informationen, weiß grad nicht wo ich nachsehen soll und wo stelle ich das um?

Lg

frankrichter, 2019-01-15:
Das Frontend berechnet die angeforderte Anzahl tuples anhand der
Fenstergröße.
Was möchtest du da umstellen, es macht ja wenig Sinn, mehr Datensätze zu
berechnen als dein Bildschirm in der Breite Pixel hat.

Falke-Design, 2019-01-15:
Naja genau diese Spitzen sind das was mich interessiert und diese werden halt "falsch" dargestellt :/

frankrichter, 2019-01-15:
Schau dir deine Daten mit passendem Zoomfaktor an, dann wird auch nichts
gemittelt :-)

Wieviele Datensätze hast du pro Tag? Du kannst höchstens mal versuchen, in
der options.js bei tuples einen Wert einzutragen, der höher ist als die
Zahl vorhandener Datensätze. Ob das dann noch sinnvoll darstellbar ist, ist
ne andere Frage...

>

Falke-Design, 2019-01-15:
hmmm ok. Also kann ich nirgens in ner Datei umstellen, dass Werte über z.b. 1kW nicht herunter gerechnet werden?

~1110
Habs probiert aber hat sich nichts wirklich verändert.

frankrichter, 2019-01-15:
Ne, eine derartige Einstellung gibt es nicht.

Nach Änderungen am Javascript immer Browsercache löschen, sonst siehst du
eh keinen Effekt.

Stell mal in der vzlogger.conf eine sinnvolle aggtime ein, damit bei hohen
Leistungen nicht sinnlos viele Datensätze erzeugt werden, z.B. 60 Sekunden,
wenn du die Tagesansicht nutzen willst. Oder versuch's zumindest mit 20
oder 30, damit deine Peaks sauber erfasst werden.

Falke-Design, 2019-01-15:
Aja Cache leeren hatte ich vergessen. Aber bringt nicht wirklich mehr.

Verwende den Logger garnicht 😅 Lasse es selber über ein kleines python programm erfassen.

frankrichter, 2019-01-15:
Dann bau dort was ein, damit die Impulse nicht mit beliebig hoher Frequenz
geloggt, sondern aufsummiert werden. Kommt der Anzeige bei längeren Zeiten
bestimmt zugute.

Falke-Design, 2019-01-15:
Ja ist sicher ne Überlegung wert danke.

In welcher Datei werden die Datensätze ausgelesen und das Diagramm gerendert? Hab die Dateienstrucktur / aufbau noch nicht ganz durchschaut.

frankrichter, 2019-01-15:
Datenaufbereitung macht die Middleware, Chart im Wesentlichen wui.js. Dort
kannst du mal Zeile 91 anschauen, da erfolgt die Berechnung der
angeforderten tuples. Kannst ja testweise mal 2000 o.ä. dort eintragen
statt dem vorhandenen Ausdruck.

Ob der Parameter tuples in der options.js wirklich was bewirkt oder
überholt ist, weiß ich grad auch nicht.

andig, 2019-01-16:
Hier werden dieDaten geladen: https://github.com/volkszaehler/volkszaehler.org/blob/master/htdocs/js/entity.js#L329, da kannst Du tuples und group rausnehmen. Tuples wird je nach Fenstergrösse gesetzt wie schon verlinkt.  

Viele Grüße, Andreas

frankrichter, 2019-01-16:
Erwartest du tatsächlich, dass auch in der Wochenansicht eine Lastspitze,
die kürzer als 1 Minute gedauert hat, korrekt wiedergegeben wird? Wie soll
das mit der Auflösung eines handelsüblichen Monitors zusammengehen?
Viel mehr als 2000 Tupel macht keinen Sinn, weil der Plot nicht mehr
wiedergegeben kann, außer du hast einen 8K-Monitor...

Falke-Design, 2019-01-16:
Hab ich nicht bedacht. 😅 
Danke mal auf jedenfall.

Kann man schließen.

andig, 2019-01-16:
Naja - Dein Problem bleibt, mal abgesehen davon dass Du die Performance ruiniert hast. 

Schön wäre (tm) wenn wir Median, Min und Max darstellen könnten so dass solche Ausreisser auch im Zoom erkennbar wären. Ich denke da eher an Banddiagramme als an Fehlerbalken. 

Wäre eine schöne Übung falls sich jemand in die Middleware einarbeiten möchte, Coaching biete ich an...

Viele Grüße,
Andreas

andig, 2019-01-24:
Wie besprochen works as designed.