volkszaehler/volkszaehler.org #724: Wrong Scaling for meter type 'water' in case of consumption of more than 1000l/h

nuccleon, 2019-02-18:
The unit of water consumption changes from l/h to m³/h in case of a consumption of more than 1000l/h.
The legend shows the correct value of ~1.03 m³/h, but the scale is still in l/h mode.

andig, 2019-02-18:
See https://github.com/andig/volkszaehler.org/issues/31#event-2147132948 for screenshot (missing here)

thhoe, 2019-02-24:
Hallo zusammen,

ich bin gerade dabei mir ein Script zu basteln, mit dem ich einen S0-Kanal überwachen kann. 

Es soll die Anzahl der im Impulse  in den letzten X Minuten ausgewertet werden.

Für das ganze habe ich eine Vorlage gefunden (wo weiß ich nicht mehr), die aber nicht funktioniert.

Problem ist das die Antwort der Middleware für mich nicht Nachvollziehbar ist.

Die URL zur Abfrage lautet: http://localhost/middleware.php/data/e2b64f10-92eb-11e2-94e4-ed0995b4eb89.csv?from=1551009686000 <http://localhost/middleware.php/data/e2b64f10-92eb-11e2-94e4-ed0995b4eb89.csv?from=1551009686000> 

Das sollte richtig sein nach: https://wiki.volkszaehler.org/development/api/reference

Antwort der Middleware:

    # source:;volkszaehler.org 
    # version:;0.3 
    # uuid:;e2b64f10-92eb-11e2-94e4-ed0995b4eb89 
    # title:;Strom_L2 
    # from:;2019-02-24 06:48:50 
    # to:;2036-02-07 07:37:55 
    # min:;2036-02-07 07:28:17; => ;0 
    # max:;2036-02-07 07:29:41; => ;1800 
    # average:;0 
    # consumption:;43.5 
    # rows:;88 
    2019-02-24 06:50:00;25.714;1 
    2036-02-07 07:28:17;0;1 
    2036-02-07 07:28:24;257.143;1 
    2036-02-07 07:28:29;360;1 
    2036-02-07 07:28:5.....

Zum testen habe ich den Zeitbereich mit to= Eingegrenzt, URL:  http://localhost/middleware.php/data/e2b64f10-92eb-11e2-94e4-ed0995b4eb89.csv?from=1551009097000&to=1551009157000 <http://localhost/middleware.php/data/e2b64f10-92eb-11e2-94e4-ed0995b4eb89.csv?from=1551009097000&amp;to=1551009157000> 

Antwort der Middleware:

    # source:;volkszaehler.org 
    # version:;0.3 
    # uuid:;e2b64f10-92eb-11e2-94e4-ed0995b4eb89 
    # title:;Strom_L2 
    # from:;2019-02-24 06:48:50 
    # to:;2036-02-07 07:28:17 
    # min:;2036-02-07 07:28:17; => ;0 
    # max:;2019-02-24 06:50:00; => ;25.714 
    # average:;0 
    # consumption:;1 
    # rows:;3 
    2019-02-24 06:50:00;25.714;1 
    2036-02-07 07:28:17;0;1

Hier sieht mann womit ich nicht klar komme. Obwohl die Abfragen mit Abstand ausgeführt wurden, bekomme ich in beiden fällen die selbe (falsche) Antwort für "# from:" und "# to:". Sogar die 1 Daten sind identisch. 

Wie kommen diese Werte zustande?

Wo kann ich noch einen Schalter für "Debug" setzen?

Oder gibt es einen besseren Weg für mein Script?

Ich hoffe mir kann jemand helfen.

Mit freundlichen Grüßen,

Thomas

Mail: thomas@thhoe.de <mailto:thomas@thhoe.de>

andig, 2019-03-16:
@thhoe deinen Kommentar habe ich weil vollkommen off-topic gelöscht. Bitte nicht willenlos fremde Themen zuspammen.

andig, 2019-03-21:
> The legend shows the correct value of ~1.03 m³/h, but the scale is still in l/h mode.

According to screenshot the problem is that the chart will show the original _value_ while using the scaled _unit_. The table shows scaled _unit_ together with scaled _value_.