volkszaehler/volkszaehler.org #727: Variable "zero point" of ordinate

beckhj, 2019-03-07:
Wenn Daten mit großen Werten nur eine kleinen variablen Anteil haben, ist die grafische Auflösung leider schlecht.
Beispiel Luftdruck: Typische Werte liegen wohl meistens zwischen 900 und 1199 hPa. Da sieht man grafisch eine Änderung von 20 hPa fast nicht.
Ideal wäre wenn man eine y-Achse so anlegen kann, dass der Minimalwert des Anzeigebereiches auf dem Ursprung liegt. Wenn sich an einem Tag beispielsweise der Luftdruck von 980 auf 1020 hPa ändert hätte man hier eine deutlich bessere grafische Auswertemöglichkeit.
Ähnliches gilt auch für Temperaturen die sich ggf. nur um 10-20 Grad ändern, aber z.B. bei 50-60 Grad liegen (hier fällt mir z.B. ein Brauchwasser-Boiler ein).
Dieser dynamische Null-Punkt gilt dann natürlich nur pro Einheit, also Temperaturen, Drücke, etc.

andig, 2019-03-07:
Du kannst selber Axen anlegen und min/max definieren.

beckhj, 2019-03-08:
Danke, das würde mir helfen.
Ich habe aber leider in der Doku/Wiki das nicht gefunden. Hast Du ein Beispiel?

Danke und Grüße
Hans-Joachim

jahir, 2019-03-08:
in htdocs/js/options.js:

```
...
vz.options.plot = {
...
        yaxes: [
...
                {
                        axisLabel: 'hPa',
                        min: null,
                        position: 'right',
                        tickFormatter: vz.wui.tickFormatter
                },
                {
                        axisLabel: '%',
                        min: 0,
                        max: 100,
                        position: 'right',
                        tickFormatter: vz.wui.tickFormatter
                },
        ],
...
```

`min: null` sollte eigentlich bedeuten "Achsen-Minimum selbst bestimmen", und ich bin mir sicher, daß das auch mal funktioniert hat, aktuell aber irgendwie nicht mehr...

andig, 2019-03-09:
Falls null nicht funktioniert bitte Ticket oder min weglassen (undefined).

Viele Grüße,
Andreas

frankrichter, 2019-03-09:
min: undefined ist ja der Originalzustand in der options.js. Nach meiner Erfahrung wird damit allerdings immer mit min = 0 skaliert, außer es sind negative Werte im Spiel. Wenn eine Achse einmal negative Werte hatte, wird sie beim weiterscrollen dynamisch skaliert (min != 0).

andig, 2019-03-16:
> min: null sollte eigentlich bedeuten "Achsen-Minimum selbst bestimmen", und ich bin mir sicher, daß das auch mal funktioniert hat, aktuell aber irgendwie nicht mehr...

Das tut er auch- negativ falls negativ. sonst 0. Für `hPa` müsste einfach `min` auf 900 gesetzt werden.

Können wir das Thema damit zu machen?