volkszaehler/volkszaehler.org #746: Create ARM docker image

andig, 2019-03-16:
...or really support multi-arch docker image (see https://github.com/andig/volkszaehler.org/issues/35 for orignal request)

andig, 2019-03-17:
See https://github.com/jorge07/alpine-php/issues/55#issuecomment-473605421.  Docker image will automatically become multi-arch once https://github.com/jorge07/alpine-php/issues/55 is resolved.

StefanSchoof, 2020-08-06:
Why using the jorge07 images and not the official ones? The official images all supports arm (v6 and v7)

I tried then it complies and show the webpage:
```Dockerfile
# Copyright (c) 2019 Andreas Goetz <cpuidle@gmx.de>

FROM composer AS builder

WORKDIR /vz

COPY composer.json /vz

RUN composer install --no-ansi --no-scripts --no-dev --no-interaction --no-progress --optimize-autoloader

COPY . /vz

FROM php:7.3

EXPOSE 8080
EXPOSE 8082
EXPOSE 5582

RUN docker-php-ext-install pcntl pdo_mysql

COPY --from=builder /vz /vz
COPY --from=builder /vz/etc/config.dist.yaml /vz/etc/config.yaml

# modify options.js
RUN sed -i "s/url: 'api'/url: '',/" /vz/htdocs/js/options.js

CMD /vz/vendor/bin/ppm start -c /vz/etc/middleware.json --static-directory /vz/htdocs --cgi-path=/usr/local/bin/php
```

andig, 2020-08-06:
Please open a pr. I cannot remember why this didn‘t seems to be an option.

andig, 2020-08-06:
Guess I felt thatbthe 140+ MB for the php image are laughable..

StefanSchoof, 2020-08-07:
There is also an alpine flavor of the official php image. That should be less MB.

I tried this dockerfile on a rp zero with arm v6 and composer is crashing with a SIGSEV. armv7 I haven't tested yet.

I will create an PR so we can discuss the details there.

StefanSchoof, 2020-08-09:
For arm32 v6 issue I created https://github.com/composer/docker/issues/117

PR #817 (2020-09-29): Add Seihon for multi arch
Resolves #746 

First test if Seihorn is in path on Travis
closed 2020-09-29 by andig