volkszaehler/volkszaehler.org #747: Zählerüberlauf berücksichtigt?

emeins, 2019-03-19:
Ich nutze in VZ einen Kanal vom Typ Zähler für die Erfassung von Netzwerkdaten (in/outByte) über die Middleware API. Dieser Counter zählt jedes Byte was rein/raus geht und startet wieder bei 0, wenn alle Bits gekippt sind. Beispiel:
| 12861161 |         41 | 1552995961000 | 4285277806 | 2019-03-19 11:46:01.0000      |
| 12861171 |         41 | 1552996021000 | 4288581075 | 2019-03-19 11:47:01.0000      |
| 12861185 |         41 | 1552996081000 | 4291640658 | 2019-03-19 11:48:01.0000      |
| 12861193 |         41 | 1552996141000 | **_4294823335_** | 2019-03-19 11:49:01.0000      |
| 12861201 |         41 | 1552996201000 |    **_3034254_** | 2019-03-19 11:50:01.0000      |
| 12861393 |         41 | 1552996261000 |    6158801 | 2019-03-19 11:51:01.0000      |
| 12861401 |         41 | 1552996321000 |    9322314 | 2019-03-19 11:52:01.0000      |
| 12861417 |         41 | 1552996381000 |   12567278 | 2019-03-19 11:53:01.0000      |

In der Darstellung von VZ erhalte ich dann durch die Deltabildung immer große Minuswerte, was so nicht passt.

Ist ein solcher Zählerüberlauf in VZ berücksichtigt bzw. gibt es einen entsprechenden Datentyp oder muss ich das vor der Middleware abfangen?

andig, 2019-03-19:
Das Problem ist Deine Logik- woher soll vz wissen dass in Deinen Daten ein Überlauf ist? Warum nimmst Du nciht 64 bit oder float?

emeins, 2019-03-19:
Die Fritzbox liefert die io/outBytes in dem Format - vgl. 
https://avm.de/fileadmin/user_upload/Global/Service/Schnittstellen/wancommonifconfigSCPD.pdf
Vielleicht frage ich aber auch mal den Owner der fritzapi, ob er das mit einbauen kann ;-)

andig, 2019-03-21:
Kannst Du machen, der wird Dich sicher auf ein wenig scripting verweisen ;) In jedem Fall kein Thema das Volkszaehler lösen sollte!

closed 2019-03-21 by andig