volkszaehler/volkszaehler.org #750: PPM static-directory

J-A-U, 2019-03-25:
`"static-directory": "htdocs",`
in middleware.json is supposed to configure the frontend instead of
`/var/www/volkszaehler.org/vendor/bin/ppm start --static-directory /var/www/volkszaehler.org/htdocs`
?

Didn't work out, frontend doesn't get started:
```
pi@volkszaehler:~ $ /var/www/volkszaehler.org/vendor/bin/ppm start -c /var/www/volkszaehler.org/etc/middleware.json --cgi-path=/usr/bin/php                     Read configuration /home/pi/volkszaehler.org/etc/middleware.json, modified by command arguments.
/home/pi
+---------------------+-----------------------------------------+
| host                | 0.0.0.0                                 |
| port                | 8080                                    |
| workers             | 8                                       |
| static-directory    | htdocs                                  |
```

Instead of:
```
pi@volkszaehler:~ $ /var/www/volkszaehler.org/vendor/bin/ppm start -c /var/www/volkszaehler.org/etc/middleware.json --static-directory /var/www/volkszaehler.org/htdocs --cgi-path=/usr/bin/php
Read configuration /home/pi/volkszaehler.org/etc/middleware.json, modified by command arguments.
/home/pi
+---------------------+-----------------------------------------+
| host                | 0.0.0.0                                 |
| port                | 8080                                    |
| workers             | 8                                       |
| static-directory    | /var/www/volkszaehler.org/htdocs        |
```

If this Config is supposed to be changed we need a "copy" like config.yml to prevent questions related to `git pull.`

andig, 2019-03-25:
Ich verstehs nicht. Dein Eintrag `htdocs` klappt natürlich als relativer Pfad nur wenn aus dem richtigen Verzeichnis aufgerufen wird. Fürs Image sollte da besser ein absoluter Pfad drin stehen.

Meinst Du das?

Prinzipiell würde ich volkszaehler.org *nicht* unter /var/www legen da dann immer sudo erforderlich ist. Bei mir liegt das unter /home/pi. `/var/www` wäre ohne Apache ja sowieso gegenstandslos?

J-A-U, 2019-03-25:
> Meinst Du das?
Ja. Wenn ich die Datei aber ändere verweigert git das nächste Update. Für Noobs (Image) bissl ungeschickt.

> Prinzipiell würde ich volkszaehler.org nicht unter /var/www legen
Liegt es auch nicht, nutze für Aufrufe aber in aller Regel den Symlink.

andig, 2019-03-25:
Dann wär die Frage nach einem allgemeingültigen Defaultpfad.

J-A-U, 2019-03-25:
Ausgehend vom Wiki wäre das /var/www/volkszaehler.org/htdocs

Fürs Image hab ich aber auch den Port auf 80 angepasst. Ist ja keine Apache mehr da ist mit dem der PPM kolidieren kann.

andig, 2019-04-02:
> Fürs Image hab ich aber auch den Port auf 80 angepasst. 

Damit musst Du jetzt doch aber weiter in der Datei ändern- Problem also noch da? Alternativ könntest Du beide Parameter einfach auf der Kommandozeile im Systemd Service angeben?

Sonst gerne PR, mir ist jede Variante recht.

closed 2019-04-24 by andig