volkszaehler/volkszaehler.org #751: Push-Server rejects vzlogger

J-A-U, 2019-03-25:
Push-Server seams so reject data delivered by vzlogger.

push-server.service:
```
Mär 25 09:16:58 volkszaehler systemd[1]: Started push-server.
Mär 25 09:16:59 volkszaehler php[3026]: Volkszaehler Push Server
Mär 25 09:16:59 volkszaehler php[3026]: Listening for updates on 5582. Clients m
Mär 25 09:17:54 volkszaehler php[3026]: Invalid channel 7a39d9a7***
Mär 25 09:17:54 volkszaehler php[3026]: Invalid channel 7a39d9a7***
Mär 25 09:17:54 volkszaehler php[3026]: Invalid channel 12f04bc0***
Mär 25 09:17:54 volkszaehler php[3026]: Invalid channel 6c2e2b10***
Mär 25 09:17:55 volkszaehler php[3026]: Invalid channel 7a39d9a7***
Mär 25 09:17:56 volkszaehler php[3026]: Invalid channel 12f04bc0***
[...]
```

vzlogger.log:
```
[Mar 25 09:17:51][main] vzlogger v0.7.0 based on heads/master-0-g79402aeb96 from Wed, 30 Jan 2019 15:56:19 +0100 started.
[Mar 25 09:17:51][mtr0] Creating new meter with protocol s0.
[Mar 25 09:17:51][mtr0] Meter configured, disabled.
[Mar 25 09:17:51]       New meter initialized (protocol=s0)
[Mar 25 09:17:51]       Configure channel.
[Mar 25 09:17:51][chn0] New channel initialized (uuid=*** api=volkszaehler id=Impulse)
[Mar 25 09:17:51][mtr1] Creating new meter with protocol sml.
[Mar 25 09:17:51][mtr1] Meter configured, enabled.
[Mar 25 09:17:51]       New meter initialized (protocol=sml)
[Mar 25 09:17:51]       Configure channel.
[Mar 25 09:17:51][chn1] New channel initialized (uuid=*** api=volkszaehler id=1-0:2.8.0)
[Mar 25 09:17:51][mtr2] Creating new meter with protocol sml.
[Mar 25 09:17:51][mtr2] Meter configured, enabled.
[Mar 25 09:17:51]       New meter initialized (protocol=sml)
[Mar 25 09:17:51]       Configure channel.
[Mar 25 09:17:51][chn2] New channel initialized (uuid=*** api=volkszaehler id=1-0:1.8.0)
[Mar 25 09:17:51]       Configure channel.
[Mar 25 09:17:51][chn3] New channel initialized (uuid=*** api=volkszaehler id=1-0:2.8.0)
[Mar 25 09:17:51]       Have 3 meters.
[Mar 25 09:17:51][main] log level is 15
[Mar 25 09:17:51][main] daemon=1, local=0
[Mar 25 09:17:51]       Daemonize process...
[Mar 25 09:17:51]       Opened logfile /var/log/vzlogger.log
[Mar 25 09:17:51][push] pushdata_thread created.
[Mar 25 09:17:51][]     ===> Start meters
[Mar 25 09:17:51][mtr0] Meter for protocol 's0' is disabled. Skipping.
[Mar 25 09:17:51][push] Start push_data_thread
[Mar 25 09:17:51][mtr1] Meter connection established
[Mar 25 09:17:51][mtr1] Meter thread started
[Mar 25 09:17:51][mtr1] Meter is opened. Starting channels.
[Mar 25 09:17:51][chn1] Logging thread started
[Mar 25 09:17:51][chn1] Start logging thread for volkszaehler-api. Running as daemon: yes
[Mar 25 09:17:51][mtr2] Meter connection established
[Mar 25 09:17:51][mtr2] Meter thread started
[Mar 25 09:17:51][mtr2] Meter is opened. Starting channels.
[Mar 25 09:17:51][chn2] Logging thread started
[Mar 25 09:17:51][chn3] Logging thread started
[Mar 25 09:17:51][]     Startup done.
[Mar 25 09:17:51][mtr1] Number of readers: 32
[Mar 25 09:17:51][mtr1] Config.daemon: 1
[Mar 25 09:17:51][mtr1] Config.local: 0
[Mar 25 09:17:51][chn1] Using default volkszaehler api.
[Mar 25 09:17:51][chn3] Start logging thread for volkszaehler-api. Running as daemon: yes
[Mar 25 09:17:51][chn3] Using default volkszaehler api.
[Mar 25 09:17:51][chn2] Start logging thread for volkszaehler-api. Running as daemon: yes
[Mar 25 09:17:51][mtr2] Number of readers: 32
[Mar 25 09:17:51][mtr2] Config.daemon: 1
[Mar 25 09:17:51][mtr2] Config.local: 0
[Mar 25 09:17:51][chn2] Using default volkszaehler api.
[Mar 25 09:17:52][mtr1] Got 4 new readings from meter:
[Mar 25 09:17:52][mtr1] Reading: id=1-0:2.8.0*255/ObisIdentifier:1-0:2.8.0*255 value=.70 ts=1553501872826
[Mar 25 09:17:52][mtr1] Reading: id=1-0:2.8.1*255/ObisIdentifier:1-0:2.8.1*255 value=.10 ts=1553501872826
[Mar 25 09:17:52][mtr1] Reading: id=1-0:2.8.2*255/ObisIdentifier:1-0:2.8.2*255 value=.60 ts=1553501872827
[Mar 25 09:17:52][mtr1] Reading: id=1-0:16.7.0*255/ObisIdentifier:1-0:16.7.0*255 value=-.00 ts=1553501872827
[Mar 25 09:17:52][chn1] Adding reading to queue (value=.70 ts=1553501872826)
[Mar 25 09:17:52][push] push: { "data": [ { "uuid": "7a39d9a7***", "tuples": [ [ 1553501872826, .70000000001 ] ] } ] }
[Mar 25 09:17:52][push] added to uuid 7a39d9a7***
[Mar 25 09:17:53][mtr2] Got 7 new readings from meter:
[Mar 25 09:17:53][mtr2] Reading: id=1-0:1.8.0*255/ObisIdentifier:1-0:1.8.0*255 value=.20 ts=1553501873185
[Mar 25 09:17:53][mtr2] Reading: id=1-0:2.8.0*255/ObisIdentifier:1-0:2.8.0*255 value=.70 ts=1553501873185
[Mar 25 09:17:53][mtr2] Reading: id=1-0:1.8.1*255/ObisIdentifier:1-0:1.8.1*255 value=.20 ts=1553501873185
[Mar 25 09:17:53][mtr2] Reading: id=1-0:2.8.1*255/ObisIdentifier:1-0:2.8.1*255 value=.70 ts=1553501873185
[Mar 25 09:17:53][mtr2] Reading: id=1-0:1.8.2*255/ObisIdentifier:1-0:1.8.2*255 value=0.00 ts=1553501873185
[Mar 25 09:17:53][mtr2] Reading: id=1-0:2.8.2*255/ObisIdentifier:1-0:2.8.2*255 value=0.00 ts=1553501873185
[Mar 25 09:17:53][mtr2] Reading: id=1-0:16.7.0*255/ObisIdentifier:1-0:16.7.0*255 value=-.10 ts=1553501873185
[Mar 25 09:17:53][chn2] Adding reading to queue (value=.20 ts=1553501873185)
[Mar 25 09:17:53][push] added to uuid 12f04bc0***
[Mar 25 09:17:53][chn3] Adding reading to queue (value=.70 ts=1553501873185)
[Mar 25 09:17:53][push] added to uuid 6c2e2b10***
[Mar 25 09:17:54][mtr1] Got 4 new readings from meter:
[Mar 25 09:17:54][mtr1] Reading: id=1-0:2.8.0*255/ObisIdentifier:1-0:2.8.0*255 value=.50 ts=1553501874248
[Mar 25 09:17:54][mtr1] Reading: id=1-0:2.8.1*255/ObisIdentifier:1-0:2.8.1*255 value=.90 ts=1553501874248
[Mar 25 09:17:54][mtr1] Reading: id=1-0:2.8.2*255/ObisIdentifier:1-0:2.8.2*255 value=.60 ts=1553501874248
[Mar 25 09:17:54][mtr1] Reading: id=1-0:16.7.0*255/ObisIdentifier:1-0:16.7.0*255 value=-.60 ts=1553501874248
[Mar 25 09:17:54][chn1] Adding reading to queue (value=.50 ts=1553501874248)
[Mar 25 09:17:54][push] added to uuid 7a39d9a7***
[Mar 25 09:17:54][push] CURL Error from url http://127.0.0.1:5582: 500 Error 500: Internal Server Error
[Mar 25 09:17:54][push] send nok to url http://127.0.0.1:5582
[Mar 25 09:17:54][push] push: { "data": [ { "uuid": "7a39d9a7***", "tuples": [ [ 1553501874248, ***.5 ] ] }, { "uuid": "6c2e2b10-***", "tuples": [ [ 1553501873185, ***.700000003 ] ] }, { "uuid": "12f04bc0***", "tuples": [ [ 1553501873185, ***.2000000002 ] ] } ] }
[Mar 25 09:17:54][push] CURL Error from url http://127.0.0.1:5582: 500 Error 500: Internal Server Error
[Mar 25 09:17:54][push] send nok to url http://127.0.0.1:5582
[Mar 25 09:17:54][mtr2] Got 7 new readings from meter:
[...]
```

Something need to be configured?

andig, 2019-03-26:
Mhhm. Can you run push-server in foreground and see if there's any error message?

J-A-U, 2019-03-26:
I'm wasn't able to reproduce the failure. :/ 
Started push-server in foreground, no errors and a short blimp of push-data at frontend.
Tried the service again, no push-data and no errors. I discovered the automatic refresh "overwrites" the push-data, disabled it: working fine.

closed 2019-03-26 by J-A-U