volkszaehler/volkszaehler.org #760: push server is not enabled in docker image

bgewehr, 2019-04-28:
Could you please explain the best way how to persistently enable push server on the docker image?

Please help a docker newbie...

andig, 2019-04-28:
Please check if https://github.com/volkszaehler/volkszaehler.org/pull/761 works for you

bgewehr, 2019-04-28:
Used the fresh image from volkszaehler/volkszaehler now.

Setup in config.yaml instead of php -> OK.
Error in frontend:

```
api/capabilities/definitions.json?unique=1556482801859:

Unknown context: 'api'
```

Google gives no hint on this one.

bgewehr, 2019-04-28:
Docker container protocol tells:

```
8 workers (starting at 5501) up and ready. Application is ready at http://0.0.0.0:8080/ | stdout
Script did not return a valid HTTP response. Maybe it has called exit() prematurely? | stdout
```

andig, 2019-04-28:
> Setup in config.yaml instead of php -> OK.

Please update image and retry (it's currently building and will hopefully publish).

bgewehr, 2019-04-29:
Volkszaehler/volkszaehler docker image now runs without errors, but still without active push-server, correct?

Started push server manually 
`nohup push-server`

```
Volkszaehler Push Server                                                                                                       
Listening for updates on 5582. Clients may connect at 8082.  
```

Frontend seems to try to connect to push-server on port 8080 but fails
[image]

bgewehr, 2019-04-29:
Changed options.js to enable live. Error is now gone, but I do not understand, why.

bgewehr, 2019-04-29:
OK, it's not gone:
[image]

bgewehr, 2019-04-29:
Container protocol sais:

```
Restart worker #5502 because it reached max requests of 1000 | stdout -- | --
Script did not return a valid HTTP response. Maybe it has called exit() prematurely? | stdout -- | --
```

andig, 2019-04-29:
What is your setup? 

Do you realize that middleware and push server are two separate containers, i.e. are NOT sharing the same IP address?

Viele Grüße,
Andreas

bgewehr, 2019-04-29:
I guess I did not yet understand how to set the tasks for a specific volkszaehler container. Is this done with an env variable? Can I create one with push-server only and another one with middleware and frontend? (and does that make sense to split? It would be still one service (=VZ) if pushserver would run inside of vz image, or do I get this wrong?)

(fyi: using docker on synology is a little bit less command line than typical docker is)

andig, 2019-04-29:
Take a look at the included docker-compose. 
On Synology you can still use docker on the command line and I really recommend to do so or it will get very painful. 

Viele Grüße,
Andreas

bgewehr, 2019-04-29:
OK, I now have two VZ-docker containers an I use the mariadb 5 from the synology services as my db.

[image]

The push Server tells he's working
`Volkszaehler Push Server`
`Listening for updates on 5582. Clients may connect at 8082.`

And the frontend / middleware does work too.

BUT:
1) Updates are way slower than it was on the pi with local middleware, now far below once a second (does docker perform so much worse than a local middleware on the pi?)
2) Changed lines that come from other sources than push service (I use node-red too to send data to the middleware) will not update their lines in the chart until the complete page is reloaded or that channel gets hidden and shown by switching the checkbox in the channels list)

Is this the expected behaviour?

bgewehr, 2019-04-29:
Well, sorry for the iterative self-responding, it shows my learning curve.

I guess 2) is why a push server actually exists and my node-red values do not trigger frontend updates without addressing the push server. I will try to send to the push service also from node-red.

bgewehr, 2019-05-01:
This works now. Thanks for helping.

closed 2019-05-01 by bgewehr