volkszaehler/volkszaehler.org #768: VZlogger ist running, after reboot System no channels

HomeAutoUser, 2019-06-27:
Hallo,
ich befasse und mich nun mit dem VZ und habe schon mehrfach eine Erkenntnis gesammelt welche nicht eindeutig nachvollziehbar ist.

Es werden Channels eingestellt (UUID´s) welche auch angezeigt werden und Options speichern gedrückt. Nach sporadischen reboots vom Pi und erneutem Zugriff auf die Oberfläche sind diese dann weg. Ernst nach einem Neueinrichten läuft das ganze wieder.

MfG

frankrichter, 2019-06-27:
Was heißt denn "Channels eingestellt"? Neu erstellt? Abboniert?

HomeAutoUser, 2019-06-27:
>Was heißt denn "Channels eingestellt"? Neu erstellt? Abboniert?
Die Channels wurden im Fonted neu erstellt und ebenso auch in der Config eingetragen, also abonniert.
Der Dämon wird gestartet und ich sehe Werte im Diagramm (Fonted).

Weil ich am Pi arbeite, starte ich ihn gelegendlich neu und dann wollte ich mal darauf schauen wie es aussieht der Ist-Zustand und schwupps, waren die Channels im Fonted "aufgelöst".

frankrichter, 2019-06-27:
Anwenderfragen gehören eigentlich nicht in GitHub-Issues. Bitte zukünftig
die Mailingliste nutzen!

Abonnieren heißt lediglich einen bereits vorhandenen Kanal im Frontend
hinzuzufügen, entweder mit Hilfe der bekannten UUID oder mittels Drop-down,
falls der Kanal auf öffentlich gesetzt wurde. Mit der vzlogger.conf hat das
alles nichts zu tun.

Die abonnierten Kanäle werden in Form eines Cookies gespeichert. Setzt du
den Haken für dieses Cookie? Speichert dein Browser Cookies über die
Session hinaus?

closed 2019-07-21 by andig