volkszaehler/volkszaehler.org #772: Datenaggregation (bei Leistungswerten) fehlerhaft

JimRockford1, 2019-07-21:
Bei mir werden die Tagesverbräuche sowie der Gesamtverbrauch nicht korrekt dargestellt.
Aktuell erfasse ich (seit Samstag, 06.07.) mit dem Volkszähler verschiedenen Leistungswerte (!) meines Piko-Wechselrichters, die (von 5 Uhr morgens bis 22 Uhr abends) alle 30 Sekunden abgefragt werden und (über die Middleware-API) in die Datenbank geschrieben werden.
Die Tageswerte für die Energie in der Spalte "Verbrauch" werden korrekt dargestellt und stimmen mit den im Piko-Wechselrichter hinterlegten Werten (für die Energie!) überein. Siehe Anhang 1. Dies gilt für den aktuellen Tag wie auch für alle vergangenen Tage.
Wechsele ich nun im Chart auf "daily" (also Balkendiagramm) und eine wochenweise Anzeige ("Woche"), dann sind die Tageswerte im Balkendiagramm nicht korrekt, sondern geringfügig zu hoch - siehe Anhang 2., zu erkennen an der Angabe links oben neben "Wechselrichter Ausgangsleistung" (der korrekte Wert am 14.07. wäre 53,4 kWh, nicht 55,8 kWh). Lediglich der laufende Tag wird korrekt darstellt - siehe Anhang 3.
In der Folge wird auch in der Spalte "Gesamt" ein zu hoher Wert angezeigt (hier: 699 kWh statt korrekt 671 kWh). Dieser zu hohe Wert stimmt mit dem Summenzug der (zu hohen) Einzelbalken überein (habe ich nachgerechnet).
Wechsele ich auf eine Tagesanzeige ("Tag") (Balkendarstellung), dann wird der Balken korrekt dargestellt.
Übrigens: Bei den Anhängen 2 und 3 scheint die Angabe in der Spalte "Verbrauch" (mit 14,4 MWh) völlig sinnfrei zu sein.
Dieser Eintrag und die Bilder sind mit Stand 15.07.2019 geschrieben. Das Problem besteht (Stand 21.07.) fort.
Diese Beschreibung habe ich auch [https://www.photovoltaikforum.com/thread/129645-volksz%C3%A4hler-2-0/?postID=1801710#post1801710](url) hier veröffentlicht. 

Gemäß [https://www.photovoltaikforum.com/thread/129645-volksz%C3%A4hler-2-0/?postID=1801722#post1801722](url) besteht das Problem nicht bei Zählerstände-Kanälen.
[image]
[image]
[image]

r00t-, 2019-07-21:
[disclaimer: nimm nicht an, dass ich zum volkszaehler projekt gehoere]

ist dir bewusst, das das loggen eines leistungswertes alle 30 sekunden _extrem_ ungenau ist?
der wert wird ein momentanwert sein, oder zumindest gemittelt ueber irgendeinen zeitraum, der vermutlich ein anderer ist, als 30 sekunden.
daraus berechnet die middleware dann energiewerte, indem sie annimt, dass dieser leistungswert fuer 30 sekunden konstant war.

insbesondere vermute ich, dass wenn du tatsaechlich nachts zwischen 22 und 5 uhr keine daten loggst, dann der letzte leistungswert von 22 uhr fuer die ganze nacht angenommen wird, bis morgens ein neuer geloggt wird.
das koennte erklaeren, dass die werte zu hoch werden, wenn ueber tagesgrenzen hinweg aggregiert wird.

JimRockford1, 2019-07-21:
Du hast offensichtlich meinen Beitrag nicht richtig gelesen oder nicht verstanden. Schließlich ist die unter "Verbrauch" angegebene Aggregation des jeweiligen Tages korrekt und stimmt mit den Angaben des Wechselrichters überein! Nur die Aggregation in der Balkendarstellung (in der Ansicht mehrerer Tage) ist fehlerhaft und weicht ab. Die Darstellung eines (Tages-)Balkens in einer Mehr-Tages-Ansicht muss aber mit der Darstellung eines gesamten Tages unter "Verbrauch" identisch sein.
Abgesehen davon ist deine Behauptung, dass das Loggen eines Leistungswertes alle 30 Sekunden extrem ungenau ist, zu pauschal. Wenn Leistungswerte ständig stark schwanken, hast du recht. Dies trifft aber bei einer Photovoltaik-Anlage nicht zu. Hier schwanken die Werte abhängig von der Bewölkung - an einem vollständig sonnigen Tag ändern sich die Werte höchstens um wenige Watt alle 30 Sekunden, und selbst da tritt der beschriebene Fehler auf.

andig, 2019-07-21:
Moin. Ich fände einen entspannten Ton in der Diskussion gut. Hier gehts nicht drum dem Anderen zu beweisen dass er Unrecht hat sondern um gemeinsame Fehlerbehebung.

andig, 2019-07-21:
> Über Nacht logge ich die Werte deshalb nicht, da der letzte Wert um 22 Uhr "0" ist und der erste Wert um fünf Uhr auch "0" ist ... ergibt im Mittel "0".

Exakt da liegt ein möglicher Fehler. Du kannst Dich nicht drauf verlassen dass dieser Nullerwert auch verwendet wird da immer mehrere Werte aus Performancegründen zusammengefasst werden. Loggen von Nullen oder nicht wurde auf der ML x-fach diskutiert- bitte Problem dort adressieren.

Ich machs hier zu da höchstwahrscheinlich kein Bug.

JimRockford1, 2019-07-21:
Nicht nur der letzte Wert ist Null, sondern derzeit bereits beginnend ab 21:30 Uhr und erst ab ca. 05:50 Uhr liegen die Werte wieder über Null. Aber wenn es damit zu tun haben könnte (weil vielleicht die Durchschnittswerte der ersten und letzten aufgezeichneten Stunde nicht bei Null liegen und dann das von dir beschriebene Problem auftritt), dann werde ich das Logging die nächsten Tage auch mal nachts durchlaufen lassen - ich werde dann hier und im photovoltaik-forum das Ergebnis posten.
P.S.1: Was bedeutet "wurde auf der ML x-fach diskutiert"?
P.S.2: Meine o.a. Antwort sollte nicht unhöflich wirken und war auch nicht so gemeint.

andig, 2019-07-21:
> Meine o.a. Antwort sollte nicht unhöflich wirken und war auch nicht so gemeint.

Passt- dann lag ich falsch. Bitte trotzdem für so allgemeine Fehlersuche die Mailingliste anfragen- da gibt es mehr Kapazität als ich hier bereitstellen kann...

JimRockford1, 2019-07-21:
Werde ich gerne künftig beachten! Allerdings hatte ich bei "photovoltaik-forum.com" den Hinweis erhalten, zu diesem Thema hier ein Issue zu öffnen.

frankrichter, 2019-07-21:
Zumindest der Gesamtverbrauch unter dem Balkendiagramm könnte schon ein Bug sein, erwarten würde man ja die Summe der einzelnen Balken.

andig, 2019-07-23:
Ich versuche mal die Textaufgabe zusammen zu fassen. Also wenn ich es richtig verstehe:

- Verbrauch unter Linechart für einen Tag ist richtig
- Verbrauch einzelner Tagesbalken im Balkendiagramm ist richtig
- Gesamtverbrauch ist immer richtig

Falsch sind "Verbrauch" in der Tabelle der Balkendiagramme, sowohl "Durchschnitt" als auch "Verbrauch"?

Dann ist die Ursache klar: anscheinend wird im Barchart nochmals mit der Zeit ausmultipliziert, wodurch dann alle Einheiten falsch sind. Jetzt wäre die Frage was da korrekterweise rein sollte?

JimRockford1, 2019-07-25:
> Exakt da liegt ein möglicher Fehler. Du kannst Dich nicht drauf verlassen dass dieser Nullerwert auch verwendet wird da immer mehrere Werte aus Performancegründen zusammengefasst werden. 

> Aber wenn es damit zu tun haben könnte (weil vielleicht die Durchschnittswerte der ersten und letzten aufgezeichneten Stunde nicht bei Null liegen und dann das von dir beschriebene Problem [...]

Die Werte der einzelnen Tages-Balken in der Wochenanzeige passen jetzt - seitdem ich auch nachts die Nullwerte erfasse - insofern liegt hier also kein Bug vor!

Nachfolgend meine Anmerkungen zum vorherigen Post in eckigen Klammern []:

> Ich versuche mal die Textaufgabe zusammen zu fassen. Also wenn ich es richtig verstehe:
> 
> * Verbrauch unter Linechart für einen Tag ist richtig [ja!]
> * Verbrauch einzelner Tagesbalken im Balkendiagramm ist richtig [ja - wenn auch Nullwerte mitgeplottet werden]
> * Gesamtverbrauch ist immer richtig [muss ich, jetzt wo ich die Nullwerte mitplotte, nochmal überprüfen - ich gehe aber davon aus
> 
> Falsch sind "Verbrauch" in der Tabelle der Balkendiagramme, sowohl "Durchschnitt" als auch "Verbrauch"? [genau - dort stehen bei mir immer noch Werte in MWh anstelle kWh]
> 
> Dann ist die Ursache klar: anscheinend wird im Barchart nochmals mit der Zeit ausmultipliziert, wodurch dann alle Einheiten falsch sind. Jetzt wäre die Frage was da korrekterweise rein sollte? [bei Durchschnitt: keine Ahnung (Durchschnitt wovon?), bei Verbrauch: der über die Zeitperiode dargestellte summierte Verbrauch (bzw. Erzeugung)]

trabant-rgb, 2020-04-03:
I have a similar issue with a workinghourssensor... maybe caused by the same underlying problem. See
https://www.mail-archive.com/volkszaehler-users@demo.volkszaehler.org/msg10654.html
for details. I can gladly supply more information or do some testing if that would help.

trabant-rgb, 2020-05-02:
My issue has been closed by #801...

andig, 2020-05-03:
@JimRockford1 working for you, too or still wrong?