volkszaehler/volkszaehler.org #773: Graph wird nur mit Lücken angezeigt

p3ab0dy, 2019-07-31:
Hi ich frage über ein Script einen Aquara Temperatur Sensor an. Neben der Temperatur erhalte ich auch einen Zeitstempel. Diesen rechne ich in ms um und schicken ihn an die API. Das Skript läuft alle 5 Minuten. Hat sich ein Wert nicht geändert würde das Skript versuchen diesen mit dem gleichen Zeitstempel einzutragen und die API würden diesen korrekt abweisen. 

Nun wollte ich vor kurzem ein Update auf die neue Version durchführen aber leider wird mir der Graph nicht mehr durchgehend angezeigt. 

Graph in alter Version:
`git describe --tags 0.7-34-g341bf0f `
[image]

`git describe --tags 1.0-54-g843fb56`
[image]

Habt ihr eine Idee woran das liegen könnte? 
Ich habe die aktuelle Version im Moment parallel installiert und an die gleiche Datenbank angeschlossen. Also die Datenbasis ist bei beiden Instanzen identisch.

frankrichter, 2019-07-31:
Schau mal in die options.js, da gibt's irgendwo eine Option "gap". Kannst
du auskommentieren oder die Zeit verlängern.

p3ab0dy, 2019-07-31:
Perfekt das war der Fehler!!! Vielen Dank!!!

closed 2019-07-31 by p3ab0dy