volkszaehler/volkszaehler.org #780: Skalierung bei Stundendarstellung stimmt nicht

BoxSo, 2019-11-12:
Wenn ich mir die Daten meines D0 (SML) Sensor am eZH darstellen lassen, scheinen die Werte initial zu stimmen. Die historischen Daten werden korrekt dargestellt. Die ab dem Zeitpunkt des Ladens der Seite gesammelten (nachgeladenen) Daten werden dann wohl durch 10 geteilt. 

Wenn ich dann wieder zwischen Tages- und Stundendarstellung wechsle, werden wieder bis zum Zeitpunkt des ladens der Seite alle Werte korrekt dargestellt. 

Es sieht so aus, als ob die Funktion, die die Daten kontinuierlich nachlädt, eine Division durch 10 durchführt. Auch der aktuelle Leistungswert unten in der Liste der Kanäle zeigt einen ggü. dem realen Verbraucht um den Faktor 10 geringeren Wert an (also z.B. statt 2.3 KW -> 230 Watt).

[image]
[image]

andig, 2019-11-12:
Wie ist der Kanal konfiguriert?

BoxSo, 2019-11-12:
Folgend die Konfiguration in der vzlogger.conf Datei
[image]
[image]
Und hier die Einstellung des Channels:
[image]
Und das ist der zugehörige Zähler:
[image]

frankrichter, 2019-11-12:
Hat deine vzlogger.conf wirklich keinen "push"-Eintrag?

BoxSo, 2019-11-12:
oh doch ... hatte einen Screenshot übersehen beim hochladen:
[image]

frankrichter, 2019-11-12:
Ok, ansonsten wäre es etwas mysteriös gewesen, wo die beschriebenen
automatischen Updates herkommen. Die Konfiguration sieht für mich ok aus,
eine Erklärung für die falsch berechnete Leistung hab ich leider nicht
parat.

J-A-U, 2019-11-13:
Erklärung hab ich auch nicht.
Da aber eh keine Aggregation genutzt wird würde ich den PUSH-Server mal deaktivieren um zu sehen ob das Problem weg geht. Wobei man den nicht nur im vzlogger abstellen kann, in der Middleware ist auch konfiguriert wo die Daten angenommen werden.

frankrichter, 2019-11-13:
sudo systemctl stop push-server genügt, um kurz zu prüfen ob die falschen
Werte vom Push-Server kommen. Der Beschreibung nach gehe ich mal davon aus.
Längerfristig muss allerdings der Eintrag in der vzlogger.conf raus, sonst
läuft das Logfile voll.

BoxSo, 2019-11-13:
Nach dem Beenden des push-server über sudo systemctl stop push-server stimmt die Anzeige des Charts und auch der Werte unterhalb des Chart. 

Ich nehme dann die push-Konfiguration aus der vzlogger.conf raus?
Hat das sonst irgendwelche Nebenwirkungen?

frankrichter, 2019-11-13:
Du verlierst halt den Komfort, den der Push-Server bietet (laufende
Frontend-Updates an der Datenbank vorbei, näheres siehe Wiki).

Wenn der Push-Server falsche Leistungswerte berechnet, ist das ein Bug und
sollte näher untersucht werden - normalerweise sollte der mit der Auflösung
aus den Kanaleigenschaften arbeiten und korrekte Werte liefern.

>

BoxSo, 2019-11-13:
Ich habe nun mittels sudo systemctl start push-server den Push-Service wieder gestartet und nun zeigt die Oberfläche korrekte Werte an (egal ob auf Stunden-Auflösung, Tag oder Woche). 

Ich hatte den Raspberry schon mehrfach neu gestartet zwischendrin.

Kann es sein, dass der push-server Änderungen am der Auflösung nicht automatisch mitbekommt?

frankrichter, 2019-11-13:
Vielleicht ein Caching-Problem - gut dass es jetzt klappt! Dann solltest du
wie von J-A-U angedeutet vielleicht noch überlegen, ob du nicht die
Aggregation im vzlogger nutzen willst, aktuell sammelst du sehr viele
Datensätze, die die Speicherkarte stressen und wahrscheinlich nicht in der
Auflösung (1-3s) gebraucht werden.

Sorry für's OT: Was für einen Lesekopf hast du da im Einsatz? Ist der per
WLAN angebunden?

>

J-A-U, 2019-11-13:
Das ist der Weidmann Lesekopf mit USB. Liegt bei ca. 45€.

frankrichter, 2019-11-13:
Okay, danke. Bin über das "host" in der vzlogger.conf gestolpert und hab
deswegen einen per Netzwerk angebundenen Lesekopf erwartet.

BoxSo, 2019-11-13:
> Vielleicht ein Caching-Problem - gut dass es jetzt klappt! Dann solltest du wie von J-A-U angedeutet vielleicht noch überlegen, ob du nicht die Aggregation im vzlogger nutzen willst, aktuell [...]

Gegen ein Caching-Problem spricht eigentlich, dass das Problem auf mehreren Clients zu beobachten war (Smartphone, Macbook, Tablet).

Die Aggregation werde ich mir zeitnah ansehen, Danke für den Tipp! 

Und ja genau, das ist ein Lesekopf von Weidmann.

andig, 2019-11-14:
Wenns so ist bitte Ticket aufmachen einschl. Kanalkonfiguration.

Viele Grüße, 
Andreas

andig, 2020-02-26:
Geht es denn jetzt? Dann könnten wir hier zumachen...

closed 2020-08-18 by andig