volkszaehler/volkszaehler.org #781: Loading VZ in a hidden iFrame gives a flot error

l3d00m, 2019-11-30:
Loading volkzszähler in a hidden iFrame/div (which happens for me when using ioBroker with vis because it has some ugly loading routine) gives the following error "Error: Invalid dimensions for plot, width = 0, height = 0" in flot.min.js:2:3666. 

I found this https://stackoverflow.com/a/22615279/4026792 which was able to solve my issue. I added `width: 100%;` to the #flot div in style.css and manually included `<script type="text/javascript" src="vendor/flot/jquery.flot.resize.js"></script>` in index.html.

I would propose to add this to volkszähler for people who are also including this as an iobroker iframe or similiar. I haven't opended a PR because I wasn't sure whether this is of interest and I also didn't know how to minify the flot.min.js file with the included resize addon because it seems your using a custom script for that.

l3d00m, 2019-11-30:
After playing around with it some more it seems not to fix the issue reliably, not sure why it fixed the issue a few minutes ago and now not any longer.

andig, 2020-02-26:
Tentatively closing.

closed 2020-02-26 by andig