volkszaehler/volkszaehler.org #782: Frontend Darstellungslücken im Liniendiagramm

Clautens, 2019-12-10:
Hallo zusammen!
ich bekomme im Frontend nicht zusammenhängende Linien zu sehen (Die Beriche zwischen zwei Datenpunkten sind nicht verbunden):
[image]

Die Bereiche sind manchmal auch größer und obwohl Datenpunkte vorhanden sind, wird die Linie nicht weitergeführt:
[image]

Version: Date:   Mon Nov 25 10:34:37 2019 +0100
Verhalten taucht bei verschiedenen Browsern auf.

Viele Grüße!
Karl

frankrichter, 2019-12-10:
Schau mal hier:
https://github.com/volkszaehler/volkszaehler.org/blob/4c11705638ce0daeb6c56f3de95d173299ecd9f0/htdocs/js/options.js#L57

Clautens, 2019-12-10:
passt!
Vielen Dank :-)

closed 2019-12-10 by Clautens