volkszaehler/volkszaehler.org #786: Help Wanted : Integration of middle ware in synology docker setup

AbDhops, 2019-12-21:
Hi All,
Need help.
Since I have found out the possibility of using volkszaehler, I am really exited and desperate to get it up and running. 
So for my smart meter Iskra MT681, I already have IR reader. I have installed volkszaehler, volkszaehler-push service and maria db databased. All in synology docker. I have verified that database is generated, and volkszaehler front end is also working on Port 8085. volkszaehler push service is on 8082.

Now I guess last piece of puzzle is to read data from IR reader. And for that I assume I need to make middleware (vzlogger) up and running. And there I am totally lost !
I have tried to make it in the docker, but its always crashing. Due to which I am not able to read any data from USB reader to volkszaehler...

Is anyone using docker setup. Any help would be highly appreciated to get this up and running... 

For better understanding of my setup, my compose file is also here with..

Many thanks.

```yaml
version: '3'
services:

  database:
    image: mariadb
    environment:
      MYSQL_DATABASE: volkszaehler
      MYSQL_USER: vz
      MYSQL_PASSWORD: demo
      MYSQL_ROOT_PASSWORD: admin_demo
    volumes:
       - './mariadb:/var/lib/mysql'
    networks:
    - volkszaehler

  volkszaehler:
    image: volkszaehler/volkszaehler
    ports:
    - 8085:8080
    # NOTE: The init command will only work once the database is fully up and running.
    #       By putting restart: always docker-compose will retry starting this container
    #       and finally complete the database setup sequence.
    command: sh -c "
      sed -i s/127.0.0.1/database/ /vz/etc/config.yaml &&
      sed -i s/vz_admin/root/ /vz/etc/config.yaml &&
      sleep 1 &&
      (/vz/bin/doctrine orm:schema-tool:update --force || /vz/bin/doctrine orm:schema-tool:create) &&
      /vz/vendor/bin/ppm start -c /vz/etc/middleware.json --static-directory /vz/htdocs --cgi-path=/usr/bin/php"
    links:
    - database
    depends_on:
    - database
    networks:
    - volkszaehler
    restart: always

  push-server:
    image: volkszaehler/volkszaehler
    ports:
    - 5582:5582
    - 8082:8082
    command: sh -c "
      sed -i s/127.0.0.1/database/ /vz/etc/config.yaml &&
      /vz/bin/push-server"
    links:
    - database
    depends_on:
    - database
    networks:
    - volkszaehler
    restart: always

  vzlogger:
    image: treban/smartmeter-vzlogger
    container_name: vzlogger
    volumes:
       - './configs/:/cfg'
       - /etc/localtime:/etc/localtime:ro
    devices:
       - "/dev/ttyUSB0"
    networks:
    - volkszaehler
    restart: always

networks:
  volkszaehler:
```

andig, 2019-12-21:
Middleware and vzlogger are two totally separate things. vzlogger is not part of the middleware docker image and currently not available in a dockerized version.

Cheers,
Andreas

AbDhops, 2019-12-21:
Thanks @andig for quick response. 
May be more stupid questions.
1. Then is it part of volkszaehler docker image.... or I will have to install middle ware separately. 
2. Who is suppose to communicate with USB IR ?

AbDhops, 2019-12-21:
I guess, vzlogger shall read the data from USB, and report it to Middleware . Middleware process (decode) the data and put it into database. And from there volkszaehler frontend shall "show" it.... If this is right, then push-server is Middleware from volkszaehler docker.

Does it makes sense...

andig, 2019-12-21:
Exactly, all correct. For howto questions please contact the mailing list.

closed 2019-12-21 by andig