volkszaehler/volkszaehler.org #795: Problem with daily bar mode scale when specifying a max for one channel in options.js

trabant-rgb, 2020-02-08:
In options.js, I have defined several channels, the m³ scale with a "max:" statement. This works fine in the frontend's "current" mode. But when switching to "daily" mode this "max: 2" definition also affects other channels.

First picture - correct scaling without max:2 for m³. Second picture - max enabled. See channel definitions here: [options.txt](https://github.com/volkszaehler/volkszaehler.org/files/4173986/options.txt)

[image]
[image]

trabant-rgb, 2020-05-11:
I traced that error back to the function `ensureAavailableAxis()` in _entity.js_...
https://github.com/volkszaehler/volkszaehler.org/blob/master/htdocs/js/entity.js#L122-L130

When no predefined y-axis is available it creates a new one which is cloned from the second one defined in _options.js_ (which is [1] in the array and just by chance is the one with the max setting in my case).

So if I add a `delete vz.options.plot.yaxes[length].max` to that function or choose another reference entry, the problem is gone... however neither of these workarounds seems to be a real solution suitable for other users with other definitions in their options.js.

So I created a `yaxis_template` entry in `options.js` and am referring to that one from `ensureAavailableAxis()`, which seems to work fine. Let me know what you think... so I know if that is worth a PR?

andig, 2020-05-11:
I would appreciate a PR without an additional config setting. Cleaning min/max on the cloned axis sounds good.

trabant-rgb, 2020-05-11:
While I still like the idea to make the automatically generated axes configurable, I see the point that an addt'l config setting will make things more complex and may break existing configurations.

Two more ideas: 

- Instead of using the second configured axis as template, use the last configured axis as `vz.options.plot.yaxes[last]`, which would be in line with the documented behaviour ("last entry contains template"). In addition, add a commend about this in options.js.

- Alternatively, don't use a template, but define only the minimum required settings directly in entity.js... as I see it, that would be `position: 'left'` or `'right'`, as well as the `tickFormatter`.

Or, a mix of both ones... import left or right from the last defined channel, and ignore all other settings but the tick formatter.

andig, 2020-05-11:
Very good- I like the idea of using the last axis. A PR on that basis would be great.

closed 2020-05-13 by andig