volkszaehler/volkszaehler.org #796: Raspberry Pi Image Download is dead

Bingo2023, 2020-02-14:
Hallo,

Ist eigentlich kein richtiges Issue, ich weiss aber nicht wo ich es sonst schreiben kann.

Der Download auf der Hauptseite geht nicht mehr:
https://wiki.volkszaehler.org/howto/raspberry_pi_image
https://demo.volkszaehler.org/downloads/volkszaehler_latest.zip

Kann das einger neu uploaden? Ich mag mir das Image nochmal ziehen.

justinotherguy, 2020-02-14:
Danke für den Hinweis - das Image ist versehentlich einer übereifrigen Aufräumaktion zum Opfer gefallen - sorry!

Das Image wird so bald als möglich wieder bereit gestellt (vrs. noch heute).

Sorry für die Umstände!

justinotherguy, 2020-02-14:
@Bingo2023 bitte testen - sollte behoben sein

closed 2020-02-14 by justinotherguy

Bingo2023, 2020-02-14:
geht, TOP.
vielen Dank für die schnelle Hilfe.

AndyO65, 2021-03-14:
Hallo volkszaehler.org,
seit ich das Projekt volkszaehler.org gefunden haben, ist das Projekt fast nicht erreichbar. 
Liegt ein Serverproblem vor oder wird die Seite nur extrem heufig aufgerufen?
Grüße
Andreas

J-A-U, 2021-03-14:
Am Server wird gearbeitet, bitte etwas Geduld.