volkszaehler/volkszaehler.org #798: Doctrine error: MariaDB new installation

Sineos, 2020-02-26:
I'm trying to install VZ on a new server:

OS: Ubuntu 19.10
DB: 5.5.5-10.3.22-MariaDB-0ubuntu0.19.10.1-log
PHP: 7.3.11-0ubuntu0.19.10.3

PHP PDO is installed and working:
```
# php --ri pdo

PDO

PDO support => enabled
PDO drivers => mysql, pgsql, sqlite
```

When calling `php bin/doctrine -v orm:schema-tool:create` the script errors out with an unspecific error:

<details>
<summary>Doctrine Error</summary>

```
# php bin/doctrine -v orm:schema-tool:create

In AbstractMySQLDriver.php line 106:

  [Doctrine\DBAL\Exception\DriverException]
  An exception occurred in driver: SQLSTATE[HY000] [2006] MySQL server has gone away

Exception trace:
  at /var/www/volkszaehler.org/vendor/doctrine/dbal/lib/Doctrine/DBAL/Driver/AbstractMySQLDriver.php:106
 Doctrine\DBAL\Driver\AbstractMySQLDriver->convertException() at /var/www/volkszaehler.org/vendor/doctrine/dbal/lib/Doctrine/DBAL/DBALException.php:166
 Doctrine\DBAL\DBALException::wrapException() at /var/www/volkszaehler.org/vendor/doctrine/dbal/lib/Doctrine/DBAL/DBALException.php:154
 Doctrine\DBAL\DBALException::driverException() at /var/www/volkszaehler.org/vendor/doctrine/dbal/lib/Doctrine/DBAL/Driver/PDOMySql/Driver.php:28
 Doctrine\DBAL\Driver\PDOMySql\Driver->connect() at /var/www/volkszaehler.org/vendor/doctrine/dbal/lib/Doctrine/DBAL/Connection.php:362
 Doctrine\DBAL\Connection->connect() at /var/www/volkszaehler.org/vendor/doctrine/dbal/lib/Doctrine/DBAL/Connection.php:428
 Doctrine\DBAL\Connection->getDatabasePlatformVersion() at /var/www/volkszaehler.org/vendor/doctrine/dbal/lib/Doctrine/DBAL/Connection.php:388
 Doctrine\DBAL\Connection->detectDatabasePlatform() at /var/www/volkszaehler.org/vendor/doctrine/dbal/lib/Doctrine/DBAL/Connection.php:330
 Doctrine\DBAL\Connection->getDatabasePlatform() at /var/www/volkszaehler.org/vendor/doctrine/orm/lib/Doctrine/ORM/Mapping/ClassMetadataFactory.php:792
 Doctrine\ORM\Mapping\ClassMetadataFactory->getTargetPlatform() at /var/www/volkszaehler.org/vendor/doctrine/orm/lib/Doctrine/ORM/Mapping/ClassMetadataFactory.php:635
 Doctrine\ORM\Mapping\ClassMetadataFactory->completeIdGeneratorMapping() at /var/www/volkszaehler.org/vendor/doctrine/orm/lib/Doctrine/ORM/Mapping/ClassMetadataFactory.php:162
 Doctrine\ORM\Mapping\ClassMetadataFactory->doLoadMetadata() at /var/www/volkszaehler.org/vendor/doctrine/persistence/lib/Doctrine/Persistence/Mapping/AbstractClassMetadataFactory.php:306
 Doctrine\Persistence\Mapping\AbstractClassMetadataFactory->loadMetadata() at /var/www/volkszaehler.org/vendor/doctrine/orm/lib/Doctrine/ORM/Mapping/ClassMetadataFactory.php:78
 Doctrine\ORM\Mapping\ClassMetadataFactory->loadMetadata() at /var/www/volkszaehler.org/vendor/doctrine/persistence/lib/Doctrine/Persistence/Mapping/AbstractClassMetadataFactory.php:185
 Doctrine\Persistence\Mapping\AbstractClassMetadataFactory->getMetadataFor() at /var/www/volkszaehler.org/vendor/doctrine/persistence/lib/Doctrine/Persistence/Mapping/AbstractClassMetadataFactory.php:91
 Doctrine\Persistence\Mapping\AbstractClassMetadataFactory->getAllMetadata() at /var/www/volkszaehler.org/vendor/doctrine/orm/lib/Doctrine/ORM/Tools/Console/Command/SchemaTool/AbstractCommand.php:62
 Doctrine\ORM\Tools\Console\Command\SchemaTool\AbstractCommand->execute() at /var/www/volkszaehler.org/vendor/symfony/console/Command/Command.php:255
 Symfony\Component\Console\Command\Command->run() at /var/www/volkszaehler.org/vendor/symfony/console/Application.php:1012
 Symfony\Component\Console\Application->doRunCommand() at /var/www/volkszaehler.org/vendor/symfony/console/Application.php:272
 Symfony\Component\Console\Application->doRun() at /var/www/volkszaehler.org/vendor/symfony/console/Application.php:148
 Symfony\Component\Console\Application->run() at /var/www/volkszaehler.org/vendor/doctrine/orm/lib/Doctrine/ORM/Tools/Console/ConsoleRunner.php:63
 Doctrine\ORM\Tools\Console\ConsoleRunner::run() at /var/www/volkszaehler.org/bin/doctrine:38

In PDOConnection.php line 31:

  [Doctrine\DBAL\Driver\PDOException (2006)]
  SQLSTATE[HY000] [2006] MySQL server has gone away

Exception trace:
  at /var/www/volkszaehler.org/vendor/doctrine/dbal/lib/Doctrine/DBAL/Driver/PDOConnection.php:31
 Doctrine\DBAL\Driver\PDOConnection->__construct() at /var/www/volkszaehler.org/vendor/doctrine/dbal/lib/Doctrine/DBAL/Driver/PDOMySql/Driver.php:22
 Doctrine\DBAL\Driver\PDOMySql\Driver->connect() at /var/www/volkszaehler.org/vendor/doctrine/dbal/lib/Doctrine/DBAL/Connection.php:362
 Doctrine\DBAL\Connection->connect() at /var/www/volkszaehler.org/vendor/doctrine/dbal/lib/Doctrine/DBAL/Connection.php:428
 Doctrine\DBAL\Connection->getDatabasePlatformVersion() at /var/www/volkszaehler.org/vendor/doctrine/dbal/lib/Doctrine/DBAL/Connection.php:388
 Doctrine\DBAL\Connection->detectDatabasePlatform() at /var/www/volkszaehler.org/vendor/doctrine/dbal/lib/Doctrine/DBAL/Connection.php:330
 Doctrine\DBAL\Connection->getDatabasePlatform() at /var/www/volkszaehler.org/vendor/doctrine/orm/lib/Doctrine/ORM/Mapping/ClassMetadataFactory.php:792
 Doctrine\ORM\Mapping\ClassMetadataFactory->getTargetPlatform() at /var/www/volkszaehler.org/vendor/doctrine/orm/lib/Doctrine/ORM/Mapping/ClassMetadataFactory.php:635
 Doctrine\ORM\Mapping\ClassMetadataFactory->completeIdGeneratorMapping() at /var/www/volkszaehler.org/vendor/doctrine/orm/lib/Doctrine/ORM/Mapping/ClassMetadataFactory.php:162
 Doctrine\ORM\Mapping\ClassMetadataFactory->doLoadMetadata() at /var/www/volkszaehler.org/vendor/doctrine/persistence/lib/Doctrine/Persistence/Mapping/AbstractClassMetadataFactory.php:306
 Doctrine\Persistence\Mapping\AbstractClassMetadataFactory->loadMetadata() at /var/www/volkszaehler.org/vendor/doctrine/orm/lib/Doctrine/ORM/Mapping/ClassMetadataFactory.php:78
[... 51 lines cut ...]
 Doctrine\DBAL\Connection->getDatabasePlatform() at /var/www/volkszaehler.org/vendor/doctrine/orm/lib/Doctrine/ORM/Mapping/ClassMetadataFactory.php:792
 Doctrine\ORM\Mapping\ClassMetadataFactory->getTargetPlatform() at /var/www/volkszaehler.org/vendor/doctrine/orm/lib/Doctrine/ORM/Mapping/ClassMetadataFactory.php:635
 Doctrine\ORM\Mapping\ClassMetadataFactory->completeIdGeneratorMapping() at /var/www/volkszaehler.org/vendor/doctrine/orm/lib/Doctrine/ORM/Mapping/ClassMetadataFactory.php:162
 Doctrine\ORM\Mapping\ClassMetadataFactory->doLoadMetadata() at /var/www/volkszaehler.org/vendor/doctrine/persistence/lib/Doctrine/Persistence/Mapping/AbstractClassMetadataFactory.php:306
 Doctrine\Persistence\Mapping\AbstractClassMetadataFactory->loadMetadata() at /var/www/volkszaehler.org/vendor/doctrine/orm/lib/Doctrine/ORM/Mapping/ClassMetadataFactory.php:78
 Doctrine\ORM\Mapping\ClassMetadataFactory->loadMetadata() at /var/www/volkszaehler.org/vendor/doctrine/persistence/lib/Doctrine/Persistence/Mapping/AbstractClassMetadataFactory.php:185
 Doctrine\Persistence\Mapping\AbstractClassMetadataFactory->getMetadataFor() at /var/www/volkszaehler.org/vendor/doctrine/persistence/lib/Doctrine/Persistence/Mapping/AbstractClassMetadataFactory.php:91
 Doctrine\Persistence\Mapping\AbstractClassMetadataFactory->getAllMetadata() at /var/www/volkszaehler.org/vendor/doctrine/orm/lib/Doctrine/ORM/Tools/Console/Command/SchemaTool/AbstractCommand.php:62
 Doctrine\ORM\Tools\Console\Command\SchemaTool\AbstractCommand->execute() at /var/www/volkszaehler.org/vendor/symfony/console/Command/Command.php:255
 Symfony\Component\Console\Command\Command->run() at /var/www/volkszaehler.org/vendor/symfony/console/Application.php:1012
 Symfony\Component\Console\Application->doRunCommand() at /var/www/volkszaehler.org/vendor/symfony/console/Application.php:272
 Symfony\Component\Console\Application->doRun() at /var/www/volkszaehler.org/vendor/symfony/console/Application.php:148
 Symfony\Component\Console\Application->run() at /var/www/volkszaehler.org/vendor/doctrine/orm/lib/Doctrine/ORM/Tools/Console/ConsoleRunner.php:63
 Doctrine\ORM\Tools\Console\ConsoleRunner::run() at /var/www/volkszaehler.org/bin/doctrine:38

orm:schema-tool:create [--dump-sql]
```

</details>

Doctrine CLI mysql have been verified to have correct DB rights etc. Also providing a wrong CLI user does not change the error message. It seems to crap out before even trying to connect.

Other applications, e.g. nextcloud, which are also using Doctrine are installing fine.

Any pointer would be appreciated.

andig, 2020-02-26:
Sounds a lot like a database error. Please check your log files.

Sineos, 2020-02-26:
Sorry, I should have mentioned it. The syslog and mysql/error.log show absolutely nothing. And, e.g. nextcloud is running fine with the mariadb server. Also phpmyadmin is working without any issues.

I dropped the volkszaehler table and db-users and tried running the /bin/install.sh script again:
- Creating the vz-admin user works
- Creating the vz user works
- Creating the volkszaehler table works
When it comes to the doctrine part it errors out with the above messages.

r00t-, 2020-02-27:
copying my comment from the MR:

please note that the prime difference between using `127.0.0.1` and `localhost` is,   
that `localhost` has mysql using a  unix-socket connection instead of tcp (thus completely bypassing dns and such),   
while `127.0.0.1` enforces use of tcp.

i can't tell which of those we want,    
actually `localhost` seems desireable,   
but the improvement you are seeing with this change is most likely not related to the actual value, but to the bypassing of tcp.

closed 2020-03-31 by andig