volkszaehler/volkszaehler.org #807: timestamp*1000 hinders easy data sharing with other visualisation tools

ogbrugge, 2020-06-04:
Upfront: I'm willing to do some of the work in order to convert to a data type that supports fractional seconds.

Problem for me is the silly casting and divide by 1000 every time i need to create a query in say grafana or some other tool that expects a timestamp and not an INTEGER. I use MySQL/MariaDB so for me there would be a DB data type like DATETIME(3) or TIMESTAMP(3) that has the fractional second property that i am looking for but i am not sure what the status is for other databases that Volkszaehler supports.

Any pointers as to where to start looking to start a conversion, if deemed possible, would be nice as well.

andig, 2020-06-04:
Thank you for a very interesting question! 

If you ask me then choosing INT over TIMESTAMP is a very old design choice pre MySQL 5, reasoning probably lost in the mists of time. I agree that manual conversion is painful and our code is quite full of INT/TIME conversions in various places, especially the aggregation code.

According to https://stackoverflow.com/questions/409286/should-i-use-the-datetime-or-timestamp-data-type-in-mysql It seems TIMESTAMP would be the way to go forward.

Considerations for making this change are:
- API continuity (JSON)
- Database schema (easily done by Doctrine) and data (?) conversion
- Performance impact

I would agree to look at a PR addressing this, however it would need to give answers to all points above. Volkszaehler is a pretty mature project and development pace has slowed down. Resources are scarce. PR author would need to commit to support follow-on issues. I could help in coaching.

ogbrugge, 2020-06-04:
Ok, I think API should stay as is, just add new API instead fo replacing, old API will simply convert to new type, new API will store as passed in. No hassle here.
Schema conversion, I'll look into Doctrine and see what is possible. Regarding data conversion, this will run for a long time on large DBs so either you lose data coming in while converting or you need to convert the data coming in to the new format before doing the conversion of the old data so conversion and new data can keep going at the same time.
Performance impact, i think only the old API will notice a slight bit of performance impact, the rest should speed up because i assume the DB is faster doing the processing on the native datatype than php is with dividing and multiplying by 1000 all the time.

I'll look into the aggregation code later today to get a feel for the amount of work needed to do the code conversion.

andig, 2020-06-04:
> Ok, I think API should stay as is, just add new API instead fo replacing, old API will simply convert to new type, new API will store as passed in. No hassle here.

I don't even think we need a new API as long as your integration pain is on the database level? If you need more then need to think about what a REST api equivalent of a TIMESTAMP really is.

> Regarding data conversion, this will run for a long time on large DBs so either you lose data coming in while converting or you need to convert the data coming in to the new format before doing the conversion of the old data so conversion and new data can keep going at the same time.

The aggregation helper utility will create a new table for doing this and swap the tables in the end. For this migration which affects only the data table (aggregate could later be rebuilt) it would suffice to add a column, populate that successively and swap columns when finished.

> Performance impact, i think only the old API will notice a slight bit of performance impact, the rest should speed up because i assume the DB is faster doing the processing on the native datatype than php is with dividing and multiplying by 1000 all the time.

It is not really doing that except in aggregation. I would not expect any change but it would be good to double-check.

andig, 2020-06-04:
I've whipped up a quick PR that might help to provide some basic infrastructure assuming that doctrine migrations might be helpful:

- added `bin/migrate`
- added `migrations` table to track progress

andig, 2020-06-04:
> Consideration needed?

You can use influx with any precision. Not sure how this should be related to the internal MySQL representation?

jahir, 2020-06-04:
It was just a historic design decision to directly use javascript timestamps without any conversion, probably because it was the easiest way at that time. Maybe not the best decision, I see not that big problem with it. There is a [request parameter](https://wiki.volkszaehler.org/development/api/reference#parameter) `tsfmt` where you can change the output timestamp format. To get your preferred format just add it to the `formatTimestamp`  function in `lib/View/View.php`. Unfortunately, it is not used by the JSON view, but it shouldn't be to hard to add it there. Something like this for the tuples:
```
--- a/lib/View/JSON.php
+++ b/lib/View/JSON.php
@@ -217,7 +217,9 @@ class JSON extends View {
                        $this->renderContent();

                        if ($key) echo(',');
-                       echo('[' . $tuple[0] . ',' . View::formatNumber($tuple[1]) . ',' . $tuple[2] . ']');
+                       $ts = $this->formatTimestamp($tuple[0]);
+                       if (!is_numeric($ts)) $ts = '"' . $ts . '"';
+                       echo('[' . $ts . ',' . View::formatNumber($tuple[1]) . ',' . $tuple[2] . ']');
                }

                // render buffered content if not rendered inside foreach loop due to Interpreter empty
```
and in the same style for `from`, `to`, `max`, `min`.

The quoting stuff of non-numeric timestamps is a little bit ugly and you need it multiple times, so it's probably better to extend `formatTimestamp()` with a boolean parameter that adds quotes if necessary.

jahir, 2020-06-04:
Oh, and if you feel the need to get something else than javascript timestamps out of the database, why not just add a database view that does the calculations?

Not saying it would not be a good thing to change the timestamp format in the database, just a little frightened about the migration...

ogbrugge, 2020-06-04:
> I don't even think we need a new API as long as your integration pain is on the database level? If you need more then need to think about what a REST api equivalent of a TIMESTAMP really is.

If timestamp passed to the API contains a '.', all is fine, new format, if it does not contain the '.' then it could be old or new without a dot, then it needs to be checked if the passed timestamp/1000 is a valid timestamp in the old format or whether it is a new format. So how do you now check whether the passed timestamp could be valid? I think you do not do such checks because I have passed in timestamps as is without multiplying by 1000 and these values get accepted without any issue, they are then not visible in the visualisation because they are way in the past...

ogbrugge, 2020-06-04:
> 
> 
> Oh, and if you feel the need to get something else than javascript timestamps out of the database, why not just add a database view that does the calculations?
> 
> Not saying it would not be a good thing to change the timestamp format in the database, just a little frightened about the migration...

Not saying it is a bad thing to use the view but it feels unnecessary because apparently all supported DBs have a timestamp type that can do so and could reduce complexity all over the place if i understand what andig is saying. In the long run this would be an improvement.

Yes testing the conversion will be the biggest problem, I'll gladly spend a lot of cycles making sure that works as expected. I have a VZ DB with almost 2 years of data now so i will definitely test on that.

andig, 2020-06-04:
One final word of warning since Jacob pointed it out. As you‘re using the api- I wasn‘t aware of that- enhancing the api is straightforward and minimally invasive. Changing the internals is the opposite. As VZ is quite grown-up I tend to not propose merging the discussed solution to avoid wide-spread breaking of users installations, confusion and support efforts.

andig, 2020-06-04:
Btw, what visualization tool are you even using vz with? Grafana? Then you could use gravo which already does the transformation for you.

ogbrugge, 2020-06-04:
Yes i use Grafana, I'll look into Gravo.

I fully understand the change is invasive but i do not want to maintain a fork so i'll do the change, create a PR and then let things roll from there on, if backwards compatibility is desperately needed one can simply add a second timestamp column with the new type and fill both, i hate redundancy like this but for compatibility it would be the easiest solution.

jahir, 2020-06-04:
> Not saying it is a bad thing to use the view but it feels unnecessary because apparently all supported DBs have a timestamp type that can do so and could reduce complexity all over the place if i understand what andig is saying. In the long run this would be an improvement.

First, to be clear, is this about storage or the API (or both)?
If storage, I don't see why one should care much. I see that it would be nice to use the DB's native timestamp types, but from a developer's point of view, it doesn't really matter that much what the exact column type is, we just use the one that fits our needs best. Our needs are a) easy handling (e.g. calculations) and b) serve to the frontend in the appropriate format.
Which brings us to the API: The main use case of VZ installations is to use it with the included web frontend, which requires JS timestamps (actually, the plot lib we use), so we would end up mass converting timestamps just to be a little more efficient and "clean" at the other end.

So, I don't see the big gain in using the native type (happy to be convinced otherwise), even in the long run (what's the benefit?). And it's a bunch of work with high potential of breakage, so the cost-value ratio is quite bad. Plus, it _will_ break all the scripts the people directly access the DB with (which is not considered good style, but shouldn't break without good reasons). Plus, somebody has to review your changes and do thorough testing, which means it will not only cost your own time but also the time of somebody familiar with the internals, which could probably be used more reasonable. Plus the support and documentation effort. So I'm afraid I'm in line with andig here: extending the API? yay. turning internals inside out? nay.

ogbrugge, 2020-06-08:
I'll drop my proposal then, no need to argue any further if the change is considered to be potentially harmful. I can only do tests on my local machine so this will not help anyone but myself, no worries, all is fine, i'll use gravo.

closed 2020-06-08 by ogbrugge