volkszaehler/volkszaehler.org #809: Flowrate is unsigned

ogbrugge, 2020-06-08:
Flowrate is an unsigned data type, I have microcirculation in my heating system at times where the heat meter starts returning negative values to indicate the flow is against the normal direction but because of the unsigned type the values become >4 billion. Is it easily possible (without data corruption or anything like this) to change the underlying type to signed or are too many location depending on it?

I can of course again create a view over this where i reinterpet the returned value but i am showing this value in volkszaehler itself, not in a different visualisation tool.

ogbrugge, 2020-06-08:
[image]

andig, 2020-06-08:
> I have microcirculation in my heating system at times where the heat meter starts returning negative values to indicate the flow is against the normal direction but because of the unsigned type the values become >4 billion.

We don't have unsigned types afaik.

ogbrugge, 2020-06-08:
This is part of the mbus-cmd perl script that gets delivered with volkszaehler, or at least used to be delivered, I did not check if it is still there. Anyway I took it from contrib and modified it a bit to suit my needs. If the value becomes larger than 2^31 i deduct 2^32 and get a negative number and pass that to the API but then the DB still stores it as a positive number or at least interprets it as a positive number.

Now that i read the code again i need to log the output in verbose mode, i cannot be sure at this time that the value is actually passed as a negative number, will report tomorrow or later when this situation occurs again.

Snippet from mbus.cmd:
                if (defined($vzUuid{$zaehlerStr})) {
                    my $uuid=$vzUuid{$zaehlerStr};
                    if ($uuid ne "") {
                        my $sendval;
                        print "Zweck is $zweck\n" if ($verbose);
                        if ( ($zweck eq "Heizenergie") || ($zweck eq "Volumen") ) {
                            # For these we need a duplicate value filter
                            if (defined($vzOld{$uuid})) {
                                if ($value eq $vzOld{$uuid}) {
                                    $sendval=undef;
                                } else {
                                    $sendval=$value;
                                    $vzOld{$uuid}=$value;
                                }
                            } else {
                                $vzOld{$uuid} = $value;
                                $sendval = $value;
                            }
                        } elsif ($zweck eq "Fliessgeschwindigkeit") {
                            **# Support negative flow speed (aka reverse direction)
                            $value -= 2**32 if $value >= 2**31;
                            $sendval = $value;**
                        } else {
                            $sendval = $value;
                        }
                        if (defined($sendval)) {
                            print "Sending POST ".$vzURL."data/".$uuid.".json  (time=$ts value=$sendval)\n" if ($verbose);
                            my $response =$browser->post(($vzURL."data/".$uuid.".json"),[
                                "ts" => $ts,
                                "value" => $sendval
                            ]);
                            if (! $response->is_success) {
                                warn "$vzURL $uuid error: ", $response->status_line;
                                open(UNPOSTEDDATA, ">>$vzUnpostedDataFile\n");
                                print UNPOSTEDDATA "$uuid\t$ts\t$value\n";
                                close(UNPOSTEDDATA);
                            }
                        } else {
                            print "NOT Sending POST ".$vzURL."data/".$uuid.".json  (value not changed)\n" if ($verbose);
                        }
                    }

andig, 2020-06-08:
> If the value becomes larger than 2^31 i deduct 2^32 and get a negative number and pass that to the API but then the DB still stores it as a positive number or at least interprets it as a positive number.

The db will store whatever you put in. It has float datatype.

ogbrugge, 2020-06-08:
Odd, ok DB contents clearly show values like below which means the negative value does not arrive in the DB, will have to wait until i can verify that the data is inserted properly through the API:

  | Bearbeiten | Kopieren | Löschen | 4294967.263
-- | -- | -- | -- | --
  | Bearbeiten | Kopieren | Löschen | 4294967.264
  | Bearbeiten | Kopieren | Löschen | 4294967.263
  | Bearbeiten | Kopieren | Löschen | 4294967.264
  | Bearbeiten | Kopieren | Löschen | 4294967.263
  | Bearbeiten | Kopieren | Löschen | 4294967.264
  | Bearbeiten | Kopieren | Löschen | 0
  | Bearbeiten | Kopieren | Löschen | 4294967.263
  | Bearbeiten | Kopieren | Löschen | 0

ogbrugge, 2020-06-13:
Problem found as far as i can tell, due to the fact that the mbus value that is returned is a 32 bit integer where the last 3 digits are the decimal digits. The value read is thus never larger than 4 million so my test cannot work, i modified the test to multiply by 1000 and divide the result again by 1000 so the condition >= 2**31 becomes true in case of negative values. I will close this bug as invalid. Sorry for the inconvenience.

closed 2020-06-13 by ogbrugge