volkszaehler/volkszaehler.org #818: ppm Worker Anzahl

StefanSchoof, 2020-08-16:
Aktuell wird die worker Anzahl auf 8 festgelegt. Mit 8 workern wird doppelt soviel Arbeitsspeicher wie mit einem Worker benötigt. Da die meisten Installationen ja wahrscheinlich von einem vzlogger und höchstens einem Nutzer gleichzeitig genutzt wird, scheint mir 8 zu viel. Wenn man das Arm Image jetzt laufen lässt, wäre es schön sparsamer mit den Ressourcen um zu gehen.

Mögliche Lösungen die mir einfallen:
 - in der middleware.json ändern (sehr einfach, aber kann vorhandene Installation beim Update stören)
 - im docker CMD worker Anzahl setzten (Änderungen nur für docker images)
 - einen Environment Variable für die Workeranzahl einführen und auslesen. (Aufwändiger, frage ob nötig)

Gibt es noch andere Ideen?

andig, 2020-08-16:
Worker als ENV Variable in Docker ausprägen und an ppm als cmdline Parameter übergeben? Das sollte ohne weitere Änderungen gehen?

StefanSchoof, 2020-08-18:
Sollte es auch eine Umgebungsvariable geben, die steuert, ob doctrine starten soll?
Also das man statt der Zeile:
https://github.com/volkszaehler/volkszaehler.org/blob/75ec0fedbf9441bf1767eb5304fc47c80ba9d744/docker-compose.yml#L25
ein bash script hat was prüft, ob `VZ_CREATE_SCHEMA` oä da ist. Wenn ja dann doctrine laufen lässt und dann ppm mit der Worker anzahl aus `VZ_WORKER_COUNT` startet.

Wenn die Idee in Ordnung ist würde ich mal versuchen ein PR dafür zu machen.

Wenn dann Umgebungsvariablen in der Config Klasse Werte aus der config.yaml überschreiben könnten, würde der lange cmd im docker-compose entfallen und nur noch über Umgebungvariablen zu steuern.

andig, 2020-08-19:
> Sollte es auch eine Umgebungsvariable geben, die steuert, ob doctrine starten soll?

Eigentlich nciht. Die Idee ist ja dass der Anwender genau nicht entscheiden muss ob die DB initialisiert werden muß?

StefanSchoof, 2020-08-19:
Aktuell startet das Dockerfile nicht doctrine
https://github.com/volkszaehler/volkszaehler.org/blob/75ec0fedbf9441bf1767eb5304fc47c80ba9d744/Dockerfile#L28
Um das zu erreichen muss in der docker-compose file doctrine als command aufgeführt werden.

Nach meinen Verständnis ist die docker-compose Datei, nur ein Beispiel. z.B. die Datenbank speichert die Daten nur im container, so das beim recreate Daten verloren gehen. Wenn ich nun mein docker-compose baue, muss ich die Zeile aus dem docker-compose kopieren. 

Daher würde ich es schöner finden, wenn der doctrine Befehl in dem dockerfile drin wäre. Da ich mir verstellen kann das nicht jeder ein Datenbank Schema Update beim Container start machen will, ist die Idee das über eine Umgebungsvariable beim dockerfile zu machen und nicht im compose.

andig, 2020-08-20:
> Daher würde ich es schöner finden, wenn der doctrine Befehl in dem dockerfile drin wäre. Da ich mir verstellen kann das nicht jeder ein Datenbank Schema Update beim Container start machen will, ist die Idee das über eine Umgebungsvariable beim dockerfile zu machen und nicht im compose.

Kannst Du gerne machen. Da das docker-compose aber nur ein Beispiel ist sollte das Beispiel am Ende so gestrickt sein, dass Doctrine auch ausgeführt wird. Wie wär mir eigentlich egal- aktuell zeit das mehr was passieren *müsste*.