volkszaehler/volkszaehler.org #819: Future of the project; The elephant in the room – Grafana

ypid, 2020-08-16:
Hi there,

I have been a long-term volkszaehler.org user and supporter (I wrote an Ansible role for it). However, It always bugged me that uses MySQL, a non-timeseries database. Today, there are far more widely used tools like Grafana and InfluxDB out there. Even the Elastic Stack could be (mis-) used for this.

How do you see the future of the project? Would it maybe make sense to rebuild what we have with volkszaehler.org as a Grafana dashboard?

andig, 2020-08-16:
Don‘t think so. Volkszaehler consists of vzlogger and the frontend/data persistency. Afaik you can use vzlogger with influxdb already. I don‘t see much value for providing anything on top of influxdb. 

But I may be missing something- what exactly would you see as part of such a rebuild?

ypid, 2020-08-16:
Ah, sorry. I mainly talk about the PHP frontend and the MySQL backend. I am not sure about the surrounding programs (vzlogger) and what possible alternatives are (maybe none, could be). My idea about the surrounding programs would be to cover them by general-purpose home automation hubs like Home Assistant, if possible. Because at the end of the day, you still want the data in one place to look at recent data and do automation.

> I don‘t see much value for providing anything on top of influxdb.

The value is avoiding redundancy and working collectively on one solution. Grafana is the general purpose metrics analytics frontend you are looking for. It has not existed when volkszaehler.org was started.

andig, 2020-08-17:
@ypid sorry if my question was not clear enough. I‘d like to understand what you think should be developed in Grafana. Grafana is more or less what the Volkszaehler frontend is today at already seems to provide the necessary visualization capabilities.

What exactly would you see to be part of a Grafana-based solution?

ypid, 2020-08-17:
I first wanted to understand your direction of the project. When you now say the direction is clear and will be Grafana, then a clear statement by the project and migration steps would help existing users. Tell me if I am too fast with that conclusion.

I agree with you that nothing in Grafana really needs to be developed.

andig, 2020-08-18:
> When you now say the direction is clear and will be Grafana, then a clear statement by the project and migration steps would help existing users.

To be clear: we have a working solution and will continue supporting our users. The upgrade path to Grafana is already seamless as we're providing https://github.com/andig/gravo which can be used as datasource for the Volkszähler API. I don't see the need to migrate anything.

ypid, 2020-08-18:
> I don't see the need to migrate anything.

OK, and what would you recommend to new users?

r00t-, 2021-03-25:
i use vzlogger+influxdb+grafana.

the only downside is that grafana does not support volkszaehler.org's "interpreters" to convert meter data on the fly.
(while it can do simply mathematical conversions, it can not calculate anything from raw s0 pulses for example.)

narc-Ontakac2, 2021-11-30:
The problem with grafana is that it is for amd64 (as a debian arcitecture) only. So it won't work on the raspberry pi image.

If you separate the backend and run it on a suitable machine grafana it will however also work with mysql.

belegdol, 2026-10-29:
I run a grafana dashboard which reproduces the main aspects of the middleware, including calculating power from S0. The main advantage of middleware I see at the moment is that vzlogger can buffer middleware records for a while. Influxdb channel cannot do this currently. Especially for S0 this has a potential to break months of records after a short downtime.

belegdol, 2026-10-29:
Here it is. Due to me using armv7hl and thus influxdb-1.8.10 locally, both flux and influxql sources are needed. InfluxQL for performance since flux does not support pushdown in this old version, and flux for S0 meter because `ELAPSED()` and `GROUP BY time()` [do not play along nicely](https://docs.influxdata.com/influxdb/v1/query_language/functions/#elapsed-with-group-by-time-clauses).
MySQL source is there for parity checking.
```json
{
  "__inputs": [
    {
      "name": "DS_INFLUXDB-INFLUXQL",
      "label": "influxdb-influxql",
      "description": "",
      "type": "datasource",
      "pluginId": "influxdb",
      "pluginName": "InfluxDB"
    },
    {
      "name": "DS_INFLUXDB-FLUX",
      "label": "influxdb-flux",
      "description": "",
      "type": "datasource",
      "pluginId": "influxdb",
      "pluginName": "InfluxDB"
    },
    {
      "name": "DS_VOLKSZÄHLER",
      "label": "volkszähler",
      "description": "",
      "type": "datasource",
      "pluginId": "mysql",
      "pluginName": "MySQL"
    }
  ],
  "__elements": {},
  "__requires": [
    {
      "type": "panel",
      "id": "barchart",
      "name": "Bar chart",
      "version": ""
    },
    {
      "type": "grafana",
      "id": "grafana",
      "name": "Grafana",
      "version": "13.2.3"
    },
    {
      "type": "datasource",
      "id": "influxdb",
      "name": "InfluxDB",
      "version": "13.1.7"
    },
    {
      "type": "datasource",
      "id": "mysql",
      "name": "MySQL",
      "version": "13.1.4"
    },
    {
      "type": "panel",
      "id": "stat",
      "name": "Stat",
      "version": ""
    },
    {
      "type": "panel",
      "id": "timeseries",
      "name": "Time series",
      "version": ""
    }
  ],
  "annotations": {
[... 1063 lines cut ...]
  "time": {
    "from": "now/d",
    "to": "now"
  },
  "timepicker": {
    "refresh_intervals": [
      "5s",
      "10s",
      "30s",
      "1m",
      "5m",
      "15m",
      "30m",
      "1h",
      "2h",
      "1d"
    ]
  },
  "timezone": "browser",
  "title": "vzlogger",
  "uid": "admbn8n",
  "version": 113,
  "weekStart": ""
}
```

r00t-, 2026-10-31:
@belegdol: that's interesting and useful, thanks.
(but a big lengthy for a discussion post, should probably be in a repository or gist.)
do you happen to know if grafana provides a path to import such a configuration (other than by copy/pasting the text) so it could be shipped to users?

belegdol, 2026-10-31:
Yes, you can import a JSON. This is how I got my [ems-esp dashboard](https://grafana.com/grafana/dashboards/15940-ems-esp/) for example. I can try sharing my vz dashboard via grafana labs too, but making nice screenshots is a challenge since rendering module does not support armv7hl.