volkszaehler/volkszaehler.org #827: PHP 7.4 ist auf der Serverseite und da tauchen erneut irreführende Fehlermeldungen auf:

WolfgangFahl, 2020-10-06:
PHP 7.4 ist auf der Serverseite und da tauchen erneut irreführende Fehlermeldungen auf:
[image]
für die Software aus 2018 die bisher lief  und konfiguriert ist.
und
[image]
für die Software mit aktuellem Stand, die nicht konfiguriert ist. Mein Problem ist nun wie ich von diesen beiden Meldungen aus zu einem funktionierenden System komme.

_Originally posted by @WolfgangFahl in https://github.com/volkszaehler/volkszaehler.org/issues/607#issuecomment-703222327_

WolfgangFahl, 2020-10-06:
Ich habe keine Ahnung wo ich anfangen soll um das Problem zu lösen. Ich habe einen vzlogger unverändert auf einem PI laufen - auf meinem Server gabs einen Ubuntu 18.04 auf 20.04 LTS Update und seitdem läuft es nicht mehr. Eine Neuinstallation mit aktuellem Stand der Software habe probiert. Hat nix geholfen. Ich habe im Wiki gesucht und findet dort nix zum Thema Debugging von Fehlermeldungen wie oben ...

frankrichter, 2020-10-06:
Hast du eine passende config.yaml mit Zugangsdaten zu deiner DB im
Verzeichnis etc/?

Hast du beim Webserver Rewriting aktiviert? Falls nicht, hilft es
alternativ, in der options.js bei Middleware 'api' zu ersetzen durch
'middleware.php'.

WolfgangFahl, 2020-10-06:
Nein eine config.yaml habe ich nicht. In der 2018 Version brauchte ich die auch nicht. In der 2020 habe ich die Konfiguration probiert aber damit würde ja mein pi nicht kommunizieren - ich weiss auch nicht was ich beim PI umstellen müsste.

https://stackoverflow.com/questions/58831791/errorexception-trying-to-access-array-offset-on-value-of-type-null verweist darauf dass es wohl ein Problem mit den Libraries wie Symfony geben könnte. Ich habe mir angeschaut, ob ich das docker image nutzen könnte aber auch das sind böhmische Dörfer für mich. Was mir fehlt ist die Verbindung zur mysql 8 Datenbank auf dem Server.  In den apache error logs findet sich: 
```
Tue Oct 06 19:05:17.943855 2020] [php7:warn] [pid 2126433] [client ...:52250] PHP Warning:  "continue" targeting switch is equivalent to "break". Did you mean to use "continue 2"? in /var/www/volkszaehler.org.2018/vendor/doctrine/orm/lib/Doctrine/ORM/UnitOfWork.php on line 2665, referer: http://vz2018.bitplan.com/vz2018/
```

was wohl aber nur ein Warning ist.

Im normalen "Apache log" findet sich:
```
... - - [06/Oct/2020:20:28:35 +0200] "POST /vz/middleware.php/data/7fb770d0-dfdb-11e3-97ce-a35538f32d73.json HTTP/1.1" 400 354 "-" "vzlogger/0.7.0 (libcurl/7.64.0 OpenSSL/1.1.1c zlib/1.2.11 libidn2/2.0.5 libpsl/0.20.2 (+libidn2/2.0.5) libssh2/1.8.0 nghttp2/1.36.0 librtmp/2.3)"
```

400 ist "Bad Request" - die middleware.php von 2018 und 2020 ist identisch - das ist ja auch nur der Entry für die magischen Dinge die Symfony dann macht - wie kann man die Lage den debuggen?

frankrichter, 2020-10-06:
Wieso sollte dein vzlogger nicht mit der aktuellen Middleware kommunizieren
können?

install.sh erzeugt eine config.yaml (= Nachfolger der
volkszaehler.conf.php). Wie hast du installiert?

WolfgangFahl, 2020-10-06:
Installation mit 
```
git clone https://github.com/volkszaehler/volkszaehler.org
```

Dann die alten Daten aus /etc der bisher funktionierenden Installation kopiert.
Tatsächlich hatte ich aber noch die middleware aus 2014 zum Abpeichern in die Datenbank im Einsatz nur das Frontend war aus 2018. Ich muss wohl in 2018 schon nicht geschafft haben die neue Middleware zu benutzen.

frankrichter, 2020-10-06:
Und die Abhängigkeiten (composer)???

WolfgangFahl, 2020-10-06:
Gabs nicht. Ich habe jetzt das install script laufen lassen so dass composer die Updates gemacht hat. Die DB-Konfiguration war etwas eigenartig weil nicht klar war was mit den existierenden Einstellungen ist. Die DB gibts ja nun seit 2014 und die hatt schon einiges an Einträgen. Nach ein paar Anläufen bin ich damit jetzt durch. Die Oberfläche zeigt jetzt 
[image]

WolfgangFahl, 2020-10-06:
Die middleware posts haben aber schon mal einen 200 Http Code das macht Hoffnung ...

frankrichter, 2020-10-06:
Was sagt die Browser-Konsole?

WolfgangFahl, 2020-10-06:
nichts. Die javascript - Netzwerkzugriffe auf die jquery -komponenten usw. tauchen auf.

WolfgangFahl, 2020-10-06:
Muss ich irgendwelche Cookies löschen?

WolfgangFahl, 2020-10-06:
Ich habe jetzt mal einen Breakpoint in 
```javascript
/**
 * Javascript functions for the frontend
 *
 * @author Florian Ziegler <fz@f10-home.de>
 * @author Justin Otherguy <justin@justinotherguy.org>
 * @author Steffen Vogel <info@steffenvogel.de>
 * @author Andreas Götz <cpuidle@gmx.de>
 * @copyright Copyright (c) 2011-2020, The volkszaehler.org project
 * @license https://www.gnu.org/licenses/gpl-3.0.txt GNU General Public License version 3
 */
/*
 * This file is part of volkzaehler.org
 *
 * volkzaehler.org is free software: you can redistribute it and/or modify it
 * under the terms of the GNU General Public License as published by the Free
 * Software Foundation, either version 3 of the License, or any later version.
 *
 * volkzaehler.org is distributed in the hope that it will be useful, but
 * WITHOUT ANY WARRANTY; without even the implied warranty of MERCHANTABILITY or
 * FITNESS FOR A PARTICULAR PURPOSE. See the GNU General Public License for more
 * details.
 *
 * You should have received a copy of the GNU General Public License along with
 * volkszaehler.org. If not, see <http://www.gnu.org/licenses/>.
 */

/**
 * Initialize the WUI (Web User Interface)
 */
vz.wui.init = function() {
	vz.wui.initEvents();
	vz.wui.resizePlot();

	// resize handling
	$(window).resize(vz.wui.resizePlot);
	$('#accordion h3').click(function() {
		// capture window height before toggling section to avoid miscalculation due to scrollbar flickering
		var windowHeight = $(window).height();
		$(this).next().toggle(0, function() {
			vz.wui.resizePlot(null, windowHeight);
		});
		return false;
	}).next().hide();
	$('#entity-list').show(); // open entity list by default

	// buttons
	$('button, input[type=button],[type=image],[type=submit]').button();
	$('button[name=options-save]').click(vz.options.saveCookies);
	$('button[name=entity-add]').click(this.dialogs.init);

	$('#export select').change(function(event) {
		vz.wui.exportData($(this).val());
		$(this).val('default');
	});

	// bind display mode change
	$('#display select').change(function(event) {
		var val = $(this).val().split('-');
		var mode = val.pop();
		var keepPeriodStartFixed = vz.options.mode !== mode;
		vz.wui.changeDisplayMode(mode);
		if (val.length) {
			vz.wui.handleControls('zoom-' + val.pop(), keepPeriodStartFixed);
		}
		else {
			var to = Math.min(new Date().getTime(), vz.options.plot.xaxis.max);
			vz.wui.zoom(vz.options.plot.xaxis.min, to);
		}
[... 817 lines cut ...]
			$(this).next().find('button:eq(0)').focus();
		}
	});
};

vz.wui.dialogs.exception = function(exception) {
	if (vz.wui.errorDialog) return; // only one error dialog at a time
	this.error(exception.type, exception.message, exception.code);
};

vz.wui.dialogs.middlewareException = function(xhr) {
	if (xhr.responseJSON && xhr.responseJSON.exception)
		// middleware exception
		this.exception(new Exception(xhr.responseJSON.exception.type, "<a href='" + xhr.requestUrl + "' style='text-decoration:none'>" + xhr.requestUrl + "</a>:<br/><br/>" + xhr.responseJSON.exception.message));
	else
		// network exception
		this.exception(new Exception("Network Error", xhr.statusText));
};

```
gesetzt und der xhr hat ganz klar eine Saubere Fehlermeldung  im reponseText dass die URL für das api 
```
api/capabilities/definitions.json?unique=1602011304708
```
nicht erreichbar ist. Die wird von der Software einfach geschlabbert.

WolfgangFahl, 2020-10-06:
warum ist api ... nicht erreichbar und was muss dafür konfigurier werden?

andig, 2020-10-06:
Du lässt uns hier mal wieder (!) völlig im dunkeln tappen. Anstatt in irgendwelchen Dateien zu wühlen wäre es hilfreich erstmal Deine Konfiguration zu teilen. Auf der Mailingliste waren auch die Schritte zur Fehlersuche beschrieben.

WolfgangFahl, 2020-10-06:
@andig - um Welche Konfiguration geht es? 

Apache?
```
# Volkszähler configuration 2020
ErrorLog  ${APACHE_LOG_DIR}/vz2020_error.log
CustomLog ${APACHE_LOG_DIR}/vz2020.log combined
# WF 2017-04-17
# WF 2018-11-04
# WF 2020-10-01
Alias /vz/  "/var/www/volkszaehler.org.2020/htdocs/"
Alias /vz2020/  "/var/www/volkszaehler.org.2020/htdocs/"
<Directory /var/www/volkszaehler.org.2020>
  Options Indexes FollowSymLinks MultiViews
	AllowOverride FileInfo Limit Options Indexes Authconfig
  Require all granted
</Directory>
```

Der Zugriff von Java aus mit http://www.bitplan.com/index.php/Vzjava funktioniert - es werden wieder Daten in die Datenbank eingetragen. Die Kommunikation zwischen PI und middleware scheint zu funktionieren. 

[image]

andig, 2020-10-06:
> Apache?

Genau. Bitte auch die htaccess und virtual hosts falls konfiguriert.

> Der Zugriff von Java aus mit http://www.bitplan.com/index.php/Vzjava funktioniert 

Toll aber ohne Aussagekraft. Siehst Du im Browser welche HTTP Requests das sind?

> Die middleware posts haben aber schon mal einen 200 Http Code das macht Hoffnung ...

Dito. Siehst Du im Browser welche HTTP Requests das sind? Konkrete URL?

Dann die fehlerhaften. Welche sind es die nicht gehen? Gibt es Unterschiede?

frankrichter, 2020-10-06:
Siehe mein erster Post oben. Vermutlich funktioniert das Rewriting von api
zu middleware.php nicht.

andig, 2020-10-06:
> Siehe mein erster Post oben. Vermutlich funktioniert das Rewriting von api
zu middleware.php nicht.

Was dann nurmalerweise im Apache Log auftauchen würde. Da wir nur 1 Zeile sehen kann man nur raten :(

andig, 2020-10-06:
> Genau. Bitte auch die htaccess und virtual hosts falls konfiguriert.

Und die options.js bitte.

WolfgangFahl, 2020-10-07:
Kommt jetzt so mit und mit. Fangen wir mit LAMP an:
**Linux: Ubuntu 20.04.1 LTS**

```bash
uname -a
Linux capri 5.4.0-48-generic #52-Ubuntu SMP Thu Sep 10 10:58:49 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.1 LTS
Release:	20.04
Codename:	focal
```
**Apache**
```bash
apache2 -version
Server version: Apache/2.4.41 (Ubuntu)
Server built:   2020-08-12T19:46:17
``
**MySQL **
```
mysql --version
mysql  Ver 8.0.21-0ubuntu0.20.04.4 for Linux on x86_64 ((Ubuntu))
```
**PHP**
```bash
 php --version
PHP 7.4.3 (cli) (built: May 26 2020 12:24:22) ( NTS )
Copyright (c) The PHP Group
Zend Engine v3.4.0, Copyright (c) Zend Technologies
    with Zend OPcache v7.4.3, Copyright (c), by Zend Technologies
```

WolfgangFahl, 2020-10-07:
Apache log:
```
127.0.0.1 - - [07/Oct/2020:11:15:01 +0200] "POST /vz/middleware.php/data/0cca0870-e695-11e3-8c38-6dc719f70ac1.json HTTP/1.1" 200 231 "-" "-"
127.0.0.1 - - [07/Oct/2020:11:15:01 +0200] "POST /vz/middleware.php/data/b8192a00-e71e-11e3-be31-e59be3e5a943.json HTTP/1.1" 200 231 "-" "-"
127.0.0.1 - - [07/Oct/2020:11:15:11 +0200] "POST /vz/middleware.php/data/7744bbf0-e74d-11e3-9ec7-11d61aec53d8.json HTTP/1.1" 200 231 "-" "-"
127.0.0.1 - - [07/Oct/2020:11:15:11 +0200] "POST /vz/middleware.php/data/460feba0-e74f-11e3-8a0d-9d48fc10b5f4.json HTTP/1.1" 200 231 "-" "-"
```
error log ist leer.

WolfgangFahl, 2020-10-07:
options.js ist wie aus git clone unverändert.

WolfgangFahl, 2020-10-07:
dito für die beiden .htaccess Dateien

WolfgangFahl, 2020-10-07:
Ein vzcompress run war erfolgreich. aggregate scheint nicht zu laufen, weil der admin-user fehlt. Wir der standardmässig gebraucht? Ich melde mich wenn ich den eingetragen habe und der aggregate crontab Eintrag wieder läuft. Das die Oberfläche nicht geht ist zur Zeit nicht so schlimm weil ich ja die vzjava Alternative habe und die zeigt im Moment korrekt die gewünschten Kanäle. Die Datenzulieferung läuft also.

andig, 2020-10-07:
> "POST /vz/middleware.php/data/0cca0870-e695-11e3-8c38-6dc719f70ac1.json HTTP/1.1" 200

Damit wissen wir dass Dein VZ unter /vz/middleware.php.... erreichbar ist. 

> options.js ist wie aus git clone unverändert.

D.h. es gilt dieses hier: https://github.com/volkszaehler/volkszaehler.org/blob/master/htdocs/js/options.js#L42. Die url ist aber "api".

> dito für die beiden .htaccess Dateien

D.h. es gilt https://github.com/volkszaehler/volkszaehler.org/blob/master/htdocs/.htaccess#L22

Jetzt nochmal: was sagt die Browserkonsole wenn das Frontend/API aufgerufen wird?

> Ein vzcompress run war erfolgreich.

Offtopic. Darum ging es hier nicht- bitte nicht alles durcheinander werfen.

J-A-U, 2020-10-07:
> Apache log:
>
> ```
> 127.0.0.1 - - [07/Oct/2020:11:15:01 +0200] "POST /vz/middleware.php/data/0cca0870-e695-11e3-8c38-6dc719f70ac1.json HTTP/1.1" 200 231 "-" "-"
> 127.0.0.1 - - [07/Oct/2020:11:15:01 [...]

Offensichtlich fanden in dem Zeitraum keine Zugriffe übers Frontend statt.
Daten an die MW schieben funktioniert weil du die MW zufällig direkt ansprichst, am bereits erwähnten rewrite vorbei.

Generell sei angeraten bei Updates auch eine Blick in die Installations- und Konfigurationsanleitung zu werfen. Es gibt einige Schritt die wir übers Script nicht abdecken können (speziell die Webserverkonfig). Das es da bei Versionswechsel zu Unstimmigkeiten kommt ist vorhersehbar.

andig, 2020-10-07:
> Jetzt nochmal: was sagt die Browserkonsole wenn das Frontend/API aufgerufen wird? Und was sagt das Apache log zu genau dem gleichen Zeitpunkt?

;)

WolfgangFahl, 2020-10-07:
after debugging with javascript and modifying the exception handling:
```javascript
vz.wui.dialogs.middlewareException = function(xhr) {
  // xhr failed
	var msg="<a href='" + xhr.requestUrl + "' style='text-decoration:none'>" 
		+ xhr.requestUrl + "</a>:<br/><br/>";
  var hint="";
  // warn if api might not have been configured
  if (xhr.requestUrl.startsWith('api/')) {
		hint="<br/></br>Did you forget to modify the middleware: entry in <a href='js/options.js'>options.js</a>? see the fineprint in <a href='https://wiki.volkszaehler.org/software/middleware/installation'>middleware installation</a>"
  }
	if (xhr.responseJSON && xhr.responseJSON.exception)
		// middleware exception
		this.exception(new Exception(xhr.response.JSON.exception.type,msg+ xhr.responseJSON.exception.message));
	else
		// network exception
		this.exception(new Exception("Network Error", msg+xhr.statusText+hint));
};
```
[image]

and I assume this is the culprit and would recommend to  adopt the above code.

WolfgangFahl, 2020-10-07:
An indeed after adding my entry:
[image]

WolfgangFahl, 2020-10-07:
There is a warning the the /ws link is not available but i hope this is another non-mandatory story ...

WolfgangFahl, 2020-10-07:
Thanks to @frankrichter  and @andig and all the other developers for making this happen:

[image]

WolfgangFahl, 2020-10-07:
To wrap up:

- I did not know that for an update if have to run "install.sh" which is essentially a composer call and updates the libraries - reading the README would have given me a hint in this direction
- I did not know nor expect that the config style of 2018 has changed and that config.yaml and options.js are now mandatory

I think the software should detect that it is not configured by itself and warn about the three points above. One of the points is covered by the change i suggested in the exception handling javascript code above. It also makes sure that the request url is shown in e.g. the "Not found" case. I'd appreciate if this bug is closed if the code is adopted. Since it's only a few line i hope we can do without a pull request.

andig, 2020-10-07:
Wir hätten den "Fehler", für den es jede Menge Ursachen geben kann, sehr schnell gefunden wenn Du die Fragen beantwortet hättest. Insbesondere: wie sehen funktionierende Requests aus und wie "fehlschlagende" Requests.

closed 2020-10-07 by andig

WolfgangFahl, 2020-10-07:
@andig 
If you don't like the hint I'd at least do:
```javascript
vz.wui.dialogs.middlewareException = function(xhr) {
        // xhr failed
	var msg="<a href='" + xhr.requestUrl + "' style='text-decoration:none'>" 
		+ xhr.requestUrl + "</a>:<br/><br/>";
	if (xhr.responseJSON && xhr.responseJSON.exception)
		// middleware exception
		this.exception(new Exception(xhr.response.JSON.exception.type,msg+ xhr.responseJSON.exception.message));
	else
		// network exception
		this.exception(new Exception("Network Error", msg+xhr.statusText));
};
```
which makes debugging easier and avoids the dreadful "Not found" with no url mentioned.

andig, 2020-10-07:
Ein PR wäre Klasse- das geht sogar online in Github.

DrCR77, 2021-01-30:
Moin! 
I am running VZ on my Strato server with an Oct 2020 version (ie with config.yaml and options.js) - installation process see [here](https://gist.github.com/DrCR77/469fe9c6d9d4f495603b808ef8a213cc). The local raspberry pi is running on an old 2015/2016 version, reads the meters and transmits the data to the Strato server. When changing the php-version on the server from 7.3 to 7.4 I get the same error as mentioned above. I just cannot make any sense of the solution adopted here. Could you please elaborate? Since I cannot even view the frontend with DB access that runs on the server and hence independently from the my old/outdated local pi I guess there is something amiss with the server installation. 
What needs to be done?

WolfgangFahl, 2021-01-30:
Ich dachte ich hätte einen PR gemacht ...

J-A-U, 2021-01-31:
Hallo Wolfgang Fahl,

du meinst PR#829?
Der ging nicht durch Travis, ist noch immer offen.

Grüße
Daniel Lauckner

notifications@github.com schrieb am Samstag, 30. Januar 2021 um 18:00 Uhr:
> Ich dachte ich hätte einen PR gemacht ...

WolfgangFahl, 2021-01-31:
Bei den paar Zeilen https://github.com/WolfgangFahl/volkszaehler.org/commit/79b0df714f1301e89dca70c045c575a7976e8630 lässt sich dass doch bestimmt händisch ändern, damit nicht andere in die gleiche Falle laufen.

DrCR77, 2021-01-31:
Hat sicher erledigt. Hab zwar nicht verstanden, was ich hier hätte wo und wie ändern sollen, aber bei der Fehlersuche festgestellt: Die Version auf dem aktuellen raspberry image von Juli 2019 läuft nicht unter PHP 7.4. Die aktuelle Version hier auf github läuft reibungslos. Danke!